tmk/tmk_keyboard #645: usb-usb converter:Filco majestouch convertible 2:no output

LeonardoYang, 2020-03-30:
Hi, I am using Filco majestouch convertible 2 with usb-usb converter which have no output,here is the debugger info, pls have a help.

*** HID device connected: t.m.k. USB to USB keyboard converter (FEED:005B:0814)

TMK:45e45691/LUFA

usb_state: usb_state: 20
usb_state: 40
usb_state: 50
usb_state: 51
BM Init
Addr:01
NC:01
HID_PROTOCOL_KEYBOARD
bNumEP:02
Cnf:01
bIfaceNum:00

bNumIface:00
Interface:00
PROTOCOL SET HID_BOOT rcode:00
SET_IDLE rcode:00
RPIPE rcode:00

Interface:01
PROTOCOL SET HID_BOOT rcode:00
SET_IDLE rcode:00
RPIPE rcode:00
BM configured
host.Task: 
which device ID is(exported form USB Device Tree Viewer):
USB\VID_04B4&PID_120E\5&1DBACDBF&0&10
\?\usb#vid_04b4&pid_120e#5&1dbacdbf&0&10#{a5dcbf10-6530-11d2-901f-00c04fb951ed}

is there anything I need to submit?
Great thanks

LeonardoYang, 2020-03-30:
updated the latest info,Great thanks for your view...

tmk, 2020-03-31:
How did you build the firmware or where did you get it from?

Try default firmware downloaded from Keymap Editor if not yet.
http://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?usb_usb

And refer this to see HID report descriptor.
https://github.com/tmk/tmk_keyboard/wiki/USB:-HID-Report-Descriptor

LeonardoYang, 2020-03-31:
I am using the firmware directly from your keymap editor,should be the latest verison, I am working on the HID descriptor with report format,will update later once i figure that out.

tmk, 2020-03-31:
Then, you have to download it again to update the latest one.
Your firmware is not the latest one.

The latest firmware should display: TMK:b813d62a+/LUFA

LeonardoYang, 2020-03-31:
last night,I went to GEEKHACK downloaded this usb-usb debug.hex(which version info is 45e45691/LUFA) to see if i could solve it by myself. I know the latest fw is b813d62a+/LUFA.,but they all have the same phenomenon.I will submit report descriptor to see if helps.
Great thanks~~

LeonardoYang, 2020-04-04:
updated with latest HID descriptor exported from USBlyzer, hope this could helps and add support for Filco majestouch convertible 2.
Great thanks for your reply.

tmk, 2020-04-04:
Thanks for the update.
I don't find any clue in its descriptors.

What does hid_listen shows with the latest firmware instead of the debug.hex?

LeonardoYang, 2020-04-05:
I update the latest FW, and the USB hid listen shows info. below:
TMK:b813d62a+/LUFA

usb_state:12

USB configured

Keyboard start.
usb_state:20
usb_state:40
usb_state:50
usb_state:51
hsot.Task:2332
usb_state:90
speed:full
usb_state:12

i press button on keyboard, and it shows nothing in USB hid listen and also in text which measns there is nothing output.