tmk/tmk_keyboard #648: working chibios version

cubiq, 2020-04-04:
What is the latest working Chibios version to be used? Documentation suggests 17.6.x but TMK calls for a mk/rules.mk file that was added around 19.1.x (but compilation fails with that version).

Latest master calls for a /rt/osal.mk that I believe now should be in /rt-nil/osal.mk (and compilation still fails with master after updating the osal.mk path).

All previous issues about this seem obsolete now.

tmk, 2020-04-04:
Chibios protocol module have not been used nor maintained actively by me and I believe it needs to be fixed somehow.

chibios.mk change in commit 8345571e broke something?

If you have idea and could make PR for this it would be appreciated.

cubiq, 2020-04-04:
> chibios.mk change in commit 8345571 broke something?

that doesn't seem the problem. I'll see if I can sort it out somehow, I'll send you a PR if I'm able to solve it

aeremichev, 2020-06-22:
Hi, I have the same problem.
Did you find a solution?

thomas-profitt, 2024-06-23:
I had this exact problem when I tried to upgrade my local tmk_keyboard repo.
If I checkout commit a976ecc and use ChibiOS stable_17.6.x and ChibiOS-Contrib chibios-17.6.x, it works normally.