tmk/tmk_keyboard #649: Can't compile: LUFA may be too old or not found

Detege, 2020-05-02:
I have been following [this tutorial on Deskthority](https://deskthority.net/viewtopic.php?f=7&t=7177&start=) but I'm stuck in the compiling phase, I'm getting this message as soon as I type the make -f Makefile command:
_dani@MacBook-Pro gh60 % make -f Makefile

../../tmk_core/protocol/lufa.mk:12: *** LUFA may be too old or not found: try 'git submodule update --init'.  Stop._

I have been trying to install LUFA but I really don't know how. I'm a total beginner on all this.

I'm running MacOs 10.15.4 and I have tried installing Mojave on another partition too, but I get the same error.

Could anyone please assist or point me on the right direction here?

tmk, 2020-05-03:
The tutorial may be too old, perhaps.
The `git submodule` should work as the message says.

This is also a bit old but it is still useful to check.https://github.com/tmk/tmk_keyboard/wiki#build-firmware

Detege, 2020-05-03:
WOOOP! Thanks a ton!
I made a new clone of the whole repository including all submodules and it is compiling fine now :)

closed 2020-05-03 by Detege