tmk/tmk_keyboard #650: adb_usb letters "er" getting changed to "n"

skalg, 2020-05-07:
Sometimes when I write the word "docker" the letters "er" are changed to "n". So I get "dockn".
I have attached the result of the debug of writing "docker" quickly "dockn" and by pressing and releasing each key (the correct one). Is it a software problem or am I doing something that shouldn't be? Thank you

```
dockn

keyboard: 00 00 07 00 00 00 00 00
keyboard: 00 00 07 12 00 00 00 00
keyboard: 00 00 00 12 00 00 00 00
keyboard: 00 00 00 00 00 00 00 00
keyboard: 00 00 06 00 00 00 00 00
keyboard: 00 00 06 0E 00 00 00 00
keyboard: 00 00 06 0E 11 00 00 00
keyboard: 00 00 06 0E 00 00 00 00
keyboard: 00 00 00 0E 00 00 00 00
keyboard: 00 00 00 00 00 00 00 00

docker

keyboard: 00 00 07 00 00 00 00 00
keyboard: 00 00 00 00 00 00 00 00
keyboard: 00 00 12 00 00 00 00 00
keyboard: 00 00 00 00 00 00 00 00
keyboard: 00 00 06 00 00 00 00 00
keyboard: 00 00 00 00 00 00 00 00
keyboard: 00 00 0E 00 00 00 00 00
keyboard: 00 00 00 00 00 00 00 00
keyboard: 00 00 08 00 00 00 00 00
keyboard: 00 00 00 00 00 00 00 00
keyboard: 00 00 15 00 00 00 00 00
keyboard: 00 00 00 00 00 00 00 00

        - Version -
DESC: convert ADB keyboard to USB
VID: 0xFEED(t.m.k.) PID: 0x0ADB(ADB keyboard converter) VER: 0x0101
BUILD: ce5e565f+ (09:02:13 Feb 13 2020)
OPTIONS: LUFA MOUSEKEY EXTRAKEY CONSOLE COMMAND KEYMAP_SECTION 4096
GCC: 5.4.0 AVR-LIBC: 2.0.0 AVR_ARCH: avr5

        - Status -
host_keyboard_leds(): 1
keyboard_protocol: 0
keyboard_idle: 0
timer_read32(): 5ACF0126
```

tmk, 2020-05-07:
It seems to me like ghost key or phantom key.
I guess  you get 'n' with pressing any 3-key combination of 'd o k e r' simultaneously.
Hold 'ker' and 'oer' or other combination to see what happens.

What keyboard are you using?
I think my ADB keyboards are 2-key block and can't register third key at all. They should not have ghost key, though.

skalg, 2020-05-07:
I am not able to reproduce it if I press 2 or 3 keys at the same time no matter the combination. "ker", "cke", "cker". But I am able to reproduce it by writing the whole word quickly. My keyboard is a Apple Extended Keyboard II M3501 with ATmega32U4

tmk, 2020-05-07:
Then, you can add code below to see scan codes.

```
diff --git a/converter/adb_usb/matrix.c b/converter/adb_usb/matrix.c
index f4bf50c1..714bb484 100644
--- a/converter/adb_usb/matrix.c
+++ b/converter/adb_usb/matrix.c
@@ -387,6 +387,7 @@ uint8_t matrix_scan(void)
         tick_ms = timer_read();
 
         codes = adb_host_kbd_recv(ADB_ADDR_KEYBOARD);
+        if (codes) xprintf("%04X ", codes);
 
         // Adjustable keybaord media keys
         if (codes == 0 && has_media_keys &&
```

skalg, 2020-05-07:
dockn
02FF keyboard: 00 00 07 00 00 00 00 00
821F keyboard: 00 00 00 00 00 00 00 00
keyboard: 00 00 12 00 00 00 00 00
08FF keyboard: 00 00 12 06 00 00 00 00
9FFF keyboard: 00 00 00 06 00 00 00 00
28FF keyboard: 00 00 0E 06 00 00 00 00
2DFF keyboard: 00 00 0E 06 11 00 00 00
AD88 keyboard: 00 00 0E 06 00 00 00 00
keyboard: 00 00 0E 00 00 00 00 00
A8FF keyboard: 00 00 00 00 00 00 00 00
24FF keyboard: 00 00 28 00 00 00 00 00
8FFF A4FF keyboard: 00 00 00 00 00 00 00 00

docker
02FF keyboard: 00 00 07 00 00 00 00 00
82FF keyboard: 00 00 00 00 00 00 00 00
1FFF keyboard: 00 00 12 00 00 00 00 00
9FFF keyboard: 00 00 00 00 00 00 00 00
08FF keyboard: 00 00 06 00 00 00 00 00
88FF keyboard: 00 00 00 00 00 00 00 00
28FF keyboard: 00 00 0E 00 00 00 00 00
A8FF keyboard: 00 00 00 00 00 00 00 00
0EFF keyboard: 00 00 08 00 00 00 00 00
8EFF keyboard: 00 00 00 00 00 00 00 00
0FFF keyboard: 00 00 15 00 00 00 00 00
8FFF keyboard: 00 00 00 00 00 00 00 00

tmk, 2020-05-07:
That output itself is no problem  and what is expected.

You got 'enter' in addition to 'dockn' this time?

It seems to receive scan codes sent from keyboard. It still looks to me like phantom key or keyboard hardware problem.

Try this patch to see if there is difference. You may need `make clean` before build.

```
diff --git a/converter/adb_usb/Makefile b/converter/adb_usb/Makefile
index fe94d9f7..9c195005 100644
--- a/converter/adb_usb/Makefile
+++ b/converter/adb_usb/Makefile
@@ -54,7 +54,7 @@ ARCH ?= AVR8
 F_USB ?= $(F_CPU)
 
 # Interrupt driven control endpoint task(+60)
-OPT_DEFS += -DINTERRUPT_CONTROL_ENDPOINT
+#OPT_DEFS += -DINTERRUPT_CONTROL_ENDPOINT
 
 
 # Boot Section Size in *bytes*
diff --git a/converter/adb_usb/matrix.c b/converter/adb_usb/matrix.c
index f4bf50c1..714bb484 100644
--- a/converter/adb_usb/matrix.c
+++ b/converter/adb_usb/matrix.c
@@ -387,6 +387,7 @@ uint8_t matrix_scan(void)
         tick_ms = timer_read();
 
         codes = adb_host_kbd_recv(ADB_ADDR_KEYBOARD);
+        if (codes) xprintf("%04X ", codes);
 
         // Adjustable keybaord media keys
         if (codes == 0 && has_media_keys &&
```

tmk, 2020-05-07:
Found key matrix here.
https://ttic.uchicago.edu/~cotter/projects/aek2/

The n can be appeared as phantom with c e k if AEK firmware has a bug. Though I never heard of bug like that on the AEK so far.

skalg, 2020-05-07:
Same thing with the latest patch. So it's an AEK bug ?

tmk, 2020-05-08:
I'd say so.

closed 2020-05-08 by skalg