tmk/tmk_keyboard #651: MC9S08JM16 support

szafranski, 2020-05-26:
Hello, would I be able to program chip "MC9S08JM16" with that software? Any additional info about it? Thanks!

tmk, 2020-05-26:
I don't think so. I don't know about the chip at all.

closed 2020-05-26 by tmk

szafranski, 2020-05-27:
OK, I should probably explain little more why I am asking. I have FNatic rush G1 that seems someone broken but I was thinking about trying to flash new firmware on it. Any thoughts what might be helpful? Sorry if not 100% related. Thanks!

tmk, 2020-05-27:
To port this project into the microcontroller,
1. you need to write codes specific to it under `common/s08`,  you can refer to codes for AVR for example.
https://github.com/tmk/tmk_keyboard/tree/master/tmk_core/common/avr

2. and USB stack for the microcontroller should be placed under `protocol`. This is LUFA stack for example.
https://github.com/tmk/tmk_keyboard/tree/master/tmk_core/protocol/lufa

3. then place codes specific to the keyboard in `keyboard`, like this.
https://github.com/tmk/tmk_keyboard/tree/master/keyboard/alps64

szafranski, 2020-05-27:
Thank you!! Last question - where can I find a list of all supported
microcontrollers? I mean for now?

tmk, 2020-05-27:
I use:
atmega32u4
atmega32u2