tmk/tmk_keyboard #653: Docs on matrix definition or some guide-example

andreabenini, 2020-07-04:
Hi,
I've built a custom keyboard from scratch and I'd like to use TMK firmware for it. It's pretty huge (119 Keys!) and after a lot of manual wirings and hardware setup I'm focusing on the firmware part with this project.
My custom project is based on HHKB code (made a copy of that dir) and I've created the KEYMAP() and the matrix following wiki markdown docs and FAQ, now I'm trying to define pin mapping but I don't understand where to start with it.
In `config.h` I've defined:
```C
/* matrix size */
#define MATRIX_ROWS 6
#define MATRIX_COLS 20
```
and now I need to assign Teensy++ 2 pins (with is my MCU) to cols/rows, with another project (QMK) I basically just need to define this:
```C
/* key matrix pins */ // Real PINS on my project
#define MATRIX_ROW_PINS { C2, C3, C4, C5, C6, C7 }
#define MATRIX_COL_PINS { F2, F1, F0, E6, E7, B0, B1, B2, B3, B4, B5, B6, B7, D4, D5, D7, E0, E1, C0, C1 }
```
in the same _config.h_ file. I'm trying to do the same because I'd like to use tmk, how can I do that ? I cannot find a quite recent guide for it. I'm a noob with this project but I've a quite solid C experience so coding it's not a problem.
Can someone point me out on that ?

I can also: push a pull request, publish my entire code if someone is interested, document the process or add info on the wiki where needed. Probably there's something already written for it but I cannot find it

_Thanks in advance
Ben_

tmk, 2020-07-04:
You can't define matrix like that in TMK, you have to write code in `matrix_scan()` to scan matrix.

andreabenini, 2020-07-04:
Oh thanks, I guess I've found something interesting..
`matrix_init()` and `matrix_scan()` are using `KEY_INIT|SELECT|ENABLE..*()` defines. In my case I've simply copied hhkb project because this seems to be the most actively developed above all and everything goes through `hhkb_avr.h` where all these defines are stored. Now my pain will be focused on _KEY_INIT()_, _KEY_SELECT()_ defs in order to try to convert them with my mapping, there's something in the wiki but it's rather complex and I'm still missing the point.
I was grabbing ideas from (https://github.com/tmk/tmk_keyboard/issues/632) but I need to study it a little bit more.

Thanks

closed 2020-07-04 by andreabenini

andreabenini, 2020-07-04:
Ok, This is even better: https://deskthority.net/viewtopic.php?f=7&t=7177&sid=c84a79014281cc2ff9a62c18201ab230
Matt3o docs are scattered over the years and really long to read, it takes a while to order them carefully but this seems to be the best way (even if 24 pages from forum threads are a lot !).

tmk, 2020-07-05:
You can base on alps64 or gh60 instead of hhkb, the project uses simple matrix circuit and usual keyboard switches.