tmk/tmk_keyboard #655: Use TMK for virtual devices

ivanjermakov, 2020-09-16:
Instead of scanning matrix, receive state changes from existing event device.
Run TMK on PC, instead of keyboard controller.
Send reports to virtual devices, for example to linux kernel module [uinput](https://www.kernel.org/doc/html/v4.12/input/uinput.html).
So TMK can be ran for any non-programmable input device.
I am not requesting this as a feature, but I would like to hear some feedback and ideas on how this can be implemented around TMK.

tmk, 2020-09-17:
On host computer you can and want to implement more high-level complex functions than ones on TMK keyboard firmware. 
You will have to refer to Xorg XKB, AHK or Karabiner rather than keyboard firmware?
I don't know much input system on Linux, Windows and Mac, though.

ivanjermakov, 2020-09-18:
For those who are interested I found project that fit well to my requirements: [KMonad](https://github.com/david-janssen/kmonad)

closed 2020-09-18 by ivanjermakov