tmk/tmk_keyboard #663: integrix keyboard layout

sharkka, 2020-12-11:
This is my integrix keyboard layout and scan code
<pre>
/* Integrix Model 500 keyboard
 * ,-------.  ,---,  ,---------------. ,---------------. ,---------------.                 ,---------------.
 * |  76   |  | 1D|  | 05| 06| 08| 0A| | 0C| 0E| 10| 11| | 12| 07| 09| 0B|                 | 15| 16| 17| 62|
 * `-------'  `---'  `---------------' `---------------' `---------------'                 `---------------'
 * ,-------.  ,-----------------------------------------------------------.  ,-----------. ,---------------.
 * | 01| 03|  | 2A| 1E| 1F| 20| 21| 22| 23| 24| 25| 26| 27| 28| 29| 58| 42|  | 02| 04| 0F| | 2D| 2E| 2F| 47|
 * |-------|  |-----------------------------------------------------------|  |------------ |---------------|
 * | 19| 1A|  |  35 | 36| 37| 38| 39| 3A| 3B| 3C| 3D| 3E| 3F| 40| 41| 2B  |  | 14| 18| 1B| | 44| 45| 46|   |
 * |-------|  |-----------------------------------------------------------|  `-----------' |-----------| 7D|
 * | 31| 33|  |  4C  | 4D| 4E| 4F| 50| 51| 52| 53| 54| 55| 56| 57|   59   |                | 5B| 5C| 5D|   |
 * |-------|  |-----------------------------------------------------------|      ,---.     |-----------|---|
 * | 48| 49|  | 63 |+7C| 64| 65| 66| 67| 68| 69| 6A| 6B| 6C| 6D|*6F|  6E  |      | 34|     | 70| 71| 72|   |
 * |-------|  |-----------------------------------------------------------|  .-----------. |-----------| 5A|
 * | 5F| 61|  | 77 | 13| 78 |    |       79         |    |   | 7A | 43| 0D|  | 73| 74| 75| |   5E  | 32|   |
 * `-------'  `-----------------------------------------------------------'  `-----------' `---------------'
 * 
 * ,-------.  ,---,  ,---------------. ,---------------. ,---------------.                 ,---------------.
 * | Help  |  |ESC|  | F1| F2| F3| F4| | F5| F6| F7| F8| | F9|F10|F11|F12|                 |Pau|PrS|ScL|NmL|
 * `-------'  `---'  `---------------' `---------------' `---------------'                 `---------------'
 * ,-------.  ,-----------------------------------------------------------.  ,-----------. ,---------------.
 * |Stp|Agn|  |`|  1|  2|  3|  4|  5|  6|  7|  8|  9|  0|  -|  =|  \|  Del|  |Ins|Hom|PgU| |  =|  /|  *|  -|
 * |-------|  |-----------------------------------------------------------|  |------------ |---------------|
 * |Prp|Und|  |Tab  |  Q|  W|  E|  R|  T|  Y|  U|  I|  O|  P|  [|  ]| Bspc|  |Del|End|PgD| |  7|  8|  9|   |
 * |-------|  |-----------------------------------------------------------|  `-----------' |-----------|  +|
 * |Frt|Cpy|  |Ctrl  |  A|  S|  D|  F|  G|  H|  J|  K|  L|  ;|  '|  Return|                |  4|  5|  6|   |
 * |-------|  |-----------------------------------------------------------|      ,---.     |-----------|---|
 * |Opn|Pst|  |Shft| +\|  Z|  X|  C|  V|  B|  N|  M|  ,|  .|  /|   | Shift|      | Up|     |  1|  2|  3|   |
 * |-------|  |-----------------------------------------------------------|  .-----------. |-----------|Ent|
 * |Fnd|Cut|  |Caps|Alt|Meta|    |      Space       |    |   |Meta|Cmp|Alt|  |Lef|Dow|Rig| |      0|  .|   |
 * `-------'  `-----------------------------------------------------------'  `-----------' `---------------'
 */
</pre>

I try to modify UNIMAP_INTEGRIX macro to make, but failed.

tmk, 2020-12-12:
The scan codes almost looks like Sun compatible with some differences.

Give us info of the 'Integrix Model 500 keyboard' for future reference. Do you have any pics or URL of the keyboard?

TMK Sun-USB converter works with the keyboard somewhat, right? And what about LED indicators?

This is default key mapping of Sun-USB converter. There are some diffrences but it seems all keys of the keyboard should be recogized by the converter and keymap editor. Key layout of editor is not useful for the keyboard though, you can remap the keys with keymap editor somehow.

For example, 'Insert' key on your keyboard registers scan code 0x02 but the scan code is 'Vol Down' on editor.

https://github.com/tmk/tmk_keyboard/blob/8898747b098b02ad60b278da6abbb4bae6029033/converter/sun_usb/unimap_trans.h#L27-L59

If you still want to define your own 'UNIMAP_INTEGRIX' macro, post your code for us to check.

tmk, 2020-12-12:
I found some reated references but no info of 'Integrix Model 500'.

It seems that Integrix TYPE 5 KEYBOARD S5 + 100 is Sun compatible.
https://deskthority.net/viewtopic.php?f=62&t=7700
https://deskthority.net/viewtopic.php?p=171614#p171614

Support for Integrix Type 5 model S5 + 100:
https://github.com/xelalexv/suniversal/issues/4

Pics of Integrix Type 5 model S5 + 100:
https://oc.honet.ch/index.php/s/kSNDRXcckLBmeja#/

> Integrix is a company that makes Sun clones, RAID solutions, etc.  An
S5+100 keyboard is a particular model.  Looks like a Sun keyboard, has a
Sun Interface, yet devices that use it (ie Tektronix X terminals) won't
work with standard Sun keyboards.

http://www.silug.org/lists/silug-discuss/200104/msg00031.html

waveform of 's':
https://oc.honet.ch/index.php/s/kSNDRXcckLBmeja?dir=undefined&%2F%2FIMG_20190402_195012.jpg&openfile=615069

sharkka, 2020-12-12:
This is my integrix keyboard,

And I also have not found any information for this model 500 keyboard,
and sun-usb converter works fine except some keys as refer I sent email to you
I try to remap those keys, now keys(UP,DOWN,LEFT,RIGHT,INT.DEL,HOME,END,PAGEUP,PAGEDOWN) have correct define,works good,but PAUSE/SCROLL/PRINT

tmk, 2020-12-12:
Thanks for the pics. They would be helpful for future reference.
[image]
[image]

Definition of 'UNIMAP_INTEGRIX_MODEL_500' looks good at a glance.
How do you define keymap in unimap.c?

sharkka, 2020-12-12:
Just replace macro UNIMAP_TYPE5 with UNIMAP_INTEGRIX_MODEL_500 in unimap.c will be ok,
I create a folder named integrix_usb, and modify unimap_trans.h and replace macro UNIMAP_TYPE5 unimap.c.

tmk, 2020-12-13:
Then that won't work. You don't need to change code for the keys(PAUSE/SCROLL/PRINT SCREEN DELETE) they uses same scan codes as genuine Sun keyboard.

You better change position of arguments in 'UNIMAP_INTEGRIX_MODEL_500' completely according to its actual key layout like this.

```
#include "unimap_trans.h"

/* Integrix Model 500 keyboard
 * ,-------.  ,----,  ,---------------. ,---------------. ,---------------.                ,---------------.
 * |  76   |  | 1D |  | 05| 06| 08| 0A| | 0C| 0E| 10| 11| | 12| 07| 09| 0B|                | 15| 16| 17| 62|
 * `-------'  `----'  `---------------' `---------------' `---------------'                `---------------'
 * ,-------.  ,-----------------------------------------------------------.  ,-----------. ,---------------.
 * | 01| 03|  | 2A| 1E| 1F| 20| 21| 22| 23| 24| 25| 26| 27| 28| 29| 58| 42|  | 02| 04| 0F| | 2D| 2E| 2F| 47|
 * |-------|  |-----------------------------------------------------------|  |------------ |---------------|
 * | 19| 1A|  |  35 | 36| 37| 38| 39| 3A| 3B| 3C| 3D| 3E| 3F| 40| 41| 2B  |  | 14| 18| 1B| | 44| 45| 46|   |
 * |-------|  |-----------------------------------------------------------|  `-----------' |-----------| 7D|
 * | 31| 33|  |  4C  | 4D| 4E| 4F| 50| 51| 52| 53| 54| 55| 56| 57|   59   |                | 5B| 5C| 5D|   |
 * |-------|  |-----------------------------------------------------------|      ,---.     |-----------|---|
 * | 48| 49|  | 63 |+7C| 64| 65| 66| 67| 68| 69| 6A| 6B| 6C| 6D|  6E  | 6F|      | 34|     | 70| 71| 72|   |
 * |-------|  |-----------------------------------------------------------|  .-----------. |-----------| 5A|
 * | 5F| 61|  | 77 | 13| 78 |    |       79         |    |   | 7A | 43| 0D|  | 73| 74| 75| |   5E  | 32|   |
 * `-------'  `-----------------------------------------------------------'  `-----------' `---------------'
 *
 * ,-------.  ,----,  ,---------------. ,---------------. ,---------------.                ,---------------.
 * | Help  |  |ESC |  | F1| F2| F3| F4| | F5| F6| F7| F8| | F9|F10|F11|F12|                |Pau|PrS|ScL|NmL|
 * `-------'  `----'  `---------------' `---------------' `---------------'                `---------------'
 * ,-------.  ,-----------------------------------------------------------.  ,-----------. ,---------------.
 * |Stp|Agn|  |  `|  1|  2|  3|  4|  5|  6|  7|  8|  9|  0|  -|  =|  \|Del|  |Ins|Hom|PgU| |  =|  /|  *|  -|
 * |-------|  |-----------------------------------------------------------|  |-----------| |---------------|
 * |Prp|Und|  |Tab  |  Q|  W|  E|  R|  T|  Y|  U|  I|  O|  P|  [|  ]| Bspc|  |Del|End|PgD| |  7|  8|  9|   |
 * |-------|  |-----------------------------------------------------------|  `-----------' |-----------|  +|
 * |Frt|Cpy|  |Ctrl  |  A|  S|  D|  F|  G|  H|  J|  K|  L|  ;|  '|  Return|                |  4|  5|  6|   |
 * |-------|  |-----------------------------------------------------------|      ,---.     |-----------|---|
 * |Opn|Pst|  |Shft| +\|  Z|  X|  C|  V|  B|  N|  M|  ,|  .|  /| Shift| LF|      | Up|     |  1|  2|  3|   |
 * |-------|  |-----------------------------------------------------------|  .-----------. |-----------|Ent|
 * |Fnd|Cut|  |Caps|Alt|Meta|    |      Space       |    |   |Meta|Cmp|Alt|  |Lef|Dow|Rig| |      0|  .|   |
 * `-------'  `-----------------------------------------------------------'  `-----------' `---------------'
 */
#define UNIMAP_INTEGRIX_MODEL_500( \
    K76,      K1D,  K05,K06,K08,K0A,   K0C,K0E,K10,K11,   K12,K07,K09,K0B,                K15,K16,K17,K62, \
    K01,K03,  K2A,K1E,K1F,K20,K21,K22,K23,K24,K25,K26,K27,K28,K29,K58,K42,  K02,K04,K0F,  K2D,K2E,K2F,K47, \
    K19,K1A,  K35,K36,K37,K38,K39,K3A,K3B,K3C,K3D,K3E,K3F,K40,K41,    K2B,  K14,K18,K1B,  K44,K45,K46,K7D, \
    K31,K33,  K4C,K4D,K4E,K4F,K50,K51,K52,K53,K54,K55,K56,K57,        K59,                K5B,K5C,K5D,     \
    K48,K49,  K63,K7C,K64,K65,K66,K67,K68,K69,K6A,K6B,K6C,K6D,    K6E,K6F,      K34,      K70,K71,K72,K5A, \
    K5F,K61,  K77,K78,K7A,K4A,        K79,            K7B,K1C,K13,K43,K0D,  K73,K74,K75,  K5E,    K32      \
) UNIMAP ( \
            K01,K03,K19,K1A,K31,K33,K48,K49,K5F,K61,K76,K0F,                                     \
    K1D,    K05,K06,K08,K0A,K0C,K0E,K10,K11,K12,K07,K09,K0B,      K16,K17,K15,      K02,K04,K2D, \
    K2A,K1E,K1F,K20,K21,K22,K23,K24,K25,K26,K27,K28,K29,NO, K2B,  NO, K34,NO,   K62,K2E,K2F,K47, \
    K35,K36,K37,K38,K39,K3A,K3B,K3C,K3D,K3E,K3F,K40,K41,    K58,  K42,K4A,K7B,  K44,K45,K46,K7D, \
    K77,K4D,K4E,K4F,K50,K51,K52,K53,K54,K55,K56,K57,    NO, K59,                K5B,K5C,K5D,NO,  \
    K63,K7C,K64,K65,K66,K67,K68,K69,K6A,K6B,K6C,K6D,    K6F,K6E,      K14,      K70,K71,K72,K5A, \
    K4C,K78,K13,K73,        K79,        K74,K75,K0D,K7A,K43,NO,   K18,K1B,K1C,  K5E,    K32,NO   \
)

#ifdef KEYMAP_SECTION_ENABLE
const action_t actionmaps[][UNIMAP_ROWS][UNIMAP_COLS] __attribute__ ((section (".keymap.keymaps"))) = {
#else
const action_t actionmaps[][UNIMAP_ROWS][UNIMAP_COLS] PROGMEM = {
#endif
    UNIMAP_INTEGRIX_MODEL_500(
    F23,        ESC, F1,  F2,  F3,       F4,  F5,  F6,  F7,  F8,       F9,  F10, F11, F12,                    PAUS,PSCR,SLCK,NLCK,
    F13, F14,   GRV, 1,   2,   3,   4,   5,   6,   7,   8,   9,   0,   MINS,EQL, BSLS,DEL,   INS, HOME,PGUP,  PEQL,PSLS,PAST,PMNS,
    F15, F16,   TAB, Q,   W,   E,   R,   T,   Y,   U,   I,   O,   P,   LBRC,RBRC,     BSPC,  DEL, END, PGDN,  P7,  P8,  P9,  PPLS,
[... cut ...] P5,  P6,
    F19, F20,   LSFT,NUBS,Z,   X,   C,   V,   B,   N,   M,   COMM,DOT, SLSH,     RSFT,RO,         UP,         P1,  P2,  P3,  PENT,
    F21, F22,   CAPS,LALT,LGUI,MHEN,          SPC,                HENK,KANA,RGUI,APP, RALT,  LEFT,DOWN,RGHT,  P0,       PDOT
    ),
};
```

Just copy this into file named 'unimap_integrix.c' in sun_usb directory. And build it with commands below.
```
    $ make -f Makefile.atmega32u4 KEYMAP=integrix clean
    $ make -f Makefile.atmega32u4 KEYMAP=integrix
```

sharkka, 2020-12-13:
OK,thanks very much,I will rebuild it as your description tonight.

sharkka, 2020-12-13:
emmm, I rebuild it with unimap_integrix.c, PAUSE/PrintScreen/ScrollLock works ok,but UP/DOWN/LEFT/DOWN and Ins/del/home/end/pageup/pagedown won't work
so I do like this,
cp -r sun_usb integrix_usb
cd integrix_usb
and modify unimap_trans.h with defination of UNIMAP_INTEGRIX_MODEL_500

<pre>
#define UNIMAP_INTEGRIX_MODEL_500( \
    K76,      K60,  K05,K06,K08,K0A,   K0C,K0E,K10,K11,   K12,K07,K09,K0B,  K16,K17,K15,  K2C,K42,K2D,  K30, \
    K01,K03,  K1D,K1E,K1F,K20,K21,K22,K23,K24,K25,K26,K27,K28,K29,K58,K2A,  K02,K04,K0F,  K62,K2E,K2F,K47, \
    K19,K1A,  K35,K36,K37,K38,K39,K3A,K3B,K3C,K3D,K3E,K3F,K40,K41,    K2B,  K14,K18,K1B,  K44,K45,K46,K7D, \
    K31,K33,  K77,K4D,K4E,K4F,K50,K51,K52,K53,K54,K55,K56,K57,        K59,                K5B,K5C,K5D,     \
    K48,K49,  K63,K7C,K64,K65,K66,K67,K68,K69,K6A,K6B,K6C,K6D,    K6F,K6E,      K34,      K70,K71,K72,K5A, \
    K5F,K61,  K4C,K78,K7A,K4A,        K79,            K7B,K1C,K13,K43,K0D,  K73,K74,K75,  K5E,    K32      \
) UNIMAP ( \
              K01,K03,K19,K1A,K31,K33,K48,K49,K5F,K61,K76,K0F,                                     \
    K1D,      K05,K06,K08,K0A,K0C,K0E,K10,K11,K12,K07,K09,K0B,      K16,K17,K15,      K02,K04,K2D, \
    K2A,K1E,  K1F,K20,K21,K22,K23,K24,K25,K26,K27,K28,K29,NO, K2B,  K2C,K34,K60,  K62,K2E,K2F,K47, \
    K35,K36,  K37,K38,K39,K3A,K3B,K3C,K3D,K3E,K3F,K40,K41,    K58,  K42,K4A,K7B,  K44,K45,K46,K7D, \
    K77,K4D,  K4E,K4F,K50,K51,K52,K53,K54,K55,K56,K57,    NO, K59,                K5B,K5C,K5D,NO,  \
    K63,K7C,  K64,K65,K66,K67,K68,K69,K6A,K6B,K6C,K6D,    K6F,K6E,      K14,      K70,K71,K72,K5A, \
    K4C,K78,  K13,K73,        K79,        K74,K75,K0D,K7A,K43,K30,  K18,K1B,K1C,  K5E,    K32,NO   \
)
</pre>

and edit Makefile replace sun_usb with integrix_usb
make
make dfu

So, now, Pause/PrintScreen/ScrollLock works good, just 'delete' map to 'Volume Down', but it is ok, not all keys have defination, lineFeed also have no defination.

On the whole，current firmware of converter is enough for me, of course, the integrix keyboard is not general layout.

tmk, 2020-12-13:
Great.
It would be helpful for future reference if you can place your code in your repository.

closed 2020-12-13 by tmk

sharkka, 2020-12-13:
OK, 
[https://github.com/sharkka/integrix_keyboard.git](url)

If necessary you can modify code as your own style, maybe my code has no general applicability.