tmk/tmk_keyboard #666: output periodically stops while holding down a key with PC98 converter

weebicneet, 2021-01-15:
I've been trying to game on my FC-9801-KB6, but I keep running into this problem, where when holding keys down, the keyboard stops outputting whatever key is held down for a very small amount of time. Every 0.2 seconds or so. It makes my character stutter when moving forwards for example. This doesn't happen with modifier keys like shift or alt.

tmk, 2021-01-16:
I guess the keyboard sends scan code repeatedly for the key for some reason.

The converter send a command for keyboard to prevent key repeating only once at startup. If the command works the keyboard should not repeat key.

Try these steps and post debug outputs on 'hid_listen'.
1. Connect keyboard to converter first.
2. Then, Plug the converter into compter.

weebicneet, 2021-01-16:
Output holding down a:
```
C:\Users\yuyu\Desktop>hid_listen.exe
Waiting for device:
Listening:
9C 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D
 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D 1D 9D
```

tmk, 2021-01-16:
The keyboard is repeating the key with make and break scan code. This behaviour is not compatible to mordern USB keyboard.

Some PC98 keyboards can stop the repeating behaviour by command but others doesn't support the command unfortunately.

Apply this patch and check debug outputs again to see if the keyboard supports the command.
```
diff --git a/converter/pc98_usb/matrix.c b/converter/pc98_usb/matrix.c
index b44d3ef5..14ddd85d 100644
--- a/converter/pc98_usb/matrix.c
+++ b/converter/pc98_usb/matrix.c
@@ -50,6 +50,7 @@ static uint8_t matrix[MATRIX_ROWS];
 
 static void pc98_send(uint8_t data)
 {
+    xprintf("s%02X ", data);
     PC98_RDY_PORT |= (1<<PC98_RDY_BIT);
     _delay_ms(1);
     serial_send(data);
@@ -62,6 +63,7 @@ static int16_t pc98_wait_response(void)
     int16_t code = -1;
     uint8_t timeout = 255;
     while (timeout-- && (code = serial_recv2()) == -1) _delay_ms(1);
+    xprintf("r%04X ", code);
     return code;
 }
 
@@ -143,7 +145,7 @@ uint8_t matrix_scan(void)
         return 0;
     }
 
-    dprintf("%02X ", code);
+    xprintf("r%02X ", code);
 
     if (code&0x80) {
         // break code
```

If you can't build firmware yourself I'll do for you. Let me know microcontroller model name of your converter?

tmk, 2021-01-16:
Also, Does Capslock LED work as expected on your keyboard?

weebicneet, 2021-01-16:
Please do build it for me. The microcontroller model name is ATmega32U2.

Actually, capslock doesn't work properly. When I press the capslock key, the LED lights up, however, the whole keyboard becomes unresponsive until I replug it. And then when it does start working again, it's stuck in capslock for some reason. The same thing doesn't happen with the kana key, which is also a locking key.

tmk, 2021-01-16:
Hmm, the converter may fail to send commands for some reason.

Can you try this firmware to see debug outputs?
https://gist.github.com/tmk/cb55ff53cb9a52308d1992d266804c25

You will see outputs like below.
```
TMK:93b260a8/LUFA

s9F rFFFF old-type
s9C r00FA s70 r00FA 
USB configured.

Keyboard start.
s9D r00FA s71 r00FA
```

weebicneet, 2021-01-16:
```
Waiting for device:.......
Listening:

TMK:93b260a8/LUFA

s9F rFFFF old-type
s9C r00FA s70 r00FA
USB configured.

Keyboard start.
s9D r00FA s74 r00FA
```

tmk, 2021-01-16:
It seems that the keyboard accept the 'repeat setting' command(9C, 70) with ACK(FA) successfully.
If the key still repeats I think the keyobard doesn't supports 'stop-repeat' feature unfortunately. Some of old keyobards doesn't support it according to PC98 documents. (My 'Dboard sono1' and PC-9801V keyboard don't support too, while PC-9801-98-S02 does.)

But pressing CapsLock should not be stuck, what happen on hid_listen when pressing CapsLock?

weebicneet, 2021-01-16:
These are the results of opening hid_listen, plugging in the converter, then plugging in the keyboard, and pressing caps twice. The keyboard doesn't work when plugging in the keyboard before the converter.
```
Waiting for device:
Listening:
r71 rF1 s9D r00FA s70 r00FA
```

tmk, 2021-01-16:
The converter doesn't support hot-plug basically at this point, with some keyobards you will have to connect keyboard into converter first. Some keyboards need to be intinitialized and the converter does initialization procedure only on startup. (Dboard sono1 works with hot-plug but PC-9801V keyboard doesn't.)

Does the CapsLock key work when you connect keyboard first?

I'll look into diffrence between my two keyboards to improve support for hot-plug.

weebicneet, 2021-01-16:
The keyboard doesn't work at all if it's plugged into the converter before the converter is plugged into the computer.

tmk, 2021-01-16:
Consulting with PC98 documents I changed code of sending command slightly. This may change the behaviour.

Try updated firmware and can you post log with press capslock?
https://gist.github.com/tmk/cb55ff53cb9a52308d1992d266804c25

After some research I found that hot-pluging keyboard is impossible or very difficult due to limitation of PC98 keyboard hardware design. So you have to connect keyboard with converter ant then plug them into USB port. Otherwirse PC98 keyboard may not be initialized correctly.

weebicneet, 2021-01-17:
Amazing! The keyboard now functions as it should. Both the key repeat and capslock issue have been fixed. Thank you, hasu-san!

Log with pressing caps four times:
```
Waiting for device:..........
Listening:

TMK:70e85185/LUFA

s9F rFFFF old-type
s9C r00FA s70 r00FA
USB configured.

Keyboard start.
r71 s9D r00FA s74 r00FA rF1 s9D r00FA s70 r00FA r71 s9D r00FA s74 r00FA rF1 s9D
r00FA s70 r00FA
```

tmk, 2021-01-17:
Great. Thank you for your report.
I just updated codes in repositroy for this problem, you can download new firmware from Keymap Editor too.

closed 2021-01-17 by tmk