tmk/tmk_keyboard #667: English Correction PR

oliversandli, 2021-01-21:
First of all, I want to say what an awesome project this is! The documentation is extremely thorough and there is an impressive set of features.

Anyway, would you be willing to accept a PR with English corrections in the README's? It would add a polish to how the project looks. The writing is technically acceptable English, but there are a couple phrases and sentence structures that are not idiomatic English. Note that I haven't started anything yet, just asking if you would accept such a PR if I were to do this.

tmk, 2021-01-24:
Thank you for your suggestion.
Yes, I'm happy to accept corrections. 

Which README? I tend not to update documents so often, due to my laziness.
One in top directory is too old at least and it should be updated by me before applying corrections from you.

oliversandli, 2021-01-24:
It's nothing serious. Mostly just missing articles (a, an, the), punctuation, and the occasional run-on sentence. I can start working on the PR before or after you rework the main README; whatever works best for you.

oliversandli, 2021-01-24:
I didn't take a look at _all_ the README's, but most that I read had possible improvements.

tmk, 2021-01-25:
OK. Could you make PR?

I can't get clear idea on how README should be updated at this time, I'll do it later.
Thanks

oliversandli, 2021-01-25:
Ok, thanks! I'll start on a PR. It will take at week at most.

oliversandli, 2021-01-26:
Hehe, after starting work it looks like it might take more like 2 weeks. Sorry for the initial incorrect estimate!

tmk, 2021-01-26:
No rush at all. Take time!

I think it would be nice to start with small PR, rather than one big PR.
This will save our time in case that we have different preference for somehting.

closed 2023-12-20 by oliverdelancey