tmk/tmk_keyboard #672: Swapped keys and power key hint on ADB-USB with AEK-II

svbnet, 2021-02-07:
In [unimap_common.h](https://github.com/tmk/tmk_keyboard/blob/master/converter/adb_usb/unimap_common.h) of the ADB - USB converter, the translation table has the GUI and ALT keys switched. If this was intended, I think it would've been better to define it as a mapping in `unimap_plain.c` as it had me scratching my head for quite a while.

Additionally, and admittedly due to me not reading the wiki, I didn't realise for windows the `KC_PWR` code must be used. It would be nice if perhaps there was a comment or something explaining this. I could submit a pull request if you like.

Regardless, I am thankful for this project - keep up the good work!

tmk, 2021-02-08:
As for LGUI and LALT they were swapped unintentionally last month and will be fixed on next update, probably in a month.

As for KC_PWR, can you confirm patch below works actually on Windows? If so I would fix this as well.
I can't remember that `ACTION_MODS_TAP_KEY` accept KC_PWR instead of KC_POWER, and can't test this myself now.

```
diff --git a/converter/adb_usb/unimap_plain.c b/converter/adb_usb/unimap_plain.c
index 4aa59628..1da5f4df 100644
--- a/converter/adb_usb/unimap_plain.c
+++ b/converter/adb_usb/unimap_plain.c
@@ -19,7 +19,7 @@ along with this program.  If not, see <http://www.gnu.org/licenses/>.
 
 #define AC_FN0  ACTION_LAYER_TAP_KEY(1, KC_GRV)
 #define AC_FN1  ACTION_LAYER_TAP_KEY(2, KC_BSLS)
-#define AC_PKEY ACTION_MODS_TAP_KEY(MOD_NONE, KC_POWER)
+#define AC_PKEY ACTION_MODS_TAP_KEY(MOD_NONE, KC_PWR)
 
 #ifdef KEYMAP_SECTION_ENABLE
 const action_t actionmaps[][UNIMAP_ROWS][UNIMAP_COLS] __attribute__ ((section (".keymap.keymaps"))) = {
```

svbnet, 2021-02-08:
I couldn't get the above to work. Though the ADB scancode is logged in `hid_listen` it doesn't appear to be picked up by Windows. Only `#define AC_PKEY ACTION_KEY(KC_PWR)` works, although with the caveat being the magic key no longer works properly by default.

tmk, 2021-02-08:
Ah, ok. This is probably reason why it still uses `KC_POWER`.
You can't use `ACTION_MODS_TAP_KEY` with  `KC_PWR` unfortunately, due to current implementation limitation perhaps.

Linux and MacOS can recoginze both codes while Windows knows only `KC_PWR`.

tmk, 2021-02-08:
Power key problem was fixed at 06d354c06b473864ddc97ca347a6d4ffd00bbda3
and Swap problem with ea4baf1cb6ed34a88c0d478f71d9f17ae069c3c6 as well.

tmk, 2021-02-08:
Let me know if you still have the problem. Thanks

closed 2021-02-08 by tmk

svbnet, 2021-02-08:
Yes, it works perfect now with `KC_PWR`. Thank you!