tmk/tmk_keyboard #674: ADB->USB Dead key on ISO Apple Keyboard II

M3kH, 2021-02-09:
First of all thanks for this amazing drivers I'm currently happy using my AKII!

The only issue that I found is that one key is not working, and although I try to map it seems that OS doesn't receive a recognise key press.

This is the keyboard:
[image]

The death key is the one next to the enter which should be `\` or `NUBS  || BLSL`, I'm currently using the binding from `old_keymap/keymap_iso.c` since was the only one that let my keyboard work on **Teensy 2.0**.

If I initialise `hid_listen` and press the dead key the return output is `2AFF AAFF`, but even if rebind the key has no use.

Any suggestion how an I fix this?
With a bit of guidance I will also be happy to share my keymap for this layout if useful to someone else.

tmk, 2021-02-09:
How did you built with the old_keymap file? I beleive that file is not compatible anymore.

Check the first post for notes about ISO keyboard especially.
https://geekhack.org/index.php?topic=14290.0

Test the key using  prebuilt firmware, then copy and paste output of hid_listen.

M3kH, 2021-02-10:
Thanks @tmk I later use the editor and allow me to full set the keyboard.
Anyway to dig a bit deeper here are my compilation steps from the `adb_usb`:

```
mkdir -p obj_adb_usb_rev1
cp old_keymap/keymap_common.h obj_adb_usb_rev1
cp old_keymap/keymap_iso.c obj_adb_usb_rev1
make -f Makefile.rev1 KEYMAP=iso
```

I will edit later what the hid_listen return on prebuild firmware.

M3kH, 2021-02-10:
@tmk I run the `hid_listen` this what does return:
```
2AFF \AAFF
```

Now what I notice is that the `KGUI` and `LAlt` are swapped which indeed is minor thing, maybe I shouldn't have use the `old_keymap` begin with.

M3kH, 2021-02-10:
I close this issue because mainly relates to deprecate keymap. Thanks for the support.

closed 2021-02-10 by M3kH