tmk/tmk_keyboard #677: Kensington Turbo 5 Mapping

BosungLim, 2021-03-05:
I am using Turbo 5 and wondered if I can change the key mapping. It will be great if it can make the scroll function.

tmk, 2021-03-06:
You can do with tools on computer like xmodmap and xbindkeys, though, it is still nice idea. PR would be appreciated.

For someone who want to try scroll function,
ps2_mouse has codes for the function:
https://github.com/tmk/tmk_keyboard/blob/6360eb308c4b81d5effde77e9b0f357dd9673696/tmk_core/protocol/ps2_mouse.c#L124-L160

You can apply this code into `adb_mouse_task()` in `matrix.c` of adb_usb probably.

BosungLim, 2021-03-06:
> You can do with tools on computer like xmodmap and xbindkeys, though, it is still nice idea. PR would be appreciated.
>
> For someone who want to try scroll function,
> ps2_mouse has codes for the [...]

BosungLim, 2021-03-06:
sorry, I had closed the issue by a mistake and reopened it.

anyway, I will try and leave the comments. thank you for your helps.

closed 2021-05-31 by tmk