tmk/tmk_keyboard #682: Add support for extended Microsoft scancodes keys to keymap

Tabby, 2021-04-16:
Hi :)

I've been using one of your PS/2 to USB converters for a few years now and I love it, but I want to try and get all the keys on my keyboard working if possible.

I have a Microsoft Natural Multimedia keyboard which has a whole bunch of extra buttons (Scancodes are listed here: https://www.win.tue.nl/~aeb/linux/kbd/scancodes-6.html#ss6.4) but I can't work out how to add them to the keymap. The online editor doesn't seem to have any option for adding additional keys and I can't figure out how to add them in the code either :S

Is it possible to use non-standard PS/2 scancodes? If so, are you able to help me work out how to add support?

Tabby, 2021-04-16:
So as far as I can tell, it looks like I just need to add the extra scancodes to matrix.c as listed in the FAQ entry which links to https://geekhack.org/index.php?topic=72052.msg2630974#msg2630974

Looking at that file though, my understanding is that using hid_listen, I should see the error `unexpected scan code at E0: %02X\n` being printed when I press an unsupported key but I don't see any log messages, even after ensuring that debug mode is on by turning it off and then on again with the magic keys

Is there something I'm missing to get those log messages? Should I just add the extra codes to matrix_scan() anyway and just hope they work without any useful debug output? :/

Tabby, 2021-04-16:
Well I'm utterly confused. When I force the matrix and keyboard debugging to be on in `matrix_init` (the magic keys to turn those on and off don't seem to be working) it looks like the matrix is being updated just fine, so I'm back to being unable to work out how to change the keymap to actually make use of the additional keys

tmk, 2021-04-17:
which firmware are you using?
You can see version info with pressing both shifts and v.

And how do you build firmware, which makefile are you using?

If you are using ps2_usb converter,
scan codes E0 xx are handled here. I think the scan codes are recognized and you don't change this code here.  But you should see the error message.
https://github.com/tmk/tmk_keyboard/blob/master/converter/ps2_usb/matrix.c#L221-L242

tmk, 2021-04-17:
And note that scan codes listend in this page are not raw codes from keyboards(Code Set 2). The scan codes are ones translated to Code Set 1 codes on PC.
For example, keyboard sneds and the converter recieves actually `e0 32` for  'Volume +' instead of `e0 30`. 

> (Scancodes are listed here: https://www.win.tue.nl/~aeb/linux/kbd/scancodes-6.html#ss6.4)

Tabby, 2021-04-17:
I'm currently using this firmware, and using `make -f Makefile.rev2 KEYMAP=plain` to build it:

```
        - Version -
DESC: convert PS/2 keyboard to USB
VID: 0xFEED(t.m.k.) PID: 0x6512(PS/2 keyboard converter) VER: 0x0001
BUILD: c2692ae (16:07:09 Oct  4 2016)
OPTIONS: LUFA EXTRAKEY CONSOLE COMMAND NKRO KEYMAP_SECTION 4096
GCC: 4.9.2 AVR-LIBC: 1.8.0svn AVR_ARCH: avr35
```

As you say, the scan codes are recognised, but I can't work out how to get the keymap to use them. I've tried using the keymap editor to replace F13 to F24 with the special keys that it knows about but it doesn't seem to work

So I guess in summary, I seem to have two problems:
* I need to add constants for some of the extra keys, like 'Help', 'Undo', 'Redo', and so on, and I can't work out where
* I can't figure out what changes to make to unimap_plain.c to add the extra keys to the keymap so that they work

tmk, 2021-04-17:
OK. Then, you should edit unimap_trans.h to add support the keys.

For example to add support for Help(e0 3b) as F13, you have to know real scan code of using the pdf document above or this translation table. In this case it is 'e0 05'.
https://www.win.tue.nl/~aeb/linux/kbd/scancodes-10.html#ss10.3

So the Help key is registered at  index 0x85(0x80|0x05) in converter matrix here.
https://github.com/tmk/tmk_keyboard/blob/master/converter/ps2_usb/matrix.c#L235

And edit unimap_trans.h to map the key to F13, just place `UNIMAP_F13` at the location of 0x85 in `unimap_trans` array.

At this point  the Help key is recognized as F13 with default keymap and you can remap the key in F13 position on Keymap Editor(or by editing unimap_plain.c).

Post your change if you don't get it working even after fllowing this.

Tabby, 2021-04-17:
Thanks so much, that's incredibly helpful! I've got most of the keys doing something now (albeit just set to letters at the moment for easy testing) and the only ones that aren't are the ones I haven't been able to find the scan codes for so that's just a case of debugging :)

I have one more question: the `UNIMAP_*` constants only allow me to set the keys to the basic unimap keys. If I want to bind, for example, the play/pause key to the system play/pause action, how would I do that?

I've tried putting `AC_MEDIA_PLAY_PAUSE` into the `unimap_trans` array in the right place, but it won't build because of loss of precision

tmk, 2021-04-17:
Unimap uses universal 128-key keyboard layout like below and `unimap_trans` maps scan code into key position in the layout with `UNIMAP_` constants, which are not actual action.
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/unimap.h#L18-L48

As I said before,
You can map actual action in `unimap_plain.c` which is user keymap file and you can edit. You can use action codes defined here.
https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/common/actionmap.h

Tabby, 2021-04-17:
If there's no way to use more than 128 keys with this adapter, then no worries

I'm still very grateful for your help in getting these keys mapped to something I can at least use. Thanks again

closed 2021-04-17 by Tabby