tmk/tmk_keyboard #683: Support for multiple devices?

evprkr, 2021-05-19:
Built the adapter using an Adafruit ItsyBitsy, works great so far. I've got it wired up with a 4.7k Ohm pull-up resistor as recommended.

However, when I plug the adapter into the computer, my M0116 keyboard into the adapter, and then my Kensington Turbo Mouse into the keyboard's ADB port, the keyboard continues to work but the trackball no longer works. I also tried the other way around (Adapter > Mouse > Keyboard) and the mouse stops working once the keyboard is plugged in.

Is this a limitation of the adapter, or something in software causing this? Would a 10k Ohm resistor enable daisy-chaining multiple devices?

tmk, 2021-05-19:
The ADB conveter should work with multiple devices. Use hid_listen to see debug outputs.

I usually use 1k Ohm but 4.7k won't be problem.

evprkr, 2021-05-27:
> The ADB conveter should work with multiple devices. Use hid_listen to see debug outputs.
> 
> I usually use 1k Ohm but 4.7k won't be problem.

Can't use the hid_listen.mac file because it's 32-bit only and I'm on Catalina on one machine and Big Sur on another. Used the HID console in QMK Toolbox, I get normal readouts for mouse movement on the trackball, then I plug in another ADB device to the second port on the trackball (in this case an Apple Desktop Bus Mouse), and I get this output in the console:

```
RRM:addr3 reg3: 6801
M:addr15 reg3: 6702
Classic 200cpi
M:addr3 reg3: 6232
TM5: ext
```
Then, the ADB Mouse works but the trackball no longer works. Unplugging the ADB Mouse does not restore the trackball's functionality, it has to be unplugged and re-plugged to work again. The same thing happens if I use an ADB keyboard instead.

tmk, 2021-05-28:
what  version of firmware are you using? You can see its version by pressing Power and v.
https://github.com/tmk/tmk_keyboard/wiki#debug-console

I remember that 1) two keyboards worked, 2) one keyboard and one pointing device worked at same time.
But not sure that I confirmed two pointing devices worked before.

I don't have ADB converter in hand now to test this, but multiple pointing devices seems to be supported at a glance of source code.

tmk, 2021-05-28:
Device hotplug is kind of supported but you may get better result with coldplug.
That is, connect devices before plug the converter into USB port.

evprkr, 2021-05-28:
I'm using rev 1 from this link: http://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?adb_usb_rev1

Sorry, not at my work computer right now, which is where I have the adapter/devices. I was trying two mice today, because that's what I happened to have near me, but in my original post I noted I was trying with the Kensington Turbo Mouse and Apple M0116 keyboard and had the same results.

I also tried hotplug and coldplug, and different combinations of cable (Adapter > Keyboard > Mouse & Adapter > Mouse > Keyboard) and neither worked.

tmk, 2022-03-30:
See 'pull-up resistor' seciton in this post also.
https://geekhack.org/index.php?topic=14290.msg277407#msg277407

closed 2022-03-30 by tmk