tmk/tmk_keyboard #685: 124-key 5576 support in ibmpc_usb

RaoulDuke-Esq, 2021-05-28:
Hi Hasu,

Could you please add support for the 124-key 5576 to your ibmpc_usb converter?  I want to be able to map all keys on the keyboard but currently the keys on either side of the space bar also register as spaces and I don't see how I would change this using an existing keymap or your web  keymap editor.

This keyboard registers as AB92 in the toolbox.

tmk, 2021-05-29:
Yes, I'm happy to do that.
Let me know model number of your keyboard model and post debug outputs at startup.

And what are the keys you are interested exactly? Posting pic or ascii art to explain them would be useful. Did you see scan codes on debug outputs when you press those keys

RaoulDuke-Esq, 2021-05-29:
Awesome!  Thank you!!

Here's what I get when I flash the default 122-key Terminal keymap onto my converter and plug in the keyboard.

`*** HID console connected: t.m.k. IBM PC keyboard converter (FEED:1BEE:0001)
  > 
    
    TMK:e473e8ce/LUFA
    
    
    USB configured.
    
    Keyboard start.
    I220 wFF rFA A1225 
    PRT:10 ISR:FE90 rAA W1694 wF2 rFA rAB r92 R2199 
    ID:AB92(AT) S2200 wED rFA w00 rFA L2205 `

Here are the console output, observed events, and key codes from those keys:
[image]

Here is the model sticker on the keyboard:
[image]

This is the overall layout:
[image]

I really appreciate you doing this.  I use TMK on my 4704 Pingmaster and my M122 and am looking forward to using it for my 5576 too.  I usually use your Keymap Editor site to make layouts but am comfortable editing the files to change Keymaps and then recompiling as well.

tmk, 2021-05-29:
Thanks for the info. Great.

The keyboard(5576-001) and other 5576s seems to start up with Scancode Set 2 and doesn't make  Japanese specific keys enabled by default. And 5576-001 doesn't have lock indicators though, it accepts 'set indicator' command(wED).

Reportedly 5576 series supports other Scancode Sets and 3(Terminal) and 82h(5576 specific) out of them appear to be promising. http://radioc.web.fc2.com/column/ps55/ps55kbd_scancode.htm

Try this firmware which makes keyboard speakk on Terminal scan codes.
https://gist.githubusercontent.com/tmk/cc23299ff1e2d2a891db593a9d4d1c82/raw/82b9596fb48740cc7afabe1b897d92dbb54deb5f/ibmpc_usb_atmega32u4.hex

RaoulDuke-Esq, 2021-05-29:
I expected this keyboard not to have lock lights but I was surprised when I took it apart for cleaning that it has the LEDs.  They blink when powering up the keyboard and the caps lock LED turns on when caps locked is pressed.  It looks like there's just a removable plastic blanking panel over the cutouts for the lights that could be replaced with one that has proper legends.

I think you're on to something here.  I loaded up the firmware hex you sent- here's what I get:

`*** HID console connected: t.m.k. IBM PC keyboard converter (FEED:1BEE:0001)
  > 
    
    TMK:693454c8/LUFA
    
    
    USB configured.
    
    Keyboard start.
    I389 wFF rFA A1394 
    PRT:10 ISR:FE90 rAA W1863 wF2 rFA rAB r92 R2369 wF0 rFA w03 rFA 
    ID:AB92(TERMINAL) S2375 wF8 rFA wED rFA w00 rFA L2383 `

And here are the outputs from console.  We're getting different codes from those keys now!
[image]

tmk, 2021-05-29:
haha, hidden LEDs are hilarious :)

Great. Code Set 3(Terminal) makes the japanese specific keys usable.
Two keys with scan code 85 and 86 are conflict with TMK default mapping unfortunately and we will have to tune the mapping for some keys on 5576 keyboards.

Can you check and correct scan codes of 5576-001 in following table? so that I can come up with new mapping without conflict.

```
IBM 5576-001
             ,---------------.  ,---------------.  ,---------------.
             | 08| 10| 18| 20|  | 28| 30| 38| 40|  | 48| 50| 57| 5F|
             |----------------  |----------------  |---------------|
             | 07| 0F| 17| 1F|  | 27| 2F| 37| 3F|  | 47| 4F| 56| 5E|
             `---------------'  `---------------'  `---------------'
,-------. ,-----------------------------------------------------------. ,-----------. ,---------------.
| 05| 06| | 0E| 16| 1E| 26| 25| 2E| 36| 3D| 3E| 46| 45| 4E| 55| 5D| 66| | 67| 6E| 6F| | 76| 77| 7E| 84|
|-------| |-----------------------------------------------------------| |-----------| |---------------|
| 04| 0C| | 0D  | 15| 1D| 24| 2D| 2C| 35| 3C| 43| 44| 4D| 54| 5B|     | | 64| 65| 6D| | 6C| 75| 7D| 7C|
|-------| |------------------------------------------------------`    | `-----------' |---------------|
| 03| 0B| | 14   | 1C| 1B| 23| 2B| 34| 33| 3B| 42| 4B| 4C| 52| 53| 5A |     | 63|     | 6B| 73| 74|   |
|-------| |-----------------------------------------------------------| ,-----------. |-----------|   |
| 83| 0A| | 12     | 1A| 22| 21| 2A| 32| 31| 3A| 41| 49| 4A| 51|  59  | | 61| 62| 6A| | 69| 72| 7A| 79|
|-------| |-----------------------------------------------------------| `-----------' |-----------|   |
| 01| 09| | 11  |  | 19  |  85   |  29   |  86   | 39  |        | 58  |     | 60|     | 70    | 71|   |
`-------' `-----'  `-----------------------------------'        `-----'     `---'     `---------------'
```

RaoulDuke-Esq, 2021-05-29:
Absolutely!

`IBM 5576-001
             ,---------------.  ,---------------.  ,---------------.
             | 08| 10| 18| 20|  | 28| 30| 38| 40|  | 48| 50| 57| 5F|
             |----------------  |----------------  |---------------|
             | 07| 0F| 17| 1F|  | 27| 2F| 37| 3F|  | 47| 4F| 56| 5E|
             `---------------'  `---------------'  `---------------'
,-------. ,-----------------------------------------------------------. ,-----------. ,---------------.
| 05| 06| | 0E| 16| 1E| 26| 25| 2E| 36| 3D| 3E| 46| 45| 4E| 55| 13| 66| | 67| 6E| 6F| | 7E| 77| 76| 84|
|-------| |-----------------------------------------------------------| |-----------| |---------------|
| 04| 0C| | 0D  | 15| 1D| 24| 2D| 2C| 35| 3C| 43| 44| 4D| 54| 5B|     | | 64| 65| 6D| | 6C| 75| 7D| 7C|
|-------| |------------------------------------------------------`    | `-----------' |---------------|
| 03| 0B| | 14   | 1C| 1B| 23| 2B| 34| 33| 3B| 42| 4B| 4C| 52| 53| 5A |     | 63|     | 6B| 73| 74|   |
|-------| |-----------------------------------------------------------| ,-----------. |-----------|   |
| 83| 0A| | 12     | 1A| 22| 21| 2A| 32| 31| 3A| 41| 49| 4A| 5C|  59  | | 61| 62| 6A| | 69| 72| 7A| 79|
|-------| |-----------------------------------------------------------| `-----------' |-----------|   |
| 01| 09| | 11  |  | 19  |  85   |  29   |  86   | 39  |        | 58  |     | 60|     | 70    | 71|   |
`-------' `-----'  `-----------------------------------'        `-----'     `---'     `---------------'
`

Summary of Changes:
5D -> 13
76 -> 7E
7E -> 76
51 -> 5C

tmk, 2021-05-29:
Thanks for the test.  I'll try to tune default mapping avoiding conflict with IBM 122-key terminal later.
  

I just built another firmware for Code Set 82h support, this supports 5576 specific keys better perhaps.

Could you try this firmware to check scan codes of Codes Set 82h in table below?
Especially left function key part and around ￥, @, [, \  and 'Dup' key.

https://gist.githubusercontent.com/tmk/57e6a1804e2817f3980e0d0aea617055/raw/a4f00033dd227510d095d38a184d361d98c49e31/ibmpc_usb_cs82h.hex

```
5576-001 Code Set 82h:
             ,---------------.  ,---------------.  ,---------------.
             | 08| 10| 18| 20|  | 28| 30| 38| 40|  | 48| 50| 57| 5F|
             |----------------  |----------------  |---------------|
             | 05| 06| 04| 0C|  | 03| 0B| 83| 0A|  | 01| 09| 78| 07|
             `---------------'  `---------------'  `---------------'
,-------. ,-----------------------------------------------------------. ,-----------. ,---------------.
|   |   | | 0E| 16| 1E| 26| 25| 2E| 36| 3D| 3E| 46| 45| 4E| 55| 5D| 66| |*70|*6C|*7D| | 7C|*4A|*41| 7B|
|-------| |-----------------------------------------------------------| |-----------| |---------------| 
|   |   | | 0D  | 15| 1D| 24| 2D| 2C| 35| 3C| 43| 44| 4D| 0E| 54|     | |*71|*69|*7A| | 6C| 75| 7D| 79| 
|-------| |------------------------------------------------------`    | `-----------' |---------------| 
|   |   | | 14   | 1C| 1B| 23| 2B| 34| 33| 3B| 42| 4B| 4C| 52| 53| 5A |     |*75|     | 6B| 73| 74|   | 
|-------| |-----------------------------------------------------------| ,-----------. |-----------|   | 
|   |   | | 12     | 1A| 22| 21| 2A| 32| 31| 3A| 41| 49| 4A| 51|  59  | |*6B| 62|*74| | 69| 72| 7A|*5A| 
|-------| |-----------------------------------------------------------| `-----------' |-----------|   | 
|   |   | | 13  |  | 19  |  67   |  29   |  64   | *11 |        | 58  |     |*72|     |     70| 71|   | 
`-------' `-----'  `-----------------------------------'        `-----'     `---'     `---------------'
*: E0 prefixed

```

RaoulDuke-Esq, 2021-05-29:
Happy to test!

```
5576-001 Code Set 82h:
             ,---------------.  ,---------------.  ,---------------.
             | ^12| 7E| 1F| 27|  | 2F| 5E| 08| 10|  | 50| 18| 17| 37|
             |----------------  |----------------  |---------------|
             | 05| 06| 04| 0C|  | 03| 0B| 83| 0A|  | 01| 09| 78| 07|
             `---------------'  `---------------'  `---------------'
,-------. ,-----------------------------------------------------------. ,-----------. ,---------------.
| F5| 48| | 76| 16| 1E| 26| 25| 2E| 36| 3D| 3E| 46| 45| 4E| 55| 5D| 66| |*7D|*69|*7A| | 7C|*4A|*41| 7B|
|-------| |-----------------------------------------------------------| |-----------| |---------------| 
| 38|^77 | | 0D  | 15| 1D| 24| 2D| 2C| 35| 3C| 43| 44| 4D| 0E| 54|     | |*70|*71|*40| | 6C| 75| 7D| 79| 
|-------| |------------------------------------------------------`    | `-----------' |---------------| 
| 30| 62| | 13   | 1C| 1B| 23| 2B| 34| 33| 3B| 42| 4B| 4C| 52| 5B| 5A |     |*75|     | 6B| 73| 74|   | 
|-------| |-----------------------------------------------------------| ,-----------. |-----------|   | 
| 20| 28| | 12     | 1A| 22| 21| 2A| 32| 31| 3A| 41| 49| 4A| 51|  59  | |*6B| *6C|*74| | 69| 72| 7A|*5A| 
|-------| |-----------------------------------------------------------| `-----------' |-----------|   | 
| 60| 14| | 11  |  | ^58  |  67   |  29   |  64   | *11 |        | *14  |     |*72|     |     70| 71|   | 
`-------' `-----'  `-----------------------------------'        `-----'     `---'     `---------------'
*: E0 prefixed
*7D: rE0 r7D rE0 rF0 r7D - others with * follow this pattern
^12: rE0 r12 rE0 r7C rE0 rF0 r7C rE0 rF0 r12
^77: rE1 r14 r77 rE1 rF0 r14 rF0 r77 
^58: r58 wED rFA w00 rFA rF0 r58 

```
Lots of differences in this one.

tmk, 2021-05-30:
Thanks for the test.

I decided to use Code Set 3 for 5576-001 and added support for the Japanese specific keys including￥(next to backspace), ろ(next to right shift).

https://gist.githubusercontent.com/tmk/537070e91e894fa18cc5f5f201856147/raw/7d9931664a8198462c35c9336b6b48549a517334/ibmpc_usb_5576_001.hex

All keys should be registered on computer as default keymap on Keymap Editor now, except two keys. The two keys(labeled as '無変換' and '変換') next to space bar are mapped to Keypad,(P,) and KeyPad.(P.) because of avoiding conflict with IBM 122-key Terminal layout.

Try new firmware and let me know if you find glitch or something.

I'll add 5576-001 support to source repo and update prebuilt firmwares later. Also  '無変換' and '変換' key of 5576 will be added to '122-key Terminal layout' of Keymap Editor. So that you can edit keymap for 5576-001 on Keymap Editor.

http://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?ibmpc_usb_32u4_terminal

RaoulDuke-Esq, 2021-05-30:
Thanks Hasu!

I'm using the new hex now.  Everything seems to match the default and what you described except I get KP Enter instead of KP Dot for the key to the right of the spacebar but I suppose that makes more sense for not causing conflicts with the 122 layout anyway.

Again, I'm so grateful for you taking this on!

tmk, 2021-05-30:
Sorry, I meant P=(KP equal) for '変換' key, not P.(KP dot).

The key should registers KP equal instead of KP Enter. I may be missing something.

RaoulDuke-Esq, 2021-05-30:
Ok- it is right then! The key does register KP Equal.

tmk, 2021-05-31:
I just updated Keymap Editor and prebuilt fimrwares to add support for 5576-001.
You can remap 5576-001 specific keys on Keymap Editor now, try this.

http://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?ibmpc_usb_32u4_terminal 

Thanks

RaoulDuke-Esq, 2021-05-31:
It's working great!  Thanks so much Hasu!!

closed 2021-05-31 by RaoulDuke-Esq