tmk/tmk_keyboard #687: [IBMPC_USB] Terminal Model M recognized as AT board

Razerban, 2021-05-31:
Hello Hasu,

I encountered an issue when using TMK's ibmpc_usb firmware on a ProMicro. (no pull-up resistors on the DATA and CLOCK lines) to convert a Model M 101-key (part number: 1394204).
[image]

The keyboard speaks codeset 3 but the firmware identifies it as an AT board:
```
*** HID console connected: t.m.k. IBM PC keyboard converter (FEED:1BEE:0001)
  > 
    
    TMK:e473e8ce/LUFA
    
    
    USB configured.
    
    Keyboard start.
    I398 wFF rFA A1403 
    PRT:10 ISR:FE90 rAA W1782 r7F W1783 r7F W1785 wF2 rFA r7F r7F R1791 
    ID:7F7F(AT) S1791 wED rFA w00 rFA L1799 
*** HID console disconnected: t.m.k. IBM PC keyboard converter (FEED:1BEE:0001)
```

When I plug the keyboard in to test it, some keys keep registering even when the they're no longer pressed. Some other keys don't even register.

On the other hand, Soarer's converter is able to identify it as a codeset 3 capable keyboard:
```
*** HID console connected: Soarer Soarer's Keyboard Converter (16C0:047D:0100)
  > wEE rEE wF2 rFA r7F r7F 
    
    remaining: FFFC
    wF8 rFA 
    
    Keyboard ID: 7F7F
    Code Set: 3
    Mode: AT/PS2
```

tmk, 2021-06-01:
I never found Keyboard ID 7F7F on the net, interestintg.
Did you open case and check controller PCB? If you have pics of the keyboard internal  can you post them somewehre?
I like to know if the controller PCB has a pin header to configure keyboard ID.   

Try this patch.
```
diff --git a/converter/ibmpc_usb/ibmpc_usb.c b/converter/ibmpc_usb/ibmpc_usb.c
index 7060480a..4f2fef6c 100644
--- a/converter/ibmpc_usb/ibmpc_usb.c
+++ b/converter/ibmpc_usb/ibmpc_usb.c
@@ -323,6 +323,8 @@ uint8_t matrix_scan(void)
                 keyboard_kind = PC_AT;
             } else if (0xBF00 == (keyboard_id & 0xFF00)) {  // CodeSet3 Terminal
                 keyboard_kind = PC_TERMINAL;
+            } else if (0x7F00 == (keyboard_id & 0xFF00)) {  // CodeSet3 Terminal
+                keyboard_kind = PC_TERMINAL;
             } else {
                 keyboard_kind = PC_AT;
             }
```

Razerban, 2021-06-01:
Thank you for your reply.

Please find hereunder the pictures you requested.
[image]

Thank you for the patch.
After applying it and compiling the firmware, I was able to make the keyboard work partially.
Some keys don't register and some keys send wrong(?) codes:
 
[image]

tmk, 2021-06-01:
Thanks for the pics.
It is intentional and default keymap for IBM 122-key and can be remap yourself.
http://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?ibmpc_usb_32u4_terminal

The default keymap for Termianal may be changed in future but there is no decisive keymap for Terminal layouts apparently.
https://www.seasip.info/VintagePC/ibm_1390876.html
https://www.seasip.info/VintagePC/ibm_6110344.html
https://www.seasip.info/VintagePC/ibm_1397000.html

Assuming you are using macos ScrollLock is recognized as F14.

You can use hid_listen to see scan codes for the keys which don't register. I think the keys send something but macos can't recognize. I guess the keys are mapped as 'Vol Down' and '無変換' on default keymap.
https://github.com/tmk/tmk_keyboard/wiki#debug

With Soarer's converter what do those keys register?

Razerban, 2021-06-01:
Yes, I'm using macOS.

This is the default keymap that I have when I flash Soarer's firmware without any configuration:
[image]

When trying to remap the keys using the [web UI](http://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?ibmpc_usb_32u4_terminal), I fail to understand which keys of the 122-key layout refer to which keys on the 102-key model.

I would appreciate it if you could you help me identify the position of the keys on the 122-key and their respective positions on the 102-key layout  :) ?

Thanks again Hasu!

tmk, 2021-06-02:
For example, to remap 'Home' on top right cornor in your keyboard you can edit 'Home' key(you can locate it in middle of cursors) on Keymap Eiditor. But just wait for now. I have something to know  about the keyboard before starting remap. 

Can you check scan codes in table below and correct if there is wrong code?
```
IBM 101-key Terminal Code Set 3:
,---.   ,---------------. ,---------------. ,---------------. ,-----------. 
| 83|   | 07| 0F| 17| 1F| | 27| 2F| 37| 3F| | 47| 4F| 56| 5E| | ??| ??| 62| 
`---'   `---------------' `---------------' `---------------' `-----------' 
,-----------------------------------------------------------. ,-----------. ,---------------.
| 0E| 16| 1E| 26| 25| 2E| 36| 3D| 3E| 46| 45| 4E| 55|   66  | | 67| 6E| 6F| | 76| 77| 7E| 84|
|-----------------------------------------------------------| |-----------| |---------------|
| 0D  | 15| 1D| 24| 2D| 2C| 35| 3C| 43| 44| 4D| 54| 5B|  5C | | 64| 65| 6D| | 6C| 75| 7D| 7C|
|-----------------------------------------------------------| `-----------' |---------------|
| 14   | 1C| 1B| 23| 2B| 34| 33| 3B| 42| 4B| 4C| 52| 53| 5A |               | 6B| 73| 74| 7B|
|-----------------------------------------------------------|     ,---.     |-----------|---|
|   12   | 1A| 22| 21| 2A| 32| 31| 3A| 41| 49| 4A|    59    |     | 63|     | 69| 72| 7A|   |
|-----------------------------------------------------------| ,-----------. |-----------| 79|
| 11  |   |19  |        29                   |39  |   | 58  | | 61| 62| 6A| |     70| 71|   |
`-----'   `---------------------------------------'   `-----' `-----------' `---------------'
```

tmk, 2021-06-02:
Also can try this patch and post debug outputs of startup?

This check if the keyboard supports Code Set 2. If so the keyboard may be supported with better keymap by default.

```
diff --git a/converter/ibmpc_usb/ibmpc_usb.c b/converter/ibmpc_usb/ibmpc_usb.c
index 4f2fef6c..d655b947 100644
--- a/converter/ibmpc_usb/ibmpc_usb.c
+++ b/converter/ibmpc_usb/ibmpc_usb.c
@@ -323,10 +323,19 @@ uint8_t matrix_scan(void)
                 keyboard_kind = PC_AT;
             } else if (0xBF00 == (keyboard_id & 0xFF00)) {  // CodeSet3 Terminal
                 keyboard_kind = PC_TERMINAL;
-            } else if (0x7F00 == (keyboard_id & 0xFF00)) {  // CodeSet3 Terminal
-                keyboard_kind = PC_TERMINAL;
             } else {
-                keyboard_kind = PC_AT;
+                xprintf("\nUnknown ID: Report to TMK ");
+                if ((0xFA == ibmpc_host_send(0xF0)) &&
+                    (0xFA == ibmpc_host_send(0x02))) {
+                    // switch to code set 2
+                    keyboard_kind = PC_AT;
+                } else if ((0xFA == ibmpc_host_send(0xF0)) &&
+                           (0xFA == ibmpc_host_send(0x03))) {
+                    // switch to code set 3
+                    keyboard_kind = PC_TERMINAL;
+                } else {
+                    keyboard_kind = PC_AT;
+                }
             }
 
             xprintf("\nID:%04X(%s) ", keyboard_id, KEYBOARD_KIND_STR(keyboard_kind));
```

Razerban, 2021-06-02:
> For example, to remap 'Home' on top right cornor in your keyboard you can edit 'Home' key(you can locate it in middle of cursors) on Keymap Eiditor. But just wait for now. I have something to know [...]

Here are the scan codes that I have:
```
IBM 101-key Terminal Code Set 3:
,---.   ,---------------. ,---------------. ,---------------. ,-----------. 
| ^ |   | 07| 0F| 17| 1F| | 27| 2F| 37| 3F| | 47| 4F| 56| 5E| | 57| 5F| 62| 
`---'   `---------------' `---------------' `---------------' `-----------' 
,-----------------------------------------------------------. ,-----------. ,---------------.
| 0E| 16| 1E| 26| 25| 2E| 36| 3D| 3E| 46| 45| 4E| 55|   66  | | 67| 6E| 6F| | 76| 77| 7E| 84|
|-----------------------------------------------------------| |-----------| |---------------|
| 0D  | 15| 1D| 24| 2D| 2C| 35| 3C| 43| 44| 4D| 54| 5B|  5C | | 64| 65| 6D| | 6C| 75| 7D| 7C|
|-----------------------------------------------------------| `-----------' |---------------|
| 14   | 1C| 1B| 23| 2B| 34| 33| 3B| 42| 4B| 4C| 52| 53| 5A |               | 6B| 73| 74| 7B|
|-----------------------------------------------------------|     ,---.     |-----------|---|
|   12   | 1A| 22| 21| 2A| 32| 31| 3A| 41| 49| 4A|    59    |     | 63|     | 69| 72| 7A|   |
|-----------------------------------------------------------| ,-----------. |-----------| 79|
| 11  |   |19  |        29                   |39  |   | 58  | | 61| * | 6A| |     70| 71|   |
`-----'   `---------------------------------------'   `-----' `-----------' `---------------'

Changes:
----------
^: was 83, found 08
*: was 62 (typo?), found 60
```

Razerban, 2021-06-02:
> Also can try this patch and post debug outputs of startup?
>
> This check if the keyboard supports Code Set 2. If so the keyboard may be supported with better keymap by default.
>
> ```
> diff --git [...]

This is the debug message at startup after applying the patch:
```
*** HID console connected: t.m.k. IBM PC keyboard converter (FEED:1BEE:0001)
  > 
    
    TMK:91b8bd02+/LUFA
    
    
    USB configured.
    
    Keyboard start.
    I402 wFF rFA A1407 
    PRT:10 ISR:FE90 rAA W1786 r7F W1788 r7F W1789 wF2 rFA r7F r7F R1795 
    Unknown ID: Report to TMK wF0 rFE wF0 rFE 
    ID:7F7F(AT) S1803 wED rFA w00 rFA L1811
```

tmk, 2021-06-02:
Thank you for the info and test.
The keyboard doesn't support Code Set 2 unfortunately. Ditch the patch and let go back with previous one.

'PrintScreen'(0x08) on left top corner can be remap with 'F13' key on Keymap Editor.
And 0x57 and 0x5F key can be remap with 'F23' and F24'

It seems that on macos F13 and F14 are recognized as 'PrintScreen' and 'ScrollLock' and 'PrintScreen' and 'ScrollLock' are recognized as F13 and F14 for some reason. Weird but Apple do that things.

```
/* 
 * Scan Code Set 3:
 *               ,-----------------------------------------------.
 *               |F13|F14|F15|F16|F17|F18|F19|F20|F21|F22|F23|F24|
 *               |-----------------------------------------------|
 *               |F1 |F2 |F3 |F4 |F5 |F6 |F7 |F8 |F9 |F10|F11|F12|
 *               `-----------------------------------------------'
 * ,-------. ,-----------------------------------------------------------. ,-----------. ,---------------.
 * |Mut|HEN| |  `|  1|  2|  3|  4|  5|  6|  7|  8|  9|  0|  -|  =|JPY| BS| |  /|PgU|PgD| |Esc|NmL|ScL|  *|
 * |-------| |-----------------------------------------------------------| |-----------| |---------------|
 * |VUp|Pau| |Tab  |  Q|  W|  E|  R|  T|  Y|  U|  I|  O|  P|  [|  ]|    \| |End|Ins|Del| |  7|  8|  9|  +|
 * |-------| |-----------------------------------------------------------| `-----------' |-----------|---|
 * |VDn|MHE| |CapsL |  A|  S|  D|  F|  G|  H|  J|  K|  L|  ;|  '|  #| Ret| |JPY|Up |  #| |  4|  5|  6|  -|
 * |-------| |-----------------------------------------------------------| ,-----------. |---------------|
 * |PrS|App| |Shif|  \|  Z|  X|  C|  V|  B|  N|  M|  ,|  ,|  /| RO| Shift| |Lef|Hom|Rig| |  1|  2|  3|Ent|
 * |-------| |-----------------------------------------------------------| `-----------' |-----------|---|
 * |Gui|Gui| |Ctrl|    |Alt |          Space              |Alt |    |Ctrl| | RO|Dow|KAN| |  ,|  0|  .|  =|
 * `-------' `----'    `---------------------------------------'    `----'  ---`---'---  `---------------'
 * 
 *               ,-----------------------------------------------.
 *               | 08| 10| 18| 20| 28| 30| 38| 40| 48| 50| 57| 5F|
 *               |-----------------------------------------------|
 *               | 07| 0F| 17| 1F| 27| 2F| 37| 3F| 47| 4F| 56| 5E| 
 *               `-----------------------------------------------' 
 * ,-------. ,-----------------------------------------------------------. ,-----------. ,---------------.
 * | 05| 06| | 0E| 16| 1E| 26| 25| 2E| 36| 3D| 3E| 46| 45| 4E| 55| 5D| 66| | 67| 6E| 6F| | 76| 77| 7E|*84|
 * |-------| |-----------------------------------------------------------| |-----------| |---------------|
 * | 04| 0C| | 0D  | 15| 1D| 24| 2D| 2C| 35| 3C| 43| 44| 4D| 54| 5B|  5C | | 64| 65| 6D| | 6C| 75| 7D| 7C|
 * |-------| |-----------------------------------------------------------| `-----------' |---------------|
 * | 03| 0B| | 14   | 1C| 1B| 23| 2B| 34| 33| 3B| 42| 4B| 4C| 52| 53| 5A | |-26| 63|-25| | 6B| 73| 74| 7B|
 * |-------| |-----------------------------------------------------------| ,-----------. |---------------|
 * |*83| 0A| | 12 | 13| 1A| 22| 21| 2A| 32| 31| 3A| 41| 49| 4A| 51|  59  | | 61| 62| 6A| | 69| 72| 7A| 79|
 * |-------| |-----------------------------------------------------------| `-----------' |---------------|
 * | 01| 09| | 11  |   |19  |        29                   |39  |   | 58  | |-16| 60|-1E| | 68| 70| 71| 78|
 * `-------' `-----'   `---------------------------------------'   `-----'  ---`---'---  `---------------'
 * *: Special codes remapped 83->02, 84->7F
 * -: G80-2551 specific 80-prefixed codes remapped: 26->5D, 25->53, 16->51, 1E->00
 * 51, 5C, 5D, 68, 78: Hidden keys in IBM 122-key terminal keyboard [7]
 */
```

Razerban, 2021-06-02:
> Thank you for the info and test.
> The keyboard doesn't support Code Set 2 unfortunately. Ditch the patch and let go back with previous one.
> 
> 'PrintScreen'(0x08) on left top corner can be remap with 'F13' key on Keymap Editor.
> And 0x57 and 0x5F key can be remap with 'F23' and F24'
> 
> It seems that on macos F13 and F14 are recognized as 'PrintScreen' and 'ScrollLock' and 'PrintScreen' and 'ScrollLock' are recognized as F13 and F14 for some reason. Weird but Apple do that things.
> 
> Following is scan code table for TMK 122-key Termainal layout, you can know how 101-key is mapped to 122-key there.
> 
> ```
> /* 
>  * Scan Code Set 3:
>  *               ,-----------------------------------------------.
>  *               |F13|F14|F15|F16|F17|F18|F19|F20|F21|F22|F23|F24|
>  *               |-----------------------------------------------|
>  *               |F1 |F2 |F3 |F4 |F5 |F6 |F7 |F8 |F9 |F10|F11|F12|
>  *               `-----------------------------------------------'
>  * ,-------. ,-----------------------------------------------------------. ,-----------. ,---------------.
>  * |Mut|HEN| |  `|  1|  2|  3|  4|  5|  6|  7|  8|  9|  0|  -|  =|JPY| BS| |  /|PgU|PgD| |Esc|NmL|ScL|  *|
>  * |-------| |-----------------------------------------------------------| |-----------| |---------------|
>  * |VUp|Pau| |Tab  |  Q|  W|  E|  R|  T|  Y|  U|  I|  O|  P|  [|  ]|    \| |End|Ins|Del| |  7|  8|  9|  +|
>  * |-------| |-----------------------------------------------------------| `-----------' |-----------|---|
>  * |VDn|MHE| |CapsL |  A|  S|  D|  F|  G|  H|  J|  K|  L|  ;|  '|  #| Ret| |JPY|Up |  #| |  4|  5|  6|  -|
>  * |-------| |-----------------------------------------------------------| ,-----------. |---------------|
>  * |PrS|App| |Shif|  \|  Z|  X|  C|  V|  B|  N|  M|  ,|  ,|  /| RO| Shift| |Lef|Hom|Rig| |  1|  2|  3|Ent|
>  * |-------| |-----------------------------------------------------------| `-----------' |-----------|---|
>  * |Gui|Gui| |Ctrl|    |Alt |          Space              |Alt |    |Ctrl| | RO|Dow|KAN| |  ,|  0|  .|  =|
>  * `-------' `----'    `---------------------------------------'    `----'  ---`---'---  `---------------'
>  * 
>  *               ,-----------------------------------------------.
>  *               | 08| 10| 18| 20| 28| 30| 38| 40| 48| 50| 57| 5F|
>  *               |-----------------------------------------------|
>  *               | 07| 0F| 17| 1F| 27| 2F| 37| 3F| 47| 4F| 56| 5E| 
>  *               `-----------------------------------------------' 
>  * ,-------. ,-----------------------------------------------------------. ,-----------. ,---------------.
>  * | 05| 06| | 0E| 16| 1E| 26| 25| 2E| 36| 3D| 3E| 46| 45| 4E| 55| 5D| 66| | 67| 6E| 6F| | 76| 77| 7E|*84|
>  * |-------| |-----------------------------------------------------------| |-----------| |---------------|
>  * | 04| 0C| | 0D  | 15| 1D| 24| 2D| 2C| 35| 3C| 43| 44| 4D| 54| 5B|  5C | | 64| 65| 6D| | 6C| 75| 7D| 7C|
>  * |-------| |-----------------------------------------------------------| `-----------' |---------------|
>  * | 03| 0B| | 14   | 1C| 1B| 23| 2B| 34| 33| 3B| 42| 4B| 4C| 52| 53| 5A | |-26| 63|-25| | 6B| 73| 74| 7B|
>  * |-------| |-----------------------------------------------------------| ,-----------. |---------------|
>  * |*83| 0A| | 12 | 13| 1A| 22| 21| 2A| 32| 31| 3A| 41| 49| 4A| 51|  59  | | 61| 62| 6A| | 69| 72| 7A| 79|
>  * |-------| |-----------------------------------------------------------| `-----------' |---------------|
>  * | 01| 09| | 11  |   |19  |        29                   |39  |   | 58  | |-16| 60|-1E| | 68| 70| 71| 78|
>  * `-------' `-----'   `---------------------------------------'   `-----'  ---`---'---  `---------------'
>  * *: Special codes remapped 83->02, 84->7F
>  * -: G80-2551 specific 80-prefixed codes remapped: 26->5D, 25->53, 16->51, 1E->00
>  * 51, 5C, 5D, 68, 78: Hidden keys in IBM 122-key terminal keyboard [7]
>  */
> ```

Thanks for your help Hasu :)

tmk, 2021-06-02:
I'll merge it in a few weeks with other fixes

tmk, 2021-06-03:
Just merged now!

Default keymap of 101-key/102-key Terminal looks like below.

```
,---.   ,---------------. ,---------------. ,---------------. ,-----------.
| 08|   | 07| 0F| 17| 1F| | 27| 2F| 37| 3F| | 47| 4F| 56| 5E| | 57| 5F| 62|
`---'   `---------------' `---------------' `---------------' `-----------'
,-----------------------------------------------------------. ,-----------. ,---------------.
| 0E| 16| 1E| 26| 25| 2E| 36| 3D| 3E| 46| 45| 4E| 55|   66  | | 67| 6E| 6F| | 76| 77| 7E| 84|
|-----------------------------------------------------------| |-----------| |---------------|
| 0D  | 15| 1D| 24| 2D| 2C| 35| 3C| 43| 44| 4D| 54| 5B|  5C | | 64| 65| 6D| | 6C| 75| 7D| 7C|
|-----------------------------------------------------------| `-----------' |---------------|
| 14   | 1C| 1B| 23| 2B| 34| 33| 3B| 42| 4B| 4C| 52| 53| 5A |               | 6B| 73| 74| 7B|
|-----------------------------------------------------------|     ,---.     |---------------|
| 12 | 13| 1A| 22| 21| 2A| 32| 31| 3A| 41| 49| 4A| 51|  59  |     | 63|     | 69| 72| 7A| 79|
|-----------------------------------------------------------| ,-----------. |---------------|
| 11  |   |19  |        29                   |39  |   | 58  | | 61| 60| 6A| | 68| 70| 71| 78|
`-----'   `---------------------------------------'   `-----' `-----------' `---------------'
Unimap:
,---.   ,---------------. ,---------------. ,---------------. ,-----------. 
|F13|   | F1| F2| F3| F4| | F5| F6| F7| F8| | F9|F10|F11|F12| |F23|F24|Hom| 
`---'   `---------------' `---------------' `---------------' `-----------' 
,-----------------------------------------------------------. ,-----------. ,---------------.
|  `|  1|  2|  3|  4|  5|  6|  7|  8|  9|  0|  -|  =|JPY|Bsp| |  /|PgU|PgD| |Esc|NmL|ScL|  *|
|-----------------------------------------------------------| |-----------| |---------------|
|Tab  |  Q|  W|  E|  R|  T|  Y|  U|  I|  O|  P|  [|  ]|  \  | |End|Ins|Del| |  7|  8|  9|  +|
|-----------------------------------------------------------| `-----------' |-----------|---|
|CapsL |  A|  S|  D|  F|  G|  H|  J|  K|  L|  ;|  '|  #|Entr|               |  4|  5|  6|  -|
|-----------------------------------------------------------|     ,---.     |---------------|
|Shft|  <|  Z|  X|  C|  V|  B|  N|  M|  ,|  .|  /| RO|Shift |     |Up |     |  1|  2|  3|Ent|
|-----------------------------------------------------------| ,-----------. |-----------|---|
|Ctrl|    |Alt |          Space              |Alt |    |Ctrl| |Lef|Dow|Rig| |  ,|  0|  .|  =|
`----'    `---------------------------------------'    `----' `-----------' `---------------'
```

closed 2021-06-03 by tmk