tmk/tmk_keyboard #688: IBM 5576-002/003 support

tmk, 2021-06-01:
The keyboard starts up with Code Set 2 by default and keeps Japanese specific keys disabled and emulates US layout.

To enable the Japanese keys you have to use Code Set 82h and some tweaks.

https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-AT-Keyboard-Protocol#ibm-5576-scan-codes-set

This is related to 5576-001 support. #685

tmk, 2021-06-02:
```
# IBM 5576-002 106-key:
Speculation from 5576-001 CS82h and Quckey! translation table.

,---.   ,-----------------------------------------------.     ,-----------.
| 76|   | 05| 06| 04| 0C| 03| 0B| 83| 0A| 01| 09| 78| 07|     |+7C| 7E|+77|
`---'   `-----------------------------------------------'     `-----------'
,-----------------------------------------------------------. ,-----------. ,---------------.
| 62| 16| 1E| 26| 25| 2E| 36| 3D| 3E| 46| 45| 4E| 55| 5D| 66| |*70|*6C|*7D| | 7C|*4A|*41| 7B|
|-----------------------------------------------------------| |-----------| |---------------|
| 0D  | 15| 1D| 24| 2D| 2C| 35| 3C| 43| 44| 4D| 0E| 54|     | |*71|*69|*7A| | 6C| 75| 7D|   |
|------------------------------------------------------`    | `-----------' |-----------| 79|
| 58   | 1C| 1B| 23| 2B| 34| 33| 3B| 42| 4B| 4C| 52| 5B| 5A |               | 6B| 73| 74|   |
|-----------------------------------------------------------|     ,---.     |---------------|
| 12     | 1A| 22| 21| 2A| 32| 31| 3A| 41| 49| 4A| 51|  59  |     |*75|     | 69| 72| 7A|   |
|-----------------------------------------------------------| ,-----------. |-----------|*5A|
| 14  |   |  13 |  67  |  29   |  64  | *11 |  11 |   | *14 | |*6B|*72|*74| |     70| 71|   |
`-----'   `---------------------------------------'   `-----' `-----------' `---------------'

Quckey! Translation:
KP*    7C -> 77     NumLock
KP'   *41 -> 7C     KP*
半/全  62 -> 0E     `~
       0E -> 54     [
       54 -> 5B     ]
       5B -> 5D     \
       5C -> 6A     JYEN
       5D -> 6A     JYEN
       11 -> *11    RAlt
       13 -> 11     LAlt
      *11 -> 13     KANA
```

InstantComet, 2022-07-13:
Apologies for my dumb questions, I wonder if the ibm 5576-002 has been supported? Can I use the IBMPC-USB Converter (ATmega32U4) preset on an ibm 5576-002 keyboard?

tmk, 2022-07-13:
The latest firmware built has preliminary support for the keyboard but has not been tested.
Please try it and report if you own the board. 

Posting debug log here would be helpful. You can chceck debug prints using `hid_listen`. 
https://github.com/tmk/tmk_keyboard/wiki#debug

InstantComet, 2022-07-21:
And thanks for your awesome work!

tmk, 2022-07-21:
Thanks for your report!

I fixed scan code translation of the key. It should work as 'Keypad *' key.
Can you try attached firmware?

https://gist.githubusercontent.com/tmk/10e103c7a38622199aca3e533721b9d1/raw/1a95841f7492664e1ec907f7218ccb2868d0fac7/ibmpc_usb_atmega32u4.hex

This is patch, just for reference.
```
commit 074d55b4e3744a015b7ab7e3f6cb4b262f80d518
Author: tmk <hasu@tmk-kbd.com>
Date:   Thu Jul 21 16:33:39 2022 +0900

    ibmpc_usb: 5576-002 Keypad , fix

diff --git a/converter/ibmpc_usb/ibmpc_usb.cpp b/converter/ibmpc_usb/ibmpc_usb.cpp
index 601fa10e..831abf69 100644
--- a/converter/ibmpc_usb/ibmpc_usb.cpp
+++ b/converter/ibmpc_usb/ibmpc_usb.cpp
@@ -880,8 +880,16 @@ int8_t IBMPCConverter::process_cs1(uint8_t code)
  */
 uint8_t IBMPCConverter::cs2_e0code(uint8_t code) {
     switch(code) {
-        // E0 prefixed codes translation See [a].
-        case 0x11: return 0x0F; // right alt
+        case 0x11:  if (0xAB90 == keyboard_id || 0xAB91 == keyboard_id)
+                        return 0x13; // Hiragana(5576) -> KANA
+                    else
+                        return 0x0F; // right alt
+
+        case 0x41:  if (0xAB90 == keyboard_id || 0xAB91 == keyboard_id)
+                        return 0x7C; // Keypad ,(5576) -> Keypad *
+                    else
+                        return (code & 0x7F);
+
         case 0x14: return 0x19; // right control
         case 0x1F: return 0x17; // left GUI
         case 0x27: return 0x1F; // right GUI
@@ -982,13 +990,6 @@ uint8_t IBMPCConverter::translate_5576_cs2(uint8_t code) {
     }
     return code;
 }
-uint8_t IBMPCConverter::translate_5576_cs2_e0(uint8_t code) {
-    switch (code) {
-        case 0x11: return 0x13; // Hiragana -> KANA
-        case 0x41: return 0x7C; // Keypad , -> Keypad *
-    }
-    return code;
-}
 
 int8_t IBMPCConverter::process_cs2(uint8_t code)
 {
@@ -1035,9 +1036,6 @@ int8_t IBMPCConverter::process_cs2(uint8_t code)
             }
             break;
         case CS2_E0:    // E0-Prefixed
-            if (0xAB90 == keyboard_id || 0xAB91 == keyboard_id) {
-                code = translate_5576_cs2_e0(code);
-            }
             switch (code) {
                 case 0x12:  // to be ignored
                 case 0x59:  // to be ignored
@@ -1082,9 +1080,6 @@ int8_t IBMPCConverter::process_cs2(uint8_t code)
             }
             break;
         case CS2_E0_F0: // Break code of E0-prefixed
-            if (0xAB90 == keyboard_id || 0xAB91 == keyboard_id) {
-                code = translate_5576_cs2_e0(code);
-            }
             switch (code) {
                 case 0x12:  // to be ignored
                 case 0x59:  // to be ignored
```

InstantComet, 2022-07-22:
The "," key on the numpad now functions currectly, awesome job! 
Also kooking forward to a fix of the online editor.
Thanks again.

tmk, 2022-07-22:
Great!

Just updated repo at 56e88e8. Online editor should be also fixed soon, you may need to reload browser. Let me know if you have any issue.

Thanks

closed 2022-07-22 by tmk