tmk/tmk_keyboard #691: Beeper For IBM PC USB

RaoulDuke-Esq, 2021-06-03:
Hi again Hasu!

On the Alps 102 4704 Pingmaster I'm using a TMK converter and the beeper works great.  

IBM PC USB converter is working great on my 5576-001!  This keyboard has a beeper and I'm wondering if there's anything I need to do in the firmware configuration prior to compiling in order to have the beeper work, or should it work natively out of the box?  When I opened it up for cleaning the solder joints between the speaker and the controller appeared to be good and when I adjust the volume slider I can hear a slight static crackling so I believe the speaker works but I don't get any beeps.

Thanks in advance!

tmk, 2021-06-03:
Hi, 
Beeper is not supported by IBM AT protocol as far as I know. So there is nothing to do with the converter at this time.

In this page author speculates that speaker is controlled through unused pins of DIN connector by PS/55 host.
https://diarywind.com/blog/e/ibm-5576002-keyboard-info.html

Also someone says that there are 5576 keyboard cables with speaker support and without, some cables have extra connections for speaker support.

Seems like there is no further information on the net.
You can trace lines on PCB to know how the speaker is connected if you are interested further.

RaoulDuke-Esq, 2021-06-03:
That’s sad, but thanks for the quick response, information, and tips for me to research further!

closed 2021-06-03 by RaoulDuke-Esq