tmk/tmk_keyboard #697: USB-USB: Ducky One 2 Mini

tmk, 2021-06-25:
Ducky One 2 Mini with 1.0.9(1.0.8) firmware doesn't have 'boot protocol' interface, os the converter can't recognize the keyboard.

- USB Descriptor - Ducky One 2 Mini 1.0.9
https://gist.github.com/tmk/79068a4fcc922e8877691d5518513914

I believe that this is not their intended desgin, but a glitch. 
Interface 0 looks like a boot keyboard interface(6KRO) but it doesn't have boot protocol configuration in descriptor.
Seems that Interface 1 is used for mouse, Interface 2 for report keyboard(NKRO) and Interace 3 for vendor specific purpose.

tmk, 2021-06-26:
This relaxes driver loading conditions in hidboot.h of UHS2.

```
diff --git a/hidboot.h b/hidboot.h
index e187189..cdca5f1 100644
--- a/hidboot.h
+++ b/hidboot.h
@@ -410,7 +410,9 @@ uint8_t HIDBoot<BOOT_PROTOCOL>::Init(uint8_t parent, uint8_t port, bool lowspeed
                                         USB_CLASS_HID,
                                         HID_BOOT_INTF_SUBCLASS,
                                         USB_HID_PROTOCOL_KEYBOARD,
-                                        CP_MASK_COMPARE_ALL> confDescrParserA(this);
+                                        CP_MASK_COMPARE_CLASS | CP_MASK_COMPARE_PROTOCOL
+                                        > confDescrParserA(this);
+                                        //CP_MASK_COMPARE_ALL> confDescrParserA(this);
 
                                 pUsb->getConfDescr(bAddress, 0, i, &confDescrParserA);
                                 if(bNumEP == (uint8_t)(totalEndpoints(BOOT_PROTOCOL)))
```

tmk, 2021-07-04:
Descriptors extracted by [usb_desc_dump](https://github.com/tmk/tmk_keyboard/tree/master/converter/usb_desc_dump) and parsed by [USB Descriptor and Request Parser](https://eleccelerator.com/usbdescreqparser/):

https://gist.github.com/tmk/6c9b88c117725446dd768a4270847019

The keyboard doesn't have Boot keyboard interface, it seems Interface0 should be that.

tmk, 2022-10-21:
TODO: New firmware 1.0.10 is available and should be tested again.

https://www.duckychannel.com.tw/en/Support/Detail/19

dinovo1, 2022-12-20:
Is there some older firmware version I can downgrade to and get it to work?

tmk, 2022-12-20:
Can you share its USB Descriptor here?

Please check 'Troubleshoot' in the first post to get USB Descriptor.
https://geekhack.org/index.php?topic=69169.0

dinovo1, 2022-12-20:
I got this using USB Tree Viewer (didn't want to reflash my TMK).

    =========================== USB Port1 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 1-1

      ========================== Summary =========================
Vendor ID                : 0x3233 (DuckyChannel International Co., Ltd.)
Product ID               : 0x6301
USB Version              : 1.1
Port maximum Speed       : Full-Speed
Device maximum Speed     : Full-Speed
Device Connection Speed  : Full-Speed
Self powered             : no
Demanded Current         : 100 mA
Used Endpoints           : 5

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : USB Composite Device
Device Path              : \\?\USB#VID_3233&PID_6301#DK-V1.10-201231#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Kernel Name              : \Device\USBPDO-12
Device ID                : USB\VID_3233&PID_6301\DK-V1.10-201231
Hardware IDs             : USB\VID_3233&PID_6301&REV_0000 USB\VID_3233&PID_6301
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0026 (GUID_DEVCLASS_USB)
Driver                   : system32\DRIVERS\usbccgp.sys (Version: 6.1.7601.18328  Date: 2013-11-26)
Driver Inf               : C:\Windows\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Service                  : usbccgp
Enumerator               : USB
Location Info            : Port_#0001.Hub_#0001
Container ID             : {58e72224-931a-5b00-b1a2-3d6b228565d4}
Manufacturer Info        : (Standard USB Host Controller)
Capabilities             : 0x94 (Removable, UniqueID, SurpriseRemovalOK)
Status                   : 0x0180400A (DN_DRIVER_LOADED, DN_STARTED, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Address                  : 1
HcDisableSelectiveSuspend: 0
EnableSelectiveSuspend   : 0
SelectiveSuspendEnabled  : 0
EnhancedPowerMgmtEnabled : 0
IdleInWorkingState       : 0
WakeFromSleepState       : 0
Power State              : D0 (supported: D0, D3, wake from D0)
 Child Device 1          : USB Input Device
  Device ID              : USB\VID_3233&PID_6301&MI_00\6&62897EB&0&0000
  Class                  : HIDClass
  Driver KeyName         : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}\0033 (GUID_DEVCLASS_HIDCLASS)
  Service                : HidUsb
  Location               : 0000.0004.0000.001.000.000.000.000.000
  LocationPaths          : PCIROOT(0)#PCI(0400)#USBROOT(0)#USB(1)#USB(1)
   Child Device 1        : HID Keyboard Device
    Device Path 1        : \\?\HID#VID_3233&PID_6301&MI_00#7&14fd6a88&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} (GUID_DEVINTERFACE_HID)
    Device Path 2        : \\?\HID#VID_3233&PID_6301&MI_00#7&14fd6a88&0&0000#{884b96c3-56ef-11d1-bc8c-00a0c91405dd} (GUID_DEVINTERFACE_KEYBOARD)
    Kernel Name          : \Device\000000ec
    Device ID            : HID\VID_3233&PID_6301&MI_00\7&14FD6A88&0&0000
    Class                : Keyboard
    Driver KeyName       : {4d36e96b-e325-11ce-bfc1-08002be10318}\0003 (GUID_DEVCLASS_KEYBOARD)
    Service              : kbdhid
 Child Device 2          : USB Input Device
  Device ID              : USB\VID_3233&PID_6301&MI_01\6&62897EB&0&0001
  Class                  : HIDClass
  Driver KeyName         : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}\0034 (GUID_DEVCLASS_HIDCLASS)
  Service                : HidUsb
  Location               : 0000.0004.0000.001.000.000.000.000.000
  LocationPaths          : PCIROOT(0)#PCI(0400)#USBROOT(0)#USB(1)#USB(1)
[... 228 lines cut ...]

      -------------------- String Descriptors -------------------
             ------ String Descriptor 0 ------
bLength                  : 0x04 (4 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language ID[0]           : 0x0409 (English - United States)
Data (HexDump)           : 04 03 09 04                                       ....
             ------ String Descriptor 1 ------
bLength                  : 0x0C (12 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "Ducky"
Data (HexDump)           : 0C 03 44 00 75 00 63 00 6B 00 79 00               ..D.u.c.k.y.
             ------ String Descriptor 2 ------
bLength                  : 0x28 (40 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "Ducky One2 Mini RGB"
Data (HexDump)           : 28 03 44 00 75 00 63 00 6B 00 79 00 20 00 4F 00   (.D.u.c.k.y. .O.
                           6E 00 65 00 32 00 20 00 4D 00 69 00 6E 00 69 00   n.e.2. .M.i.n.i.
                           20 00 52 00 47 00 42 00                            .R.G.B.
             ------ String Descriptor 3 ------
bLength                  : 0x20 (32 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "DK-V1.10-201231"
Data (HexDump)           : 20 03 44 00 4B 00 2D 00 56 00 31 00 2E 00 31 00    .D.K.-.V.1...1.
                           30 00 2D 00 32 00 30 00 31 00 32 00 33 00 31 00   0.-.2.0.1.2.3.1.

tmk, 2022-12-20:
Thank you for the descriptor. They didn't fix the descriptor to mark a interface as boot device.
I fixed compatibility list on the thread.

I still have an idea on converter firmware fix to support this keyboard.
Let me know when you have time and willing to flash your converter to debug it.

dinovo1, 2022-12-21:
Thanks for your reply.
At this point I am so dependent on my TMK converter that I don't want risk interrupting my workflow.
I would be interested in buying another one (as a backup) and then we can use that one to flash and debug (maybe you can already send it to me flashed with the changes you had in mind?). what do you think?

tmk, 2022-12-21:
I'm happy to flash improved firmware for testing when the converter is shipped, of course.
Please refer this URL when you place an order, so that I make sure to do that.

dinovo1, 2022-12-21:
Just sent you a direct email with a request for a paypal invoice.
Thanks!

tmk, 2023-01-19:
The keyboard doesn't support SET_PROTOCOL request, which ends with STALL.
After the STALL the keybaord stop working for no reason.

```
Waiting for device:
Listening:
Device disconnected.
Waiting for new device:.......
Listening:
_state: 20
usb_state: 40
usb_state: 50
usb_state: 51
Ci read:0012
left:0000
vid:3233
pid:6301
klass:00
subklass:00
AC1
AC2
A2 driver:00
BM Init
Ci read:0008
left:0000
Addr:01
Ci read:0012
left:0000
NC:01
bootIf[]:00
HID_PROTOCOL_KEYBOARD
[gC]Ci read:0009
left:0000
Cl:006D
Ci ConfVal: 01
Iface: 00
    Class: 03
    SubClass: 00
    Protocol: 01
Iface: 01
    Class: 03
    SubClass: 00
    Protocol: 02
read:0040
left:002D
Iface: 02
    Class: 03
    SubClass: 00
    Protocol: 01
Iface: 03
    Class: 03
    SubClass: 00
    Protocol: 00
read:002D
left:0000
[gC]Ci read:0009
left:0000
Cl:006D
Ci ConfVal: 01
Iface: 00
    Class: 03
    SubClass: 00
    Protocol: 01
[xE1]hidboot:EndpointXtract:iface: 00
hidboot:EndpointXtract:bNumEP: 01
[xE++] Iface: 01
    Class: 03
    SubClass: 00
    Protocol: 02
read:0040
left:002D
Iface: 02
    Class: 03
    SubClass: 00
    Protocol: 01
[xE1]hidboot:EndpointXtract:iface: 02
Iface: 03
    Class: 03
    SubClass: 00
    Protocol: 00
read:002D
left:0000
bNumEP:02
Cnf:01
bootIf[]:00
bIfaceNum:00
bNumIface:00
Interface:00
crST:05
SET_PROTOCOL: 05
SET_IDLE: 00
Ci read:0040
left:0040
read:0000
left:0040
RPIPE: 00
BM configured
Init:00
AttemptConfig2:00
Configuring:00
host.Task: 2340
usb_state: 90
speed: full
Co 
```

tmk, 2023-02-01:
Fixed at https://github.com/tmk/USB_Host_Shield_2.0/commit/204bc5d1aeb85f88a0ee980d1450e8b7ca56d0dd and available on repo and keymap editor now.

```
Waiting for device:.........
Listening:
_state: 12
Device disconnected.
Waiting for new device:.
Listening:
TMK:8fd2ba/LUFA:d6a7df/UHS2:069220
USB configured.
Loop start.
usb_state: 20
usb_state: 40
usb_state: 50
usb_state: 51
Ci read:0012
left:0000
vid:3233
pid:6301
klass:00
subklass:00
AC1
AC2
A2 driver:00
BM Init
Ci read:0008
left:0000
Addr:01
Ci read:0012
left:0000
NC:01
bootIf[]:00
HID_PROTOCOL_KEYBOARD
[gC]Ci read:0009
left:0000
Cl:006D
Ci ConfVal: 01
    Iface: 00
        Class: 03
        SubClass: 00
        Protocol: 01
    Iface: 01
        Class: 03
        SubClass: 00
        Protocol: 02
read:0040
left:002D
    Iface: 02
        Class: 03
        SubClass: 00
        Protocol: 01
    Iface: 03
        Class: 03
        SubClass: 00
        Protocol: 00
read:002D
left:0000
[gC]Ci read:0009
left:0000
Cl:006D
Ci ConfVal: 01
    Iface: 00
        Class: 03
        SubClass: 00
        Protocol: 01
[xE1]hidboot:EndpointXtract:iface: 00
hidboot:EndpointXtract:bNumEP: 01
[xE++]  Iface: 01
        Class: 03
        SubClass: 00
        Protocol: 02
read:0040
left:002D
    Iface: 02
        Class: 03
        SubClass: 00
        Protocol: 01
[xE1]hidboot:EndpointXtract:iface: 02
    Iface: 03
        Class: 03
        SubClass: 00
        Protocol: 00
read:002D
left:0000
bNumEP:02
Cnf:01
bootIf[]:00
bIfaceNum:00
bNumIface:00
Interface:00
skip SET_PROTOCOLSET_IDLE: 00
Ci read:0040
left:0040
read:0000
left:0040
RPIPE: 00
BM configured
Init:00
AttemptConfig2:00
Configuring:00
host.Task: 2342
usb_state: 90
speed: full
Co input 1: 00 00 07 00 00 00 00 00
00 00 07 00 00 00 00 00 
dinput 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 
input 1: 00 00 16 00 00 00 00 00
00 00 16 00 00 00 00 00 
sinput 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 
input 1: 00 00 04 00 00 00 00 00
00 00 04 00 00 00 00 00 
ainput 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 
usb_state: 12
usb_state: 20
usb_state: 40
usb_state: 50
usb_state: 51
Ci read:0012
left:0000
vid:3233
pid:6301
klass:00
subklass:00
AC1
AC2
A2 driver:00
BM Init
Ci read:0008
left:0000
Addr:01
Ci read:0012
left:0000
NC:01
bootIf[]:00
HID_PROTOCOL_KEYBOARD
[gC]Ci read:0009
left:0000
Cl:006D
Ci ConfVal: 01
    Iface: 00
        Class: 03
        SubClass: 00
        Protocol: 01
    Iface: 01
        Class: 03
        SubClass: 00
        Protocol: 02
read:0040
left:002D
    Iface: 02
        Class: 03
        SubClass: 00
        Protocol: 01
    Iface: 03
        Class: 03
        SubClass: 00
        Protocol: 00
read:002D
left:0000
[gC]Ci read:0009
left:0000
Cl:006D
Ci ConfVal: 01
    Iface: 00
        Class: 03
        SubClass: 00
        Protocol: 01
[xE1]hidboot:EndpointXtract:iface: 00
hidboot:EndpointXtract:bNumEP: 01
[xE++]  Iface: 01
        Class: 03
        SubClass: 00
        Protocol: 02
read:0040
left:002D
    Iface: 02
        Class: 03
        SubClass: 00
        Protocol: 01
[xE1]hidboot:EndpointXtract:iface: 02
    Iface: 03
        Class: 03
        SubClass: 00
        Protocol: 00
read:002D
left:0000
bNumEP:02
Cnf:01
bootIf[]:00
bIfaceNum:00
bNumIface:00
Interface:00
skip SET_PROTOCOLSET_IDLE: 00
Ci read:0040
left:0040
read:0000
left:0040
RPIPE: 00
BM configured
Init:00
AttemptConfig2:00
Configuring:00
host.Task: 2340
usb_state: 90
speed: full
Co input 1: 00 00 04 00 00 00 00 00
00 00 04 00 00 00 00 00 
ainput 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 
input 1: 00 00 05 00 00 00 00 00
00 00 05 00 00 00 00 00 
binput 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 
input 1: 00 00 06 00 00 00 00 00
00 00 06 00 00 00 00 00 
cinput 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 
usb_state: 12
usb_state: 20
usb_state: 40
usb_state: 50
usb_state: 51
Ci read:0012
left:0000
vid:3233
pid:6301
klass:00
subklass:00
AC1
AC2
A2 driver:00
BM Init
Ci read:0008
left:0000
Addr:01
Ci read:0012
left:0000
NC:01
bootIf[]:00
HID_PROTOCOL_KEYBOARD
[gC]Ci read:0009
left:0000
Cl:006D
Ci ConfVal: 01
    Iface: 00
        Class: 03
        SubClass: 00
        Protocol: 01
    Iface: 01
        Class: 03
        SubClass: 00
        Protocol: 02
read:0040
left:002D
    Iface: 02
        Class: 03
        SubClass: 00
        Protocol: 01
    Iface: 03
        Class: 03
        SubClass: 00
        Protocol: 00
read:002D
left:0000
[gC]Ci read:0009
left:0000
Cl:006D
Ci ConfVal: 01
    Iface: 00
        Class: 03
        SubClass: 00
        Protocol: 01
[xE1]hidboot:EndpointXtract:iface: 00
hidboot:EndpointXtract:bNumEP: 01
[xE++]  Iface: 01
        Class: 03
        SubClass: 00
        Protocol: 02
read:0040
left:002D
    Iface: 02
        Class: 03
        SubClass: 00
        Protocol: 01
[xE1]hidboot:EndpointXtract:iface: 02
    Iface: 03
        Class: 03
        SubClass: 00
        Protocol: 00
read:002D
left:0000
bNumEP:02
Cnf:01
bootIf[]:00
bIfaceNum:00
bNumIface:00
Interface:00
skip SET_PROTOCOLSET_IDLE: 00
Ci read:0040
left:0040
read:0000
left:0040
RPIPE: 00
BM configured
Init:00
AttemptConfig2:00
Configuring:00
host.Task: 2339
usb_state: 90
speed: full
Co input 1: 00 00 14 00 00 00 00 00
00 00 14 00 00 00 00 00 
qinput 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 
input 1: 00 00 1A 00 00 00 00 00
00 00 1A 00 00 00 00 00 
winput 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 
input 1: 00 00 08 00 00 00 00 00
00 00 08 00 00 00 00 00 
einput 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 
input 1: 00 00 15 00 00 00 00 00
00 00 15 00 00 00 00 00 
rinput 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 
input 1: 00 00 17 00 00 00 00 00
00 00 17 00 00 00 00 00 
tinput 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 
input 1: 00 00 1C 00 00 00 00 00
00 00 1C 00 00 00 00 00 
yinput 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 
usb_state: 12
```

closed 2023-02-01 by tmk