tmk/tmk_keyboard #703: usb_usb stops working when (1) switching on a KVM switch and (2) computer start from sleep

lightingghost, 2021-08-23:
I have a usb2usb converter (pre-assembled, not the pro micro one), I am using it by attaching a unifying receiver to it.

I found the keyboard lost connection when 
(1) switching to another computer with a KVM switch
(2) computer start from sleep. (macbook pro)

I also tried to connect the usb_usb converter directly to my computer using a type-c -> type-a convertor, it still stops working when starts from sleep.

I found [this PR](https://github.com/tmk/tmk_keyboard/pull/612) to address the similar issue (if not the same), but I built the firmware from the master brach, so it should already been merged. 

I would be happy to provide more details or follow your instructions to help debug this issue. Any help will be appreciated.

tmk, 2021-08-23:
What is model number of the Unifying redeiver and KVM switch?

Let's forget the KVM and focus (2) sleep problem here.
Do you need to replug the converter to recovery the problem, or just repluging the Unifying dongle? 

Connect converter directly to macbook. Can you share debug log with hid_listen when you sleep and resume?

tmk, 2021-08-23:
Use this debug firmware attached in this post.
https://geekhack.org/index.php?topic=69169.msg3075062#msg3075062

lightingghost, 2021-08-23:
@tmk 

Thank you for helping! I am providing the answers below:

> What is model number of the Unifying receiver and KVM switch?
The model number of the unifying receiver is C-U0007
I cannot find the model number of the KVM switch, but the it's UGREEN's usb 2.0 KVM switch

> Let's forget the KVM and focus (2) sleep problem here.
> Do you need to replug the converter to recovery the problem, or just repluging the Unifying dongle?
I need to replug the converter. Replugging the unifying dongle doesn't help -- replugging the unifying dongle makes the connection light on the keyboard blinks, which means it cannot connect the receiver. However, before replugging, although the keyboard has no input, light shows it is able to connect to the receiver.

> Connect converter directly to macbook. Can you share debug log with hid_listen when you sleep and resume?
 Here is the log after long sleep
(After a short time of sleep for example 5 min, the receiver is able to connect back with no problem: pressing the keyboard cannot wake up the computer, and the connection light shows it cannot connect the receiver, but after clicking the mouse to wake up, pressing the keyboard can make it connect the receiver. However, after a long sleep, for example over night, it cannot connect back. The log below is the one after a long sleep over night.)

```
Waiting for device:
Listening:

//here I replugged the converter 
Device disconnected.
Waiting for new device:...
Listening:

TMK:9154a8/LUFA:d6a7df/UHS2:ee3e6b

USB configured.

Loop start.
usb_state: 20
usb_state: 40
usb_state: 50
usb_state: 51
Ci read:0012
left:0000
vid:046D
pid:C52B
klass:00
subklass:00
AC1
AC2
A2 driver:00
BM Init
Ci read:0008
left:0000
Addr:01
Ci read:0012
left:0000
NC:01
HID_PROTOCOL_KEYBOARD
[gC]Ci read:0009
left:0000
Cl:0054
Ci [xE1][xE++]read:0040
left:0014
read:0014
left:0000
bNumEP:02
Cnf:01
bIfaceNum:00
bNumIface:00

Interface:00
PROTOCOL SET HID_BOOT rcode:00
SET_IDLE rcode:00
Ci read:003B
left:0045
RPIPE rcode:00
BM configured
Init:00
AttemptConfig2:00
Configuring:00
host.Task: 2323
usb_state: 90
speed: full
Co input 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
```

tmk, 2021-08-24:
I don't own any mac to test, but I just confirm on Windows 10 that the converter works through sleep/resume and wakeup by pressing a key. 

With other normal wired keyboard,  can the converter wake up mac?  and resume normally after long sleep?
I want to know whether the problem is specific to Unifying receiver or not.

lightingghost, 2021-08-24:
I tested using a normal wired keyboard. The converter cannot wake up mac, and it did not work after waking up.

I tried using my window PC as well: The keyboard connected to the converter cannot wake up from hibernation, but after waking up, it works without replugging.

tmk, 2021-08-25:
So the problem seems to be not specific to Unifying receiver but specific to Mac. Can you agree to this?

In general, you can't wake up by key press(or mouse click) from hibernation or deep sleep mode. This is expected behaviour, not problem. Becase those modes shut down power completely for the converter. Meanwhile, you can wake up Windows by key press in suspend or sleep mode.
And the converter and keyboard are expected to work without repluging after resuming from any power-saving modes, of course.

On your Windows PC does the covnerter work with both wired keyboard and Unifying as expected above?

lightingghost, 2021-08-29:
I don't know what happened, but the converter stopped working again after waking up. I changed back to the debug firmware and recorded the following log
```
Waiting for device:
Listening:
input 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
input 1: 00 00 05 00 00 00 00 00
00 00 05 00 00 00 00 00
binput 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
input 1: 00 00 06 00 00 00 00 00
00 00 06 00 00 00 00 00
cinput 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00

Device disconnected.
Waiting for new device:.....
Listening:

TMK:39273d/LUFA:d6a7df/UHS2:8dc3c8
[C][S][W][R][W][c]
USB configured.

Loop start.
usb_state: 12
[I0]6[L0][L0][L0][L0][L0]usb_state: 20
usb_state: 40
usb_state: 50
usb_state: 51
Ci read:0012
left:0000
vid:046D
pid:C52B
klass:00
subklass:00
AC1
AC2
A2 driver:00
BM Init
Ci read:0008
left:0000
Addr:01
Ci read:0012
left:0000
NC:01
HID_PROTOCOL_KEYBOARD
[gC]Ci read:0009
left:0000
Cl:0054
Ci [xE1][xE++]read:0040
left:0014
read:0014
left:0000
bNumEP:02
Cnf:01
bIfaceNum:00
bNumIface:00

Interface:00
PROTOCOL SET HID_BOOT rcode:00
SET_IDLE rcode:00
Ci read:003B
left:0045
RPIPE rcode:00
BM configured
Init:00
AttemptConfig2:00
Configuring:00
host.Task: 2325
usb_state: 90
speed: full
Co input 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
[L0]
Device disconnected.
Waiting for new device:..................................
Listening:

TMK:39273d/LUFA:d6a7df/UHS2:8dc3c8
[C][S][W][R][W][c]
USB configured.

Loop start.
usb_state: 20
[I0]6[P][L0][L0][L0][L0][L0]usb_state: 40
usb_state: 50
usb_st
Device disconnected.
Waiting for new device:...
Listening:

TMK:39273d/LUFA:d6a7df/UHS2:8dc3c8
[C][S][W][R][W][c]
USB configured.

Loop start.
usb_state: 20
[I0]6[P][L0][L0][L0][L0][L0]usb_state: 40
usb_state: 50
usb_st
Device disconnected.
Waiting for new device:...
Listening:

TMK:39273d/LUFA:d6a7df/UHS2:8dc3c8
[C][S][W][R][W][c]
USB configured.

Loop start.
usb_state: 12
[I0]6[P][L0][L0][L0][L0][L0]usb_state: 20
usb_state: 40
usb_state: 50
usb_state: 51
Ci read:0012
left:0000
vid:046D
pid:C52B
klass:00
subklass:00
AC1
AC2
A2 driver:00
BM Init
Ci read:0008
left:0000
Addr:01
Ci read:0012
left:0000
NC:01
HID_PROTOCOL_KEYBOARD
Device disconnected.
Waiting for new device:..
Listening:

TMK:39273d/LUFA:d6a7df/UHS2:8dc3c8
[C][S][W][R][W][c]
USB configured.

Loop start.
usb_state: 20
[I0]6[P][L0][L0][L0][L0][L0]usb_state: 40
usb_state: 50
usb_st
Device disconnected.
Waiting for new device:...
Listening:
gured.

Loop start.
usb_state: 20
[I0]6[P][L0][L0][L0][L0][L0]usb_state: 40
usb_state: 50
usb_st
Device disconnected.
Waiting for new device:..
Listening:

TMK:39273d/LUFA:d6a7df/UHS2:8dc3c8
[C][S][W][R][W][c]
USB configured.

Loop start.
usb_state: 20
[I0]6[P][L0][L0][L0][L0][L0]usb_state: 40
usb_state: 50
usb_state: 51
Ci read:0012
left:0000
vid:046D
pid:C52B
klass:00
subklass:00
AC1
AC2
A2 driver:00
BM Init
Ci read:0008
left:0000
Addr:01
nak_limit:7FFF
crSU:0E

getDevDescr 0E
Init:0E
BUSRST
AttemptConfig2:0E
Configuring:0E
host.Task: 408
usb_state: A0
usb_state: 12
```

tmk, 2021-08-29:
You used Unifying receiver this time while wired keyboard in the last test. With wired keyboard does the converter still work as the last test? Do you think Unifying receiver causes this problem?

In the log the converer failed to initialize Unifying receiver for some reason, `crSU:0E` in end part means timeout error.

How do you turn your Mac into standby mode? I assume that you have to wait for 3 hours using less 50% battery without AC power adapter unless you change power configuration with `pmset` or preference pane. Am I right?

It there a way to know current power state of Mac or discriminate between standby and normal sleep? LED indicator, I guess?
And according to the manual `pmset -g log` displays a history of sleeps, wakes, and other power management events. You can check its outputs after resuming test.

lightingghost, 2021-08-29:
> You used Unifying receiver this time while wired keyboard in the last test. With wired keyboard does the converter still work as the last test? Do you think Unifying receiver causes this problem?

The wired keyboard does not work with the converter as well. Here is the log after sleep over night. I have to replug:

```
Waiting for device:
Listening:

Device disconnected.
Waiting for new device:...
Listening:

TMK:39273d/LUFA:d6a7df/UHS2:8dc3c8
[C][S][W][R][W][c]
USB configured.

Loop start.
usb_state: 20
[I0]6[P][L0][L0][L0][L0][L0]usb_state: 40
usb_state: 50
usb_state: 51
Ci read:0012
left:0000
vid:FEED
pid:1964
klass:00
subklass:00
AC1
AC2
A2 driver:00
BM Init
Ci read:0008
left:0000
Addr:01
Ci read:0012
left:0000
NC:01
HID_PROTOCOL_KEYBOARD
[gC]Ci read:0009
left:0000
Cl:006D
Ci [xE1][xE++]read:0040
left:002D
read:002D
left:0000
bNumEP:02
Cnf:01
bIfaceNum:00
bNumIface:00

Interface:00
PROTOCOL SET HID_BOOT rcode:00
SET_IDLE rcode:00
Ci read:0040
left:0040
read:0000
left:0040
RPIPE rcode:00
BM configured
Init:00
AttemptConfig2:00
Configuring:00
host.Task: 2326
usb_state: 90
speed: full
Co input 1: 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
[L0]
Device disconnected.
Waiting for new device:.
[... 626 lines cut ...]
 womp                 1
 halfdim              1
 hibernatefile        /var/vm/sleepimage
 proximitywake        1
 powernap             1
 gpuswitch            2
 networkoversleep     0
 disksleep            10
 standbydelayhigh     300
 sleep                30
 hibernatemode        3
 ttyskeepawake        1
 displaysleep         30
 tcpkeepalive         1
 highstandbythreshold 50
 acwake               0
 lidwake              1

```

> It there a way to know current power state of Mac or discriminate between standby and normal sleep? LED indicator, I guess?

I don't think I can see the different without powering on the mac.

> And according to the manual `pmset -g log` displays a history of sleeps, wakes, and other power management events. You can check its outputs after resuming test.

tmk, 2021-08-31:
I still can't get good clue out of debug prints so far. I wildly guess that debug print USB interface can cause somehow.
You can disable the debug interface by having a line `CONSOLE_ENABLE = no` in Makefile.

Does MacOS have system logs under directory like /var/log and don't the logs have something informational when the problem occurs? Also trying command `$ pmset -g log` may display useful infos.

lightingghost, 2021-09-02:
Thanks for the suggestion. I tried to disable the debug interface but the problem is still there.

I tried to search system logs using console with the keyword of "usb" and "hid", but I could not find anything. I also tried to search in the log from `pmset -g log`, it doesn't seem to be anything interesting.

closed 2022-06-07 by tmk