tmk/tmk_keyboard #704: NeXT non-ADB keyboard timing

spenczar, 2021-09-11:
I recently acquired a NeXT non-ADB keyboard, and have been exploring it pretty thoroughly with an oscilloscope and logic analyzer. Like you (https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/protocol/next_kbd.c#L153-L162), I noticed that the timing of the keyboard's signal is not _quite_ 50 microseconds, or even 51 microseconds.

It turns out that the timing is actually 52.75 microseconds per pulse, corresponding to a frequency of 18,957 hertz. That's because the keyboard uses a Ceralock CSB455 oscillator:

[image]

This oscillator has a frequency of 455 kHz, and drives the keyboard's signal, updating the pulse every 24 clock cycles. 455 kHz / 24 is 18.957 kHz - resulting in the 52.75 microsecond pulses.

I have no reason to believe your code is behaving _too_ badly, but at the least you might want to fix comments and the header settings, for example here:
https://github.com/tmk/tmk_keyboard/blob/a93677cbf145baf210f92a9129ded55c9ef5657b/tmk_core/protocol/next_kbd.h#L54

and here: https://github.com/tmk/tmk_keyboard/blob/8c0e245f52182055de1d10397b58373f2bdad46d/tmk_core/protocol/next_kbd.c#L62-L67

tmk, 2021-09-12:
Thanks for the headsup.
I'll look into the timing when I setup NeXT keyboard and converter next time.

spenczar, 2022-01-24:
FWIW, I wrote more about this here, including logic analyzer and oscilloscope images of the timing: https://journal.spencerwnelson.com/entries/nextkb.html

tmk, 2022-04-17:
Thank you for your detailed writeup.

I also confirmed the bit width(52.75us) with my NeXT Keyboard using Logic Analyzer.

The converter can read signal at way better timing now.

Before Fix(signal read timing  at 9b99f8f)
[image]

After Fix(signal read timing at dcebfb4f)
[image]

In this capture the converter reads 'IN' at either edge of 'Read'.

closed 2022-04-17 by tmk

tmk, 2024-10-12:
Reimplemented NeXT converter interpreting the protocol as UART 9-bit data, not a long bit sequence. 

https://github.com/tmk/tmk_keyboard/wiki/NeXT#next-keyboard-protocol

https://github.com/tmk/tmk_keyboard/tree/master/converter/next_usb_ng