tmk/tmk_keyboard #712: tmk with bluetooth on MacBook Pro M1

EdmundsEcho, 2021-11-30:
Quick question -- 

1. Is the usb connection to the MacBook Monterey supported?  I ask because I can only connect using blue-tooth.
2. It's been a while since I used the keyboard.  How can I "discover" the keyboard layout?  For instance, I have forgotten how to change layers.

Thank you in advance for any guidance.

tmk, 2021-11-30:
1. I expect this should work but I don't know if it work or not with M1. 
2. Is your keyboard HHKB? If so you can find default layout here. If you changed your keymap and forgot it is difficult to know it.
http://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?hhkb_rn42
 
As for 1. You can use debug console to get debug prints. Try pressing both Shift key and 'v' to get fimrware version info. 
https://github.com/tmk/tmk_keyboard/wiki#debug-console

EdmundsEcho, 2021-12-13:
Thank you.  Yes the keyboard is HHKB pro 2. I did customize it.  What is the general method for changing layers?  I'll be sure to give shift-v to get the version and play with it some more.

tmk, 2021-12-13:
Hi,
You can enable Layer1 with Fn key(maked 'L1' on keymap editor) with default keymap, for example. 

Could you connect it to your Mac M1 with USB?

closed 2022-03-08 by tmk