tmk/tmk_keyboard #714: Add support for IBM M-50 to ibmpc_usb

RaoulDuke-Esq, 2021-12-31:
Hi Hasu,

Would it be possible to add layout support for the M-50 (50 key IBM keypad) to ibmpc_usb?  When loaded with the baseline ibmpc_usb firmware it registers as ID:AB83(AT) and sends the following:

All letters send as lowercase but are shown shifted for ease of read
`
A W E                     R \ U                  I O ; . (keypad dot)
Z X C                      V ENTER M       , L . ?
1 2 3                       4 F10 7             8 9 0 " (numbers are NOT keypad numbers)
LSFT F1 F2            5 F9 6               = F8 [ -
LCTL LALT F3       T BSPC Y.         ] F6 F7 H

`

tmk, 2021-12-31:
Can you share `hid_listen` ouput when the keyboard is plugged in?

Keboard layout is defined just as CPP macro like this.
https://github.com/tmk/tmk_keyboard/blob/792074f49b30ea9d39becdd3364cac7883024ac9/tmk_core/common/unimap.h#L49

You can define your layout in your keymap(unimap) file, unimap_plain.c for example.

'A' is identified by code 0x04 in unimap and 'W' by 0x1A. 
https://github.com/tmk/tmk_keyboard/blob/792074f49b30ea9d39becdd3364cac7883024ac9/tmk_core/common/unimap.h#L103

So `K04` and `K1A` are placed for A and W in paramter list part. you have to do that for all other keys. 
Unused codes should be defined as `AC_NO` in the latter part.

```
#define UNIMAP_M50( \
    K04, K1A, ... K66, \
...
    K78, K7A, ... K0B \
) { \
    { AC_NO,    AC_NO,    AC_NO,    AC_##K03, AC_##K04, AC_##K05, AC_##K06, AC_##K07,   /* 00-07 */ \
      AC_##K08, AC_##K09, AC_##K0A, AC_##K0B, AC_##K0C, AC_##K0D, AC_##K0E, AC_##K0F }, /* 08-0F */ \
    { AC_##K10, AC_##K11, AC_##K12, AC_##K13, AC_##K14, AC_##K15, AC_##K16, AC_##K17,   /* 10-17 */ \
      AC_##K18, AC_##K19, AC_##K1A, AC_##K1B, AC_##K1C, AC_##K1D, AC_##K1E, AC_##K1F }, /* 18-1F */ \
    { AC_##K20, AC_##K21, AC_##K22, AC_##K23, AC_##K24, AC_##K25, AC_##K26, AC_##K27,   /* 20-27 */ \
      AC_##K28, AC_##K29, AC_##K2A, AC_##K2B, AC_##K2C, AC_##K2D, AC_##K2E, AC_##K2F }, /* 28-2F */ \
    { AC_##K30, AC_##K31, AC_##K32, AC_##K33, AC_##K34, AC_##K35, AC_##K36, AC_##K37,   /* 30-37 */ \
      AC_##K38, AC_##K39, AC_##K3A, AC_##K3B, AC_##K3C, AC_##K3D, AC_##K3E, AC_##K3F }, /* 38-3F */ \
    { AC_##K40, AC_##K41, AC_##K42, AC_##K43, AC_##K44, AC_##K45, AC_##K46, AC_##K47,   /* 40-47 */ \
      AC_##K48, AC_##K49, AC_##K4A, AC_##K4B, AC_##K4C, AC_##K4D, AC_##K4E, AC_##K4F }, /* 48-4F */ \
    { AC_##K50, AC_##K51, AC_##K52, AC_##K53, AC_##K54, AC_##K55, AC_##K56, AC_##K57,   /* 50-57 */ \
      AC_##K58, AC_##K59, AC_##K5A, AC_##K5B, AC_##K5C, AC_##K5D, AC_##K5E, AC_##K5F }, /* 58-5F */ \
    { AC_##K60, AC_##K61, AC_##K62, AC_##K63, AC_##K64, AC_##K65, AC_##K66, AC_##K67,   /* 60-67 */ \
      AC_##K68, AC_##K69, AC_##K6A, AC_##K6B, AC_##K6C, AC_##K6D, AC_##K6E, AC_##K6F }, /* 68-6F */ \
    { AC_##K70, AC_##K71, AC_##K72, AC_##K73, AC_##K74, AC_##K75, AC_##K76, AC_##K77,   /* 70-77 */ \
      AC_##K78, AC_##K79, AC_##K7A, AC_##K7B, AC_##K7C, AC_##K7D, AC_##K7E, AC_##K7F }  /* 78-7F */ \
}
```

RaoulDuke-Esq, 2022-01-01:
Thanks Hasu.  I'll try creating a unman using those instructions.

I don't seem to be able to run hid_listen on any of my computers but I am able to get the console output:
[image]