tmk/tmk_keyboard #715: PC98 keys stutters when held down

Crizender, 2022-01-16:
Similar to the the issue detailed here : https://github.com/tmk/tmk_keyboard/issues/666

I have an Elsa Notekey A106 running a PC98 internal converter which has the keys periodically stutter. I have ran the it with the latest firmware and it still occurs. My AVR is a pro micro. 

Using HID listen when holding down the "a" key for a few seconds shows the below: 

> Waiting for device:
> Listening:
> r1D r9D r1D r9D r1D r9D r1D r9D r1D r9D r1D r9D r1D r9D r1D r9D r1D r9D r1D r9D r1D r9D r1D r9D r1D r9D r1D r9D r1D r9D 

Any help would be appreciated :)

tmk, 2022-01-16:
What is displayed when keyboard is plugged into converter and when connecting the converter to usb port ? I like to see How the keyboard is initialized, particularly.

And Does the keyboard has LED and it worked?

Crizender, 2022-01-16:
When plugged in, it shows up as PC98 keyboard converter.

It has two LEDs, one for Caps lock and another for Kana. They will light up but do not sync properly with the status (for example Caps will light up both its own light and the Kana light, then turn off both. Another keyswitch is needed for it to turn the light back on for Caps).

tmk, 2022-01-16:
Didn't you get output like this?
https://github.com/tmk/tmk_keyboard/issues/666#issuecomment-761516847

Crizender, 2022-01-17:
Ah sorry I didn't see that debug firmware, is there one for atmega32u4?

tmk, 2022-01-17:
No debug firmware is available. Just try default firmware and I think you already flashed it on your converter. 

The default firmware should display this version string. If you are on Windows try several times. Sometimes hid_listen on Windows doesn't work well. Better use Linux instead if possible.
```
TMK:8c0e24/LUFA:d6a7df
```

Also check this and download hid_listen if needed.
https://github.com/tmk/tmk_keyboard/wiki#debug-console

Crizender, 2022-01-17:
Oh my bad!

here you go: 

> Waiting for new device:....
> Listening:
> 
> TMK:8c0e24/LUFA:d6a7df
> s9F rFFFF old type
> s9C rFFFF
> USB configured.
> 
> Loop start.

tmk, 2022-01-17:
Hmm,
The keyboard  doesn't seem to reply to the command(9C) which prohibits key repeating.

Can you try this procedure?
1. Connect the keyboard with the converter first, 
2. then plug the converter into USB port.

You will see this line below if your keyboard supports the command. Your keyboard looks one of later models and is expected to have the feature.

`s9C r00FA s70 r00FA`

And I like to see debug outputs when pressing Caps Lock key.

Crizender, 2022-01-17:
I tried it to no changes sadly. I do have it internally converted however if this is causing an issue. 

pressing caps lock three times emits this: 

> r71 s9D rFFFF 
> r71 
> rF1 s9D rFFFF

tmk, 2022-01-18:
Then, the keyboard still doesn't respond to any command for some reason, or it doesn't receive command signal at all perhaps?

It should support commnads as it has LED indicators probably. I would check if all pins are wired correctly again, also Pro Micro specific info. 
https://github.com/tmk/tmk_keyboard/wiki/PC-9801-Keyboard
https://github.com/tmk/tmk_keyboard/wiki/Pro-Micro

I don't test PC98 converter firmware with Pro Micro or ATmega32u4, it may be a cause. My target is ATmega32u2 for development.

Let me know if you find something.

tmk, 2022-01-18:
Can you post pic of the keyboard for future eference?
I can't find its info much.

Crizender, 2022-01-18:
Here is a few pictures of the keyboard: https://imgur.com/a/eIQiUdB
I have also not been able to find much on it online, the best I found was this: https://web.archive.org/web/20160311051243/http://kbd.rzw.jp:80/mechanical/elsa_notekey-106/

I have also done a bit of probing on the wires this evening, I had a 2 cables the wrong way around. Pressing Caps twice now outputs the below, showing the correct state on the LED.

> r71 s9D rFFFF 
> rF1 s9D rFFFF

I will continue to probe if any of the other wires are the cause of the issue.

tmk, 2024-11-18:
Refer this for key repeating issue also:

https://github.com/tmk/tmk_keyboard/wiki/PC-9801-Keyboard#key-repeating

closed 2024-11-18 by tmk