tmk/tmk_keyboard #716: PS/2 3-Button Intellimouse recognized as Default

Bmkool, 2022-01-31:
The 3 button PS/2 Intellimouse is recognized as a standard PS/2 mouse by the library. 

In `converter/ibmpc_usb/ibmpc_usb.cpp`. 
```
MOUSE_INTELLI:
                    // Intellimouse protocol: 3
                    xprintf("\nINT: ");
                    ibmpc.host_send(0xF3); ibmpc.host_send(0xC8);
                    ibmpc.host_send(0xF3); ibmpc.host_send(0x64);
                    ibmpc.host_send(0xF3); ibmpc.host_send(0x50);
                    mouse_id = ((read_keyboard_id() >> 8) == MOUSE_INTELLI ? MOUSE_INTELLI : MOUSE_DEFAULT);

                    // Intellimouse Explorer protocol: 4
                    xprintf("\nEXP: ");
                    ibmpc.host_send(0xF3); ibmpc.host_send(0xC8);
                    ibmpc.host_send(0xF3); ibmpc.host_send(0xC8);
                    ibmpc.host_send(0xF3); ibmpc.host_send(0x50);
                    mouse_id = ((read_keyboard_id() >> 8) == MOUSE_EXPLORER ? MOUSE_EXPLORER : MOUSE_DEFAULT);

                    // Not Intellimouse
                    if (mouse_id == 0) {
                        xprintf("\nDEF: ");
                        ibmpc.host_send(0xF6);  // Set Default
                    }
````

Looks like the mouse_id gets set back to default in the protocol 4 check  even if the ID `03` is read again.

tmk, 2022-01-31:
How should we fix for this?

Bmkool, 2022-02-01:
Yes that's a three button with the scroll wheel. I think I fixed it by adding a check before the Intellimouse explorer that it is indeed an Intellimouse:

```
MOUSE_INTELLI:
                    // Intellimouse protocol: 3
                    xprintf("\nINT: ");
                    ibmpc.host_send(0xF3); ibmpc.host_send(0xC8);
                    ibmpc.host_send(0xF3); ibmpc.host_send(0x64);
                    ibmpc.host_send(0xF3); ibmpc.host_send(0x50);
                    mouse_id = ((read_keyboard_id() >> 8) == MOUSE_INTELLI ? MOUSE_INTELLI : MOUSE_DEFAULT);
                    if (MOUSE_INTELLI == mouse_id) {
                        // Intellimouse Explorer protocol: 4
                        xprintf("\nEXP: ");
                        ibmpc.host_send(0xF3); ibmpc.host_send(0xC8);
                        ibmpc.host_send(0xF3); ibmpc.host_send(0xC8);
                        ibmpc.host_send(0xF3); ibmpc.host_send(0x50);
                        mouse_id = ((read_keyboard_id() >> 8) == MOUSE_EXPLORER ? MOUSE_EXPLORER : MOUSE_INTELLI);
                    }

                    // Not Intellimouse
                    if (mouse_id == 0) {
                        xprintf("\nDEF: ");
                        ibmpc.host_send(0xF6);  // Set Default
                    }
```

Although I don't have an Intellimouse explorer to test this with, according to [this](https://isdaman.com/alsos/hardware/mouse/ps2interface.htm) that appears to be the expected behavior.

Bmkool, 2022-02-01:
Yep looks good to me.

commit 57c3f4a6 (2022-02-01): ibmpc_usb: Intellimouse fix #716
closed 2022-02-01 by tmk