tmk/tmk_keyboard #718: IBMPC: Sporadic hotplug failure for PS/2 mouse

Bmkool, 2022-02-10:
I've ran across a few scenarios where a PS/2 mouse fails to re-connect after hotplugging. The converter is then frozen and not functional for that port until the processor is restarted. 

This tends to happen when:
 1) Nothing is plugged into the PS/2 port when powered on and plugged in later
	```
	TMK:57c3f4/LUFA:d6a7df

	USB configured.

	Loop start.
	I338 I338 A3339 wFF R R R R R R R R R R A4057 wFF R R R R R R R R R R
	4776 ERR:11 ISR:8000
	4777 ERR:11 ISR:8000 X5278 X5279
	26593 ERR:20 ISR:F800
	```
 2) Unplugged during the initialization process 
	```
	Loop start.
	I306 I306 rAA
	645 PRT:10 ISR:EA90 r00 A3307 wFF R R R R R R R R R R A4025 wFF R R R R R R R R R R
	4744 ERR:11 ISR:8000
	4745 ERR:11 ISR:8000 X5246 X5247
	11280 ERR:20 ISR:C000
	```
 
I have not experienced any issues hotplugging a keyboad.

tmk, 2022-02-11:
I'm also aware of the issue. I'll look into when I have time.

tmk, 2022-05-28:
Fixed at 27178e956927

closed 2022-05-28 by tmk