tmk/tmk_keyboard #723: Add support for RT protocol to ibmpc_usb

RaoulDuke-Esq, 2022-03-26:
Hi Hasu!

Would you be willing to add support for RT protocol?  Codes are similar to PS/2 but with some differences described in the documentation below page 482-502.

https://archive.org/details/bitsavers_ibmpcrt75XferenceVolume1Jun87_19098532/page/n481/mode/2up

tmk, 2022-03-26:
Hi,
I'm happy to work on. How can I help you?

Do you have the keyboard?
Did you try it with ibmpc_usb?

RaoulDuke-Esq, 2022-03-26:
Thanks.  I think the help I need is adding to ibmpc_usb handling for keyboards with RT protocol and their different key codes so that when it detects an RT keyboard it uses the right codes.

These are the RT codes:
[image]

Those codes go with this layout:
[image]

Additional codes:
[image]

I tried my RT keyboard with a standard ibmpc_usb hex.

```
 A3221 wFF rFA A3226 wFF R R R R R R R R R R rAA W3945 
3945 ERR:11 ISR:8000 rBF W3946 rB1 W3947 wF2 rFA rBF rB1 R3957 
ID:BFB1(TERMINAL) S3957 wF8 rFA wED rFA w00 rFA L3972 X4447
```

Some keys in RT protocol are different than PS/2, so we get different results.

Caps Lock turns on Num Lock light but does lock caps.  `r14 wED rFA w04 rFA rF0 r14`

ESC registers as F13. `r08 F0 r08`

Insert registers as / `r67 /rF0 r67`

Delete registers as End `r64 rF0 r64`

Home registers as Page Up `r6E rF0 r6E`

End registers as Insert `r65 rF0 r65`

Page Up registers as Page Down `r6F rF0 r6F`

Page Down registers as Delete `r6D rF0 r6D`

NumLock registers as ESC `r76 rF0 76`

/ registers as NumLock `r77 rF0 77`

* doesn't register as anything `r7E rF0 7E`

- registers as * `r84 F0 r84`

+ doesn't seem to send any code

Print Screen sends `r57 rF0 57`

Scroll Lock sends `r5F rF0 r5F`
  
Pause registers as Home `r62 rF0 62`

The keyboard has a speaker.  I have it connected to C6 on my Pro Micro 32U4.
 
It seems that Soarer implemented RT support before disappearing, but I'd prefer to use TMK.
The issues are similar to this conversation: https://deskthority.net/viewtopic.php?f=7&t=2510&start=210#p105149

tmk, 2022-03-26:
Its keyboard ID is BFB1, not BFB0. Interesting.
Technical Reference says that RT keyboard replies with BFB0. 
Also this post shows BFB0. https://deskthority.net/viewtopic.php?p=105422#p105422
I guess RT keyboard has some variants or different languag layout has different ID.
Does your keyboard has DIP switches or jumper pins to change keyboard ID?

What is layout of your keyboard?
Can you share photos of your RT keyboard for future reference?

tmk, 2022-03-27:
I will handle RT keyboard as a kind of Terminal keyboard. It looks equivalent to IBM 101-key/102-key Terminal keyboard, as for scan code.
https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-Keyboard-Converter#ibm-terminal-101-key102-key-and-ibm-rt

The current converter firmware supports keymap for 122-key Terminal by default. 
You can remap using its keymap using for RT and 101-key/102-key Terminal keyboard.

http://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?ibmpc_usb_32u4_terminal

ESC,  PrintScreen, ScrollLock and Pause on RT keyboard can be remapped with F13, F23, F24 and Home on Keymap Editor recpectively.

Current firmware doens't support RT LED indicator, this causes Lockup of CapsLock key probably. I'll work on this. 

Do you know how the speaker is connected to the pin and how we can control it? Does Soarer's support it?
It would be fun if you can look into and implement speaker function. But it is not critcal and out of my interest.

tmk, 2022-03-27:
> Current firmware doens't support RT LED indicator, this causes Lockup of CapsLock key probably. I'll work on this.

Try this patch and check if the indicators show their states.

```
diff --git a/converter/ibmpc_usb/ibmpc_usb.cpp b/converter/ibmpc_usb/ibmpc_usb.cpp
index 4dff43ca..721f5f70 100644
--- a/converter/ibmpc_usb/ibmpc_usb.cpp
+++ b/converter/ibmpc_usb/ibmpc_usb.cpp
@@ -109,14 +109,26 @@ void IBMPCConverter::set_led(uint8_t usb_led)
     // - Some other Terminals like G80-2551 supports the command.
     //   https://geekhack.org/index.php?topic=103648.msg2894921#msg2894921
 
-    // TODO: PC_TERMINAL_IBM_RT support
+    // 7   6   5   4   3   2   1   0
+    // SCR CAP NUM --- --- CAP NUM SCR
+    // RT  RT  RT          AT  AT  AT
     uint8_t ibmpc_led = 0;
-    if (usb_led &  (1<<USB_LED_SCROLL_LOCK))
-        ibmpc_led |= (1<<IBMPC_LED_SCROLL_LOCK);
-    if (usb_led &  (1<<USB_LED_NUM_LOCK))
-        ibmpc_led |= (1<<IBMPC_LED_NUM_LOCK);
-    if (usb_led &  (1<<USB_LED_CAPS_LOCK))
-        ibmpc_led |= (1<<IBMPC_LED_CAPS_LOCK);
+    if (0xBFB0 == keyboard_id || 0xBFB1 == keyboard_id) {    // IBM RT Keyboard
+        if (usb_led &  (1<<USB_LED_SCROLL_LOCK))
+            ibmpc_led |= (1<<IBMPC_LED_SCROLL_LOCK_RT);
+        if (usb_led &  (1<<USB_LED_NUM_LOCK))
+            ibmpc_led |= (1<<IBMPC_LED_NUM_LOCK_RT);
+        if (usb_led &  (1<<USB_LED_CAPS_LOCK))
+            ibmpc_led |= (1<<IBMPC_LED_CAPS_LOCK_RT);
+    } else {                                                // IBM AT Keyboard
+        if (usb_led &  (1<<USB_LED_SCROLL_LOCK))
+            ibmpc_led |= (1<<IBMPC_LED_SCROLL_LOCK);
+        if (usb_led &  (1<<USB_LED_NUM_LOCK))
+            ibmpc_led |= (1<<IBMPC_LED_NUM_LOCK);
+        if (usb_led &  (1<<USB_LED_CAPS_LOCK))
+            ibmpc_led |= (1<<IBMPC_LED_CAPS_LOCK);
+    }
+
     ibmpc.host_set_led(ibmpc_led);
 }
 
@@ -363,16 +375,18 @@ uint8_t IBMPCConverter::process_interface(void)
                         }
                     }
                 }
-            } else if (0xBFB0 == keyboard_id) {     // IBM RT Keyboard
+            } else if (0xBFB0 == keyboard_id || 0xBFB1 == keyboard_id) {    // IBM RT Keyboard
                 // https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-AT-Keyboard-Protocol#bfb0
+                // https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-AT-Keyboard-Protocol#bfb1
                 // TODO: LED indicator fix
-                //keyboard_kind = PC_TERMINAL_IBM_RT;
                 keyboard_kind = PC_TERMINAL;
             } else if (0xAB00 == (keyboard_id & 0xFF00)) {  // CodeSet2 PS/2
                 keyboard_kind = PC_AT;
             } else if (0xBF00 == (keyboard_id & 0xFF00)) {  // CodeSet3 Terminal
+                // https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-AT-Keyboard-Protocol#bfbf
                 keyboard_kind = PC_TERMINAL;
             } else if (0x7F00 == (keyboard_id & 0xFF00)) {  // CodeSet3 Terminal 1394204
+                // https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-AT-Keyboard-Protocol#7f7f
                 keyboard_kind = PC_TERMINAL;
             } else {
                 xprintf("\nUnknown ID: Report to TMK ");
diff --git a/tmk_core/protocol/ibmpc.hpp b/tmk_core/protocol/ibmpc.hpp
index d1765e7f..2e04c85f 100644
--- a/tmk_core/protocol/ibmpc.hpp
+++ b/tmk_core/protocol/ibmpc.hpp
@@ -90,6 +90,11 @@ POSSIBILITY OF SUCH DAMAGE.
 #define IBMPC_LED_NUM_LOCK    1
 #define IBMPC_LED_CAPS_LOCK   2
 
+// https://archive.org/details/bitsavers_ibmpcrt75XferenceVolume1Jun87_19098532/page/n484/mode/1up
+#define IBMPC_LED_NUM_LOCK_RT       5
+#define IBMPC_LED_CAPS_LOCK_RT      6
+#define IBMPC_LED_SCROLL_LOCK_RT    7
+
 
 class IBMPC
 {
```

RaoulDuke-Esq, 2022-03-27:
Here are a couple images of mine:
[image]
[image]

Mine is NOS and has differences compared to those shown in the IBM documentation, mainly that the split left space keys and ISO Enter keys are not printed.  There are also no identifying labels on the back of mine.  Very unusual.  

There's a DT thread with many images of the interior for this type of keyboard and how the speaker is connected - seems to be simple 2-wire https://deskthority.net/viewtopic.php?f=62&t=4101.  I'm assuming that like other IBM keyboard speakers, the host was intended to send directly.

I will have to take mine apart soon, as I've noticed that several keys only register if I press them down hard and hold them down.  There's no way it could be usable at this point because of this.  The keyboard is capacitive (Brother dome and foil) so I'm wondering if there's something going on inside that's causing the capacitance to not register properly.  The cord is long and thick, but I used 1k pullups on clock and data.  I am also directly connecting my converter to the computer - no hubs.

I tried patching the two files in question but it's possible I did something wrong because after flashing and rebooting the console gets flooded and causes QMK Toolbox to freeze.

tmk, 2022-03-27:
> I tried patching the two files in question but it's possible I did something wrong because after flashing and rebooting the console gets flooded and causes QMK Toolbox to freeze.

https://github.com/tmk/tmk_keyboard/wiki#debug-console

tmk, 2023-03-05:
Scan code Set 3

IBM RT and IBM Terminal 101-key/102-key have same scan codes.

```
,---.   ,---------------. ,---------------. ,---------------. ,-----------. 
|F13|   | F1| F2| F3| F4| | F5| F6| F7| F8| | F9|F10|F11|F12| |F23|F24|Hom| 
`---'   `---------------' `---------------' `---------------' `-----------' 
,-----------------------------------------------------------. ,-----------. ,---------------.
|  `|  1|  2|  3|  4|  5|  6|  7|  8|  9|  0|  -|  =|JPY|Bsp| |  /|PgU|PgD| |Esc|NmL|ScL|  *|
|-----------------------------------------------------------| |-----------| |---------------|
|Tab  |  Q|  W|  E|  R|  T|  Y|  U|  I|  O|  P|  [|  ]|  \  | |End|Ins|Del| |  7|  8|  9|  +|
|-----------------------------------------------------------| `-----------' |-----------|---|
|CapsL |  A|  S|  D|  F|  G|  H|  J|  K|  L|  ;|  '|  #|Entr|               |  4|  5|  6|  -|
|-----------------------------------------------------------|     ,---.     |---------------|
|Shft|  <|  Z|  X|  C|  V|  B|  N|  M|  ,|  .|  /| RO|Shift |     |Up |     |  1|  2|  3|Ent|
|-----------------------------------------------------------| ,-----------. |-----------|---|
|Ctrl|    |Alt |          Space              |Alt |    |Ctrl| |Lef|Dow|Rig| |  ,|  0|  .|  =|
`----'    `---------------------------------------'    `----' `-----------' `---------------'
,---.   ,---------------. ,---------------. ,---------------. ,-----------.
| 08|   | 07| 0F| 17| 1F| | 27| 2F| 37| 3F| | 47| 4F| 56| 5E| | 57| 5F| 62|
`---'   `---------------' `---------------' `---------------' `-----------'
,-----------------------------------------------------------. ,-----------. ,---------------.
| 0E| 16| 1E| 26| 25| 2E| 36| 3D| 3E| 46| 45| 4E| 55| 5D| 66| | 67| 6E| 6F| | 76| 77| 7E| 84|
|-----------------------------------------------------------| |-----------| |---------------|
| 0D  | 15| 1D| 24| 2D| 2C| 35| 3C| 43| 44| 4D| 54| 5B|  5C | | 64| 65| 6D| | 6C| 75| 7D| 7C|
|-----------------------------------------------------------| `-----------' |---------------|
| 14   | 1C| 1B| 23| 2B| 34| 33| 3B| 42| 4B| 4C| 52| 53| 5A |               | 6B| 73| 74| 7B|
|-----------------------------------------------------------|     ,---.     |---------------|
| 12 | 13| 1A| 22| 21| 2A| 32| 31| 3A| 41| 49| 4A| 51|  59  |     | 63|     | 69| 72| 7A| 79|
|-----------------------------------------------------------| ,-----------. |---------------|
| 11  |   |19  |        29                   |39  |   | 58  | | 61| 60| 6A| | 68| 70| 71| 78|
`-----'   `---------------------------------------'   `-----' `-----------' `---------------'
```

https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-Keyboard-Converter#ibm-terminal-101-key102-key-and-ibm-rt