tmk/tmk_keyboard #739: Commodore/XT-USB stops responding randomly

manudergrohse, 2022-05-12:
--all keys respond for 1-30(or more) keypresses
--a completely unrelated key gets activated (even keys that aren't on the keyboard like normal arrows)
--the keyboard is unresponsive for about five seconds
--repeat
PC has been restarted, controller plugged in before restart, keyboard plugged in to controller before controller in to PC

manudergrohse, 2022-05-12:
```
Device disconnected.
Waiting for new device:
Listening:

Device disconnected.
Waiting for new device:
Listening:

Device disconnected.
Waiting for new device:
Listening:

Device disconnected.
Waiting for new device:
Listening:

Device disconnected.
Waiting for new device:
Listening:

Device disconnected.
Waiting for new device:
Listening:

Device disconnected.
Waiting for new device:HID/win32: SetupDiGetClassDevs failed.HID/win32: SetupDiGetClassDevs failed.HID/win32: SetupDiGetClassDevs failed.HID/win32: SetupDiGetClassDevs failed.HID/win32: SetupDiGetClassDevs failed.HID/win32: SetupDiGetClassDevs failed.HID/win32: SetupDiGetClassDevs failed.
```

tmk, 2022-05-13:
hid_listen doesn't seem to work for some reason.
what's OS are you using? 32bit or 64bit?

- Try other AT(PS/2) keyboard to see if it works.
- Try other Windows computers, Mac or Linux if possible.

manudergrohse, 2022-05-13:
I have tried another PS2 keyboard but that one might not work at all. I will have another couple of G80s come in soon that I can test this on.
The same issue happens on another windows PC if I can find the adapter, I will try it on an android phone and on my friends macbook

tmk, 2022-05-13:
Also make sure you are using default firmware. You can download it here.
https://github.com/tmk/tmk_keyboard/tree/master/converter/ibmpc_usb/binary

You will see message like below when the latest version of default firmware.
```
Waiting for new device:...............................................................................................
Listening:

TMK:159cc2/LUFA:d6a7df

USB configured.

Loop start.
```

manudergrohse, 2022-05-13:
I didn't see this firmware file before. Thanks for the help. the keyboard works perfectly now. You can add the Commodore to your list.

manudergrohse, 2022-05-13:
r1E r9E r20 rA0 r39 rB9 r1E r9E r1F r39 r9F rB9 r26 rA6 r18 r98 r31 rB1 r22 rA2 r39 r1E rB9 r9E r1F r9F r39 rB9 r17 r97 r14 r94 r39 rB9 r17 r97 r1F r9F r39 rB9 r30 rB0 r12 r17 r92 r97 r31 r22 rB1 rA2 r39 rB9 r2E r26 rAE rA6 r17 r97 r2E r25 rAE rA5 r12 r92 r20 rA0 r39 rB9 r18 r98 r21 rA1 r21 r39 rA1 rB9 r30 rB0 r2C r39 rAC rB9 r26 r17 rA6 r97 r14 r94 r14 r94 r26 r12 rA6 r92 r39 rB9 r11 r23 r91 rA3 r12 r92 r12 r92 r26 rA6 r1F r9F r2A r33 rB3 rAA r39 rB9 r18 r98 r31 rB1 r26 rA6 r2C r39 rAC rB9 r11 r23 r91 rA3 r12 r31 r92 rB1 r39 rB9 r14 r23 r94 rA3 r12 r39 r92 rB9 r26 rA6 r0E r8E r2E rAE r26 rA6 r18 r98 r2E r25 rAE rA5 r39 rB9 r1F r9F r14 r18 r94 r98 r19 r99 r1F r39 r9F rB9 r20 r18 rA0 r98 r12 r92 r1F r9F r39 rB9 r14 r17 r94 r97 r32 rB2 r12 r92 r39 rB9 r2E r18 rAE r98 r32 r12 rB2 r92 r39 rB9 r14 r18 r94 r98 r39 rB9 r26 r17 rA6 r21 r97 r12 rA1 r92 r34 rB4 r0F r8F r1C r9C r36 r1F r9F rB6 r1E r9E r2C r33 rAC rB3 r39 rB9 r2C r18 rAC r16 r98 r96 r2A r2B rAA rAB r13 r12 r93 r92 r39 r1E rB9 r39 r9E rB9 r26 rA6 r18 r98 r14 r94 r39 rB9 r1F r9F r32 rB2 r3C
1512 ERR:20 ISR:4000 E1513 I1513
1565 PRT:21 ISR:9EA0 r9E rA6 r26 rA6 r39 rB9 A4515 wFF R R R R R R R R R R
5234 ERR:11 ISR:8000 X5735 rFFF(XT)
29813 ERR:13 ISR:8000 S29814 L29814 r20 r1F rA1 r1E rA0 r9E r9F r21 r20 r1F rA1 r1E rA0 r9F r9E r1C r9C r21 rA1 r21 r20 rA1 r1F rA0 r9F r21 r20 rA1 r1F rA0 r9F r21 r20 r1F rA1 rA0 r9F r1F r20 r21 r9F rA0 rA1 r21 r20 r1F rA1 rA0 r9F

manudergrohse, 2022-05-14:
```
Device disconnected.
Waiting for new device:....
Listening:

TMK:159cc2/LUFA:d6a7df

USB configured.

Loop start.
I317 I318
319 PRT:21 ISR:AAA0 rAA A3319 wFF R R R R R R R R R R A4037 wFF R R R R R R R R R R
4756 ERR:11 ISR:8000
4757 ERR:11 ISR:8000 X5258 X5259 r20 W11062 rA0 W11144 r17 W11308 wF2 R R11392
ID:FFFF(XT)
11393 ERR:13 ISR:97A0 S11394 L11394 r97 r1F r9F r0E r8E r0E r8E r0E r8E r1F r9F r0E r8E r20 rA0 r17 r97 r1F r9F r2E rAE r18 r98 r13 r93 r20 rA0 r1C r9C r23 rA3 r1E r9E r2F rAF r12 r39 r92 rB9 r1F r16 r9F r96 r2E r23 rAE rA3 r39 r32 rB9 rB2 r16 r96 r2E r23 rAE rA3 r39 rB9 r30 rB0 r2C r39 rB9 rAC r18 r98 r31 rB1 r12 r39 r92 rB9 r22 rA2 r12 r92 r31 rB1 r12 r13 r1E r93 r92 r26 r9E rA6 r39 rB9 r30 rB0 r12 r92 r22 rA2 r17 r31 r97 rB1 r39 rB9 r19 r99 r26 r1E rA6 r9E r39 rB9 r46 rC6 r0E r46 rC6 r8E r0E r8E r1E r9E r2E rAE r12 r92 r39 rB9 r11 r91 r17 r97 r14 r23 r94 rA3 r39 rB9 r19 r99 r18 r98 r1F r9F r1F r9F r17 r97 r30 r26 rB0 rA6 r12 r39 r92 rB9 r11 r91 r23 rA3 r18 r98 r39 rB9 r1E r9E r14 r39 r94 rB9 r22 rA2 r13 r18 r16 r93 r19 r96 r98 r99 r39 rB9 r32 rB2 r1E r31 r9E rB1 r39 rB9 r14 r94 r23 rA3 r12 r39 r92 rB9 r1F r9F r23 rA3 r18 r98 r11 r91 r39 rB9 r1F r9F r12 r92 r12 r32 r92 rB2 r39 rB9 r23 rA3 r18 r16 r98 r1F r96 r12 r9F r92 r39 rB9 r12 r31 r92 rB1 r20 r39 rA0 rB9 r0F r1C r9C r8F r1F r16 r9F r96 r2E r23 rA3 rAE r39 rB9 r18 r98 r21 rA1 r21 r39 rA1 rB9 r19 r99 r26 r1E rA6 r9E r2C r39 rAC r39
62790 ERR:20 ISR:C000 E62790 I62792
62840 PRT:21 ISR:A3A0 rA3
62901 ERR:20 ISR:4000 A257 wFF
267 ERR:17 ISR:8000 X769 r21 W1675 r24 W1733 r25 W1884 wF2 R1890
ID:FFFF(XT)
1891 ERR:13 ISR:8000 S1891 L1892 r20 r26 r1F rA4 rA1 r1E rA5 rA0 rA6 r9F r9E r39 rB9 r20 rA0 r12 r92 r2F rAF r12 r92 r26 rA6 r18 r19 r98 r99 r39 rB9 r1F r9F r14 r17 r94 r97 r26 rA6 r26 rA6 r39 rB9 r22 r18 rA2 r98 r39 rB9 r23 r17 rA3 r97 r22 r23 rA2 rA3 r39 rB9 r32 rB2 r1E r25 r9E rA5 r12 r39 r92 r17 rB9 r21 r97 rA1 r39 rB9 r11 r23 r91 r17 rA3 r97 r2E r23 rAE rA3 r39 r23 rB9 rA3 r18 r16 r98 r96 r1F r12 r9F r92 r39 rB9 r12 r92 r2C r12 rAC r39 r92 rB9 r32 rB2 r17 r22 r97 rA2 r23 r14 rA3 r94 r39 rB9 r18 r98 r11 r31 r91 rB1 r39 rB9 r1F r9F r1E r9E r0F r1C r9C r8F r2E r23 rAE r17 r26 rA3 r97 rA6 r20 r39 rA0 r17 rB9 r97 r31 rB1 r39 r32 rB9 rB2 r1E r31 r9E rB1 r39 r1F rB9 r9F r14 r17 r94 r97 r26 rA6 r26 r39 rA6 rB9 r17 r97 r31 r2E rB1 rAE r13 r12 r1E r93 r92 r9E r1F r12 r39 r9F r92 rB9 r14 r23 r17 r94 rA3 r97 r31 rB1 r25 rA5 r39 rB9 r14 r94 r0E r8E r11 r91 r23 rA3 r1E r14 r9E r39 r94 rB9 r14 r18 r94 r98 r18 r98 r39 r1F rB9 r9F r14 r1E r94 r14 r9E r12 r94 r39 r92 r18 rB9 r98 r11 r31 r91 rB1 r39 rB9 r2E r18 rAE r16 r98 r96 r26 r20 rA6 r39 rA0 rB9 r1E r31 r9E rB1 r2C r39 rAC rB9 r32 rB2 r16 r96 r2E rAE r23 rA3 r39 r17 rB9 r31 r97 r14 rB1 r94 r12 r13 r92 r12 r93 r1F r14 r92 r9F r94 r39 rB9 r30 rB0 r1E r9E r2E r25 rAE rA5 r39 rB9 r21 r18 rA1 r98 r13 r93 r39 rB9 r1E r9E r31 rB1 r2C r39 rAC rB9 r1E r9E r1F r25 r9F rA5 r39 rB9 r11 r91 r13 r17 r93 r97 r14 r12 r94 r39 rB9 r92 r23 rA3 r12 r92 r1E r9E r20 rA0 r39 rB9 r1E r31 r9E rB1 r18 r14 r23 r98 r94 rA3 r12 r13 r92 r39 r93 rB9 r14 r23 r94 rA3 r1E r31 r9E rB1 r39 rB9 r26 r12 rA6 r1E r92 r9E r20 r39 rA0 rB9 r1E r31 r9E rB1 r14 r17 r94 r18 r97 r31 r98 rB1 r0F r1C r37 r8F r9C rB7 r0F r1C r9C r8F r1F r9F r12 r92 r12 r32 r92 rB2 r39 r31 rB9 r18 rB1 r98 r11 r91 r39 rB9 r14 r23 r94 r12 rA3 r92 r1F r12 r9F r39 r92 rB9 r23 rA3 r12 r92 r39 rB9 r11 r23 r91 r17 r26 rA3 r12 r97 rA6 r92 r39 rB9 r11 r23 r91 rA3 r12 r31 r92 rB1 r39 rB9 r18 r98 r2F rAF r12 r13 r39 r92 r93 rB9 r30 rB0 r12 r92 r2E rAE r1E r16 r9E r96 r1F r12 r9F r39 r92 rB9 r11 r23 r91 rA3 r12 r31 r92 rB1 r39 r26 rB9 rA6 r1E r13 r9E r93 r22 r12 rA2 r39 r92 rB9 r30 r16 rB0 r96 r14 r39 r94 rB9 r23 r18 rA3 r98 r32 rB2 r12 r92 r39 rB9 r22 r18 rA2 r98 r2F rAF r12 r13 r92 r31 r93 rB1 r39 r32 rB9 rB2 r16 r96 r2E rAE r23 rA3 r39 rB9 r17 r97 r31 r14 rB1 r94 r12 r92 r13 r12 r93 r1F r14 r92 r9F r94 r39 rB9 r18 r98 r21 rA1 r21 r39 rA1 r26 rB9 r17 rA6 r97 r14 r94 r14 r94 r26 rA6 r12 r92 r39 rB9 r19 r99 r26 rA6 r1E r31 r9E rB1 r39 r1F rB9 r9F r1E r9E r2C r39 rAC rB9 r2E rAE r1E r26 r9E rA6 r26 rA6 r39 rB9 r14 r23 r94 r17 rA3 r97 r31 rB1 r22 r39 rA2 rB9 r19 r16 r99 r96 r26 r30 rB0 rA6 r0F r1C r8F r9C r14 r23 r94 r17 rA3 r97 r31 r25 rB1 rA5 r39 rB9 r18 r98 r13 r93 r39 rB9 r11 r23 r91 r17 rA3 r26 r97 rA6 r12 r39 r92 rB9 r31 rB1 r18 r98 r39 rB9 r22 r18 rA2 r98 r2F rAF r12 r13 r31 r92 rB1 r93 r39 rB9 r1F r9F r12 r92 r12 r92 r32 rB2 r39 rB9 r32 rB2 r12 r92 r1E r31 r9E rB1 r39 rB9 r1E r9E r21 rA1 r14 r94 r12 r13 r92 r39 r93 rB9 r2E r18 rAE r16 r98 r96 r13 r12 r1F r93 r92 r9F r39 rB9 r31 rB1 r12 r92 r2F rAF r12 r92 r13 r39 r93 rB9 r17 r97 r14 r94 r39 rB9 r21 rA1 r1E r9E r2E rAE r12 r92 r39 rB9 r0E r8E r0F r1C r37 rB7 r8F r9C r1F r9F r12 r92 r14 r39 r94 rB9 r18 r98 r19 r99 r12 r92 r31 rB1 r39 rB9 r11 r18 r98 r91 r13 r25 r93 rA5 r39 rB9 r1F r23 r9F rA3 r12 r92 r39 rB9 r11 r18 r91 r16 r98 r96 r26 r20 rA6 rA0 r39 rB9 r17 r97 r31 rB1 r14 r18 r94 r98 r39 r21 rB9 rA1 r18 r98 r13 r39 r93 rB9 r11 r23 r91 rA3 r12 r13 r92 r12 r93 r39 r92 rB9 r1F r9F r12 r92 r12 r92 r32 rB2 r39 rB9 r26 r12 rA6 r1E r92 r9E r2F r12 rAF r39 r92 rB9 r32 rB2 r1E r31 r9E rB1 r39 rB9 r19 r99 r26 r1E rA6 r9E r2E rAE r12 r92 r39 rB9 r14 r17 r94 r97 r32 r12 rB2 r92 r39 rB9 r30 rB0 r12 r14 r92 r11 r94 r91 r12 r92 r12 r92 r12 r31 r92 rB1 r39 rB9 r21 rA1 r17 r97 r13 r93 r0F r1C r9C r8F r19 r99 r13 r93 r18 r98 r22 rA2 r13 r1E r93 r32 r9E rB2 r39 rB9 r26 r12 rA6 r92 r1E r9E r2F r12 rAF r39 r92 rB9 r23 rA3 r12 r92 r [... cut ...]
9700 PRT:21 ISR:39A0 r39 rB9 r0F r1C r8F r9C r0F r1C r8F r9C r21 rA1 r21 r24 rA1 rA4 r1F r25 r26 rA5 rA6 r9F A12594 wFF R
12710 ERR:17 ISR:8000 X13212 r21 W13561 rA1 W13708 r24 W13778 wF2 R13822
ID:FFFF(XT)
13823 ERR:13 ISR:8000 S13823 L13824 r21 rA4 rA1 r39 rB9 r12 r92 r2C rAC r12 r92 r39 rB9 r23 rA3 r18 r98 r26 r20 rA6 rA0 r39 rB9 r2E r18 rAE r16 r98 r96 r13 r12 r93 r1F r92 r9F r0F r1C r8F r9C
```
I tried it on a MacBook too and it happened every 1 to 5 keystrokes. I will try replacing the capacitors.

manudergrohse, 2022-05-14:
I have found yet another micro USB cable and so far couldn't repeat the issue. Might have been that all along.

closed 2022-05-14 by manudergrohse

aeglon, 2023-09-23:
```

HID console connected: TMK IBM PC keyboard converter (FEED:1BEE:0201)
> 
> TMK:1c528a/LUFA:d6a7df
> 
> USB configured.
> 
> Loop start.
> I2500 I2500 
> 2502 PRT:21 ISR:AAA0 rAA A5501 wFF R R R R R R R R R R A6209 wFF R R R R R R R R R R 
> 6918 ERR:11 ISR:8000 
> 6919 ERR:11 ISR:8000 X7419 X7420 r39 W18597 rB9 W18826 
> 19028 PRT:22 ISR:2FC0 r2F W19029 R19029 
> ID:FFFB(XT_CLONE) S19030 L19030 
> 19191 ERR:20 ISR:4000 E19192 
> [CLR] I19192 
> 19216 PRT:21 ISR:13A0 r13 r93 A22193 wFF R R R R R R R R R R 
> 22901 ERR:11 ISR:8000 X23403 r39 W25599 rB9 W25811 r39 W26185 R26186 
HID console disconnected: TMK IBM PC keyboard converter (FEED:1BEE:0201)
HID console connected: TMK IBM PC keyboard converter (FEED:1BEE:0201)
> 
> TMK:1c528a/LUFA:d6a7df
> 
> USB configured.
> 
> Loop start.
> I2500 I2500 
> 2506 PRT:21 ISR:AAA0 rAA A5501 wFF R R R R R R R R R R A6209 wFF R R R R R R R R R R 
> 6918 ERR:11 ISR:8000 
> 6919 ERR:11 ISR:8000 X7420 X7420 r25 W12234 r26 W12235 r1E W12239 R12239 
> ID:FFFC(XT_IBM) S12240 L12240 r1F r20 r24 r9E r9F rA5 rA6 r1E rA0 rA4 r1F r20 r9E r9F rA0 r24 r1E rA4 r9E r24 r1E r1F rA4 r20 r9E r9F r39 rA0 rB9 r1E r9E r31 r14 r12 rB1 r94 r92 r39 r1E r1F rB9 r20 r9E r9F r17 rA0 r97 r39 r14 rB9 r94 r23 r12 rA3 r92 r39 r1E rB9 r1F r20 r9E r31 r9F rA0 rB1 r22 r12 r39 rB9 rA2 r26 r92 rA6 r1E r1F r20 r9E r9F r18 rA0 r98 r39 r11 rB9 r12 r91 r39 r92 r1E rB9 r1F r9E r20 r9F r31 rA0 rB1 r12 r22 r92 r26 r12 rA6 rA2 r1E r18 r92 r1F r98 r20 r9E r9F r39 rA0 r10 rB9 r12 r90 r39 
> 18285 PRT:22 ISR:23C0 r23 
> 18307 ERR:20 ISR:4000 E18307 
> [CLR] I18310 
> 18313 PRT:21 ISR:B9A0 rB9 r91 r10 r11 r12 r90 r91 r16 r17 r92 r1E r1F r96 r97 r20 r9E r9F rA0 r39 rB9 A21312 wFF R R R R R R R R R R 
> 22020 ERR:11 ISR:8000 X22522 r39 W24700 r1E W24773 r1F W24828 R24828 
> ID:FFFC(XT_IBM) S24829 L24829 rB9 r20 r9E r9F r39 r1E r1F rB9 rA0 r9E r20 r31 r9F rA0 rB1 r39 r22 rB9 r12 rA2 r1E r1F r39 r92 rB9 r20 r9E r9F r39 r1E r1F rA0 r24 rB9 r20 r9E rA4 r39 r9F r18 rA0 rB9 r98 r39 r1E r1F rB9 r9E r39 r9F r14 rB9 r12 r23 r94 r92 rA3 r39 r1E r22 rB9 r9E r1E rA2 r32 r1F rB2 r20 r9E r9F r17 rA0 r39 r97 r11 r10 rB9 r90 r04 r05 r91 r0A r09 r85 r84 r03 r8A r0B r89 r04 r8B r83 r03 r84 r04 r83 r03 r84 r04 r83 r84 r1E r1F r20 r9E r39 r9F r1E rB9 r1F rA0 r31 r9E r9F rB1 r22 r12 r26 rA2 r92 rA6 r18 r98 r39 r17 rB9 r1F r97 r39 r9F rB9 r22 r31 r12 rB1 rA2 r1E r92 r26 rA6 r9E r39 r17 rB9 r97 r39 r1F 
> 32369 PRT:22 ISR:73C0 r73 
> 32447 ERR:20 ISR:C000 E32447 
> [CLR] I32450 
> 32528 PRT:21 ISR:23A0 r23 r14 r12 rA3 r94 r1E r39 r92 rB9 r1F r20 r9E r9F r31 rB1 rA0 r39 rB9 A35452 wFF R R R R R R R R R R 
> 36160 ERR:11 ISR:8000 X36662 r1E W37753 r9E W37884 r31 W37894 R37895 
> ID:FFFC(XT_IBM) S37895 L37896 r22 rB1 r12 rA2 r92 r26 rA6 r18 r98 r39 r1F rB9 r17 r9F r97 r39 r1F rB9 r17 r9F r97 r39 r14 rB9 r23 r94 r12 rA3 r39 r92 r22 rB9 r1E rA2 r31 
> 39798 PRT:22 ISR:3DC0 r3D 
> 39841 ERR:20 ISR:C000 E39842 
> [CLR] I39844 
> 39880 PRT:21 ISR:1EA0 r1E r1F r26 r20 r9E rA6 r9F 
> 40193 PRT:22 ISR:31C0 r31 
> 40198 ERR:20 ISR:4000 E40199 
> [CLR] I40199 
> 40264 PRT:21 ISR:1EA0 r1E r98 r39 r9E rB9 r22 r31 r12 rB1 rA2 r92 r1E r1F r20 r9E r9F rA0 r39 rB9 r1E r1F r31 r20 r9E rB1 r9F rA0 r1E r1F r31 r9E r20 rB1 r1E rA0 r20 r9E r9F rA0 r31 A43200 wFF 
> 43225 ERR:17 ISR:8000 X43726 r1E W43938 r9E W44004 r31 W44030 R44030 
> ID:FFFC(XT_IBM) S44031 L44031 r22 rB1 r12 rA2 r26 r92 rA6 r1E r1F r20 r9E r9F r39 rA0 r1E rB9 r9E r31 r22 rB1 r12 rA2 r26 r92 rA6 r18 r98 r39 r1F rB9 r17 r9F r39 r97 rB9 r14 r12 r94 r39 r92 r31 r1E rB9 rB1 r1F r9E r39 r9F r17 rB9 r97 r39 rB9 r14 r12 r23 r94 rA3 r92 r39 r1E r1F rB9 r20 r9E r31 r9F rA0 r1E rB1 r39 r9E r22 rB9 r12 rA2 r1E r92 r1F r20 r9E r39 r9F r1E rA0 r1F rB9 r20 r9E r9F r23 r1E rA0 r1F r24 rA4 rA3 r20 r23 r9E r9F rA3 rA0 r1F r1E r20 r9E r9F r1E rA0 r1F r20 r9E r39 r9F r1E r1F rA0 r31 rB9 r9E r9F rB1 r26 r10 r25 r11 r12 r24 rA5 rA6 r90 r91 r26 r25 rA4 r11 r92 r10 r24 
> 4
> 49856 ERR:20 ISR:C000 E49857 
> [CLR] I
> 41 ISR:AAA0 rAA r2 r1 r12 r10 r90 26rr25 rA4 r11 r92 r10 r4 1 r12 r6 r5 r90 225 r10 r11 rA4 r92 r24 r12 rA6 rA5 r90 r91 r92 rA4 A52864 wFF R R R R R R R R R R 
> 53572 ERR:11 ISR:8000 X54074 r39 W59051 rB9 W59231 r1E W59409 R59410 
> ID:FFFC(XT_IBM) S59410 L59411 r1F r20 r9E r9F r39 rA0 r1F r1E rB9 r20 r9E r24 r9F rA0 r1F r1E rA4 r20 r25 r9E r9F rA0 
> 60599 PRT:22 ISR:49C0 r49 
> 60668 ERR:20 ISR:4000 E60668 
> [CLR] I60670 
> 60731 PRT:21 ISR:A4A0 rA4 A63672 wFF R R R R R R R R R R 
> 64380 ERR:11 ISR:8000 X64882 r39 W7382 rB9 W7562 r24 W7764 R7764 
> ID:FFFC(XT_IBM) S7765 L7765 r1E r1F r26 rA4 r25 r20 r24 r9E r9F rA5 rA6 r1E rA0 r1F rA4 r20 r9E r9F r39 rA0 rB9 r1E r9E r31 r1F r1E rB1 r20 r9E r9F r39 r1E r1F rA0 r20 rB9 r9E r9F r31 
> 9516 PRT:22 ISR:3DC0 r3D 
> 9522 ERR:20 ISR:4000 E9522 
> [CLR] I9525 
> 9552 PRT:21 ISR:A0A0 rA0 rB1 r39 r20 r9E r9F rB9 rA0 r31 r22 rB1 r12 rA2 r92 r26 r1E r1D rA6 r1F r20 r9D r9E r39 r9F rA0 rB9 A12527 wFF R R R R R R R R R R 
[... 20 lines cut ...]
> 27167 ERR:20 ISR:C000 E27168 
> [CLR] I27170 
> 27664 PRT:21 ISR:33A0 r33 rB3 r39 r2B rB9 r17 r97 rAB A30172 wFF R R R R R R R 
> 30631 ERR:17 ISR:8000 X31133 r13 W35002 r93 W35231 R35732 
> ID:FFFC(XT_IBM) S35732 L35732 r13 r93 r39 rB9 r33 rB3 r39 rB9 r17 r97 r39 rB9 r16 r1F r96 r9F r12 r92 r39 rB9 r19 r13 r99 r18 r93 r98 r39 r32 rB9 rB2 r17 r2E r97 rAE r13 r18 r93 r98 r39 rB9 r2E rAE r26 rA6 r18 r98 r31 rB1 r12 r92 r39 rB9 r14 r94 
> 49458 PRT:22 ISR:47C0 r47 
> 49523 ERR:20 ISR:4000 E49524 
> [CLR] I49525 
[... cut ...]
> ID:FFFC(XT_IBM) S34017 L34018 r84 r09 r89 r09 r89 r04 r84 r39 rB9 r25 rA5 r12 r15 r92 r95 r39 rB9 r2D rAD r14 r94 r39 rB9 r2A rAA r2A rAA r0A r2A r8A rAA r2A r0A r8A r0A r8A rAA r2A rAA r2A r0A r8A rAA r30 r27 
> 55411 PRT:22 ISR:27C0 
> 55447 ERR:20 ISR:4000 E55447 
> [CLR] I55449 
> 55451 PRT:21 ISR:B0A0 rB0 r1E r93 r9E r92 A58450 wFF 
> 58504 ERR:17 ISR:8000 X59005 r32 W11531 rB2 W11630 r32 W12067 R12067 

```
I have similar issue. However , i use pro micro clone without j1. I must try this first when I can get proper pro micro. I am also unsure how to remap the keys in different position than 83 key xt (minus,equal,bkspce)