tmk/tmk_keyboard #741: Z150 AT, Compiled Hex will not work

Alectardy98, 2022-06-05:
I have tried to compile a hex from the source code for the zenith Z150 AT and it does not work, and when the keyboard is plugged in, it repeats this code.

This only happens when I try to compile, the online configuration tool works with no problems

Here is the referenced code

Loop start.
    I173 I173 I174 I174 I174 I174 I174 I174 I175 I175 I175 I175 I176 I176 I176 I176 I176 I177 I177 I177 I177 I177 I178 I178 I178 I178 I178 I179 I179 I179 I180 I180 I180 I180 I180 I181 I181 I181 I181 I182 I182 I182 I183 I183 I183 I184 I18I193 I193 I193 I193 I193 I194 I194 I194 I194 I195 I195 I195 I195 I196 I196 I196 I196 I196 I197 I197 I197 I197 I197 I198 I198 I198 I198 I199 I199 I199 I199 I199 I200 I200 I200 I200 I200 I201 I201 I201 I201 I202 I202 I202 I202 I202 I203 I203 I203 I203 I203 I204 I204 I204 I204 I204 I205 I205 I205 I205 I206 I206 I206 I206 I206 I207 I207 I207 I207 I207 I208 I208 I208 I208 I209 I209 I209 I209 I210 I210 I210 I210 I210 I210 I211 I211 I211 I211 I212 I212 I212 I212 I212 I213 I213 I213 I213 I214 I214 I214 I214 I214 I215 I215 I215 I215 I215 I216 I216 I216 I216 I217 I217 I217 I217 I217 I218 I218 I218 I218 I218 I219 I219 I219 I219 I220 I220 I220 I220 I220 I221 I221 I221 I221 I221 I222 I222 I222 I222 I223 I223 I223 I223 I223 I224 I224 I224 I224 I225 I225 I225 I225 I225 I226 I226 I226 I226 I226 I227 I227 I227 I227 I228 I228 I228 I228 I228 I229 I229 I229 I229 I229 I230 I230 I230 I230 I231 I231 I231 I231 I231 I232 I232 I232 I232 I232 I233 I233 I233 I233 I234 I234 I234 I234 I234 I235 I235 I235 I235 I235 I236 I236 I236 I236 I237 I237 I237 I237 I237 I238 I238 I238 I238 I238 I239 I239 I239 I239 I240 I240 I240 I240 I240 I241 I241 I241 I241 I241 I242 I242 I242 I242 I243 I243 I243 I243 I243 I244 I244 I244 I244 I244 I245 I245 I245 I245 I246 I246 I246 I246 I246 I247 I247 I247 I247 I247 I248 I248 I248 I248 I249 I249 I249 I249 I249 I250 I250 I250 I250 I250 I251 I251 I251 I251 I252 I252 I252 I252 I252 I253 I253 I253 I253 I253 I254 I254 I254 I254 I255 I255 I255 I255 I255 I256 I256 I256 I256 I256 I257 I257 I257 I257 I258 I258 I258 I258 I258 I259 I259 I259 I259 I259 I260 I260 I260 I260 I261 I261 I261 I261 I261 I262 I262 I262 I262 I262 I263 I263 I263 I263 I264 I264 I264 I264 I264 I265 I265 I265 I265 I265 I266 I266 I266 I266 I267 I267 I267 I267 I267 I268 I268 I268 I268 I269 I269 I269 I269 I269 I270 I270 I270 I270 I270 I271 I271 I271 I271 I272 I272 I272 I272 I272 I273 I273 I273 I273 I273 I274 I274 I274 I274 I275 I275 I275 I275 I275 I276 I276 I276 I276 I276 I277 I277 I277 I277 I277 I278 I278 I278 I278 I279 I279 I279 I279 I280 I280 I280 I280 I280 I280 I281 I281 I281 I281 I282 I282 I282 I282 I282 I283 I283 I283 I283 I284 I284 I284 I284 I284 I285 I285 I285 I285 I285 I286 I286 I286 I286 I287 I287 I287 I287 I287 I288 I288 I288 I288 I288 I289 I289 I289 I289 I290 I290 I290 I290 I290 I291 I291 I291 I291 I291 I292 I292 I292 I292 I293 I293 I293 I293 I293 I294 I294 I294 I294 I295 I295 I295 I295 I295 I296 I296 I296 I296 I296 I297 I297 I297 I297 I298 I298 I298 I298 I298 I299 I299 I299 I299 I299 I300 I300 I300 I300 I301 I301 I301 I301 I301 I302 I302 I302 I302 I302 I303 I303 I303 I303 I304 I304 I304 I304 I304 I305 I305 I305 I305 I305 I306 I306 I306 I306 I307 I307 I307 I307 I307 I308 I308 I308 I308 I308 I309 I309 I309 I309 I310 I310 I310 I310 I310 I311 I311 I311 I311 I311 I312 I312 I312 I312 I313 I313 I313 I313 I313 I314 I314 I314 I314 I314 I315 I315 I315 I315 I316 I316 I316 I316 I316 I317 I317 I317 I317 I317 I318 I318 I318 I318 I319 I319 I319 I319 I319 I320 I320 I320 I320 I320 I321 I321 I321 I321 I322 I322 I322 I322 I322 I323 I323 I323 I323 I323 I324 I324 I324 I324 I325 I325 I325 I325 I325 I326 I326 I326 I326 I326 I327 I327 I327 I327 I328 I328 I328 I328 I328 I329 I329 I329 I329 I329 I330 I330 I330 I330 I331 I331 I331 I331 I331 I332 I332 I332 I332 I332 I333 I333 I333 I333 I334 I334 I334 I334 I334 I335 I335 I335 I335 I335 I336 I336 I336 I336 I337 I337 I337 I337 I337 I338 I338 I338 I338 I339 I339 I339 I339 I339 I340 I340 I340 I340 I340 I341 I341 I341 I341 I342 I342 I342 I342 I342 I343 I343 I343 I343 I343 I344 I344 I344 I344 I345 I345 I345 I345 I345 I346 I346 I346 I346 I346 I347 I347 I347 I347 I348 I348 I348 I348 I348 I349 I349 I349 I349 I350 I350 I350 I350 I350 I351 I351 I351 I351 I351 I352 I352 I352 I352 I353 I353 I353 I353 I353 I354 I354 I354 I354 I354 I355 I355 I355 I355 I356 I356 I356 I356 I356 I357 I357 I357 
    357 PRT:11 ISR:6A90 I358 I358 
    358 PRT:11 ISR:0000 I359 I359 
    359 PRT:11 ISR:0000 I360 I360 
    360 PRT:11 ISR:0000 I361 I361 
    361 PRT:11 ISR:0000 I362 I362 
    362 PRT:11 ISR:0000 I363 I363 
    363 PRT:11 ISR:0000 I364 I364 
    364 PRT:11 ISR:0000 I365 I365 
    365 PRT:11 ISR:0000 I365 I366 
    366 PRT:11 ISR:0000 I366 I367 
    367 PRT:11 ISR:0000 I367 I368 
    368 PRT:11 ISR:0000 I368 I368 
    369 PRT:11 ISR:0000 I369 I369 
    370 PRT:11 ISR:0000 I370 I370 
    371 PRT:11 ISR:0000 I371 I371 
    372 PRT:11 ISR:0000 I372 I372 
    373 PRT:11 ISR:0000 I373 I373 
    373 PRT:11 ISR:0000 I374 I374 
    374 PRT:11 ISR:0000 I375 I375 
    375 PRT:11 ISR:0000 I376 I376 
    376 PRT:11 ISR:0000 I377 I377 
    377 PRT:11 ISR:0000 I378 I378 
  > 378 PRT:11 ISR:0000 I379 I379 
    379 PRT:11 ISR:0000 I380 I380 
    380 PRT:11 ISR:0000 I381 I381 
    381 PRT:11 ISR:0000 I381 I382 
    382 PRT:11 ISR:0000 I382 I383 
    383 PRT:11 ISR:0000 I383 I384 
    384 PRT:11 ISR:0000 I384 I385 
    385 PRT:11 ISR:0000 I385 I385 
    386 PRT:11 ISR:0000 I386 I386 
    387 PRT:11 ISR:0000 I387 I387 

T

tmk, 2022-06-05:
What's your gcc version?
What if checkout 27e6c8d(which is used with online editor) and buid firmware?

Alectardy98, 2022-06-05:
I am currently using AVR GCC 9.3.0, on mac OS 12.4 and I'm using the most up to date build of TMK

`~ % avr-gcc  --version
avr-gcc (Homebrew AVR GCC 9.3.0_3) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.`

Alectardy98, 2022-06-05:
I checked out  27e6c8d and still was unable to compile, it must have to do with the AVR GCC Version

Alectardy98, 2022-06-05:
i also have just tested on manjaro Linux and I am getting the same issue

tmk, 2022-06-06:
which is gcc version of manjaro Linux?

Alectardy98, 2022-06-06:
no, I didn't see any errors, but it was the most up-to-date version. It was 12 something.

Alectardy98, 2022-06-06:
I have a friend who has ubuntu, and the most up to date version of TMK. He has avr-gcc 5.4.0 and the hex he made is working. I think its an issue with avr-gcc

tmk, 2022-06-06:
I will have to check GCC release notes about AVR closely. Something changed between 5.4 and 9.3 and it exposes bug or incompatible of my code probably.

https://gcc.gnu.org/releases.html

tmk, 2022-06-06:
I confirmed the problem on Archlinux with avr-gcc 12.1.0.

At this time workaround seems to be removing `-ffunction-sections` option.
I'm not sure how this option causes inside.

```
diff --git a/tmk_core/rules.mk b/tmk_core/rules.mk
index b559e57b..a3e46442 100644
--- a/tmk_core/rules.mk
+++ b/tmk_core/rules.mk
@@ -159,7 +159,7 @@ CPPFLAGS += -funsigned-bitfields
 CPPFLAGS += -fpack-struct
 CPPFLAGS += -fshort-enums
 CPPFLAGS += -fno-exceptions
-CPPFLAGS += -ffunction-sections
+#CPPFLAGS += -ffunction-sections
 CPPFLAGS += -fdata-sections
 # to supress "warning: only initialized variables can be placed into program memory area"
 CPPFLAGS += -w
```

an-achronism, 2023-03-03:
> I confirmed the problem on Archlinux with avr-gcc 12.1.0.
>
> At this time workaround seems to be removing `-ffunction-sections` option. I'm not sure how this option causes inside.
>
> ```
> diff --git [...]

I appear to be having the same issue, or at least a similar one, over here on GCC v9.4.0.

How exactly would I remove that section? I'm assuming I can just edit the makefile, but I'm not sure precisely the syntax that I would need for that to work.

Worth noting that the files generated if I run a make are much, much smaller than the ones precompiled. I've tried rolling back to old commits etc. and compiling from those, but no dice...

tmk, 2023-03-06:
# GCC 5 and 7
Confirmed that GCC 5.4.0 and 7.3.0  works adding this change in Makefile. `-Wl,--relax` resolves firmware size issue with those GCC version.

```
# Size optimization
# https://github.com/obdev/v-usb/blob/master/usbdrv/Readme.txt
# https://p5r.uk/blog/2008/avr-gcc-optimisations.html
EXTRALDFLAGS += -Wl,--relax

EXTRACFLAGS += -fno-function-sections
#EXTRACFLAGS += -fno-data-sections
EXTRACFLAGS += -fno-move-loop-invariants
EXTRACFLAGS += -fno-tree-scev-cprop
EXTRACFLAGS += -fno-inline-small-functions

#EXTRACPPFLAGS += -fno-function-sections
#EXTRACPPFLAGS += -fno-data-sections
EXTRACPPFLAGS += -fno-move-loop-invariants
EXTRACPPFLAGS += -fno-tree-scev-cprop
#EXTRACPPFLAGS += -fno-inline-small-functions
```

# GCC 12
With GCC 12.1.0 LTO(link time optimization) doesn't work with my code, firmware size becomes larger in the result. You can build with default configuration due to size issue at this time. You have to disable some feature, `IBMPC_MOUSE_ENABLE` is disabled here for example.

```
IBMPC_MOUSE_ENABLE ?= no 

# Size optimization                                                                                                         
# https://github.com/obdev/v-usb/blob/master/usbdrv/Readme.txt                                                              
# https://p5r.uk/blog/2008/avr-gcc-optimisations.html                                                                       
EXTRALDFLAGS = -Wl,--relax                                                                                                  
                                                                                                                            
# LTO doesn't work with C(GCC12)                                                                                            
EXTRACFLAGS += -fno-lto                                                                                                     
#EXTRACFLAGS += -fno-function-sections                                                                                      
#EXTRACFLAGS += -fno-data-sections                                                                                          
EXTRACFLAGS += -fno-move-loop-invariants                                                                                    
EXTRACFLAGS += -fno-tree-scev-cprop                                                                                         
EXTRACFLAGS += -fno-inline-small-functions                                                                                  
                                                                                                                            
#EXTRACPPFLAGS += -fno-lto                                                                                                  
#EXTRACPPFLAGS += -fno-function-sections                                                                                    
#EXTRACPPFLAGS += -fno-data-sections                                                                                        
#EXTRACPPFLAGS += -fno-move-loop-invariants                                                                                 
#EXTRACPPFLAGS += -fno-tree-scev-cprop                                                                                      
#EXTRACPPFLAGS += -fno-inline-small-functions
```