tmk/tmk_keyboard #746: Support for split numpad0 in IBMPC-USB and USB-USB (both in Unimap and TMK itself).

mbfarah, 2022-08-08:
I noticed a possible omission in the IBMPC-USB and USB-USB online keymap editors.

Looking at the IBMPC-USB Converter Unimap editors for 84-key AT and 122-key Terminal, all the numpad keys are fully split, including numpad 0. However, in the Full-key pages, numpad 0 is not split, despite Model M full-size keyboards supporting this as well; this occurs in the USB-USB Converter page as well.

I use a Soarer's converter on one of my Model M (PC) keyboards, that has with a 20-key numpad; S.C. handles the split numpad 0 without issue (the scan codes it recognizes are EXTRA_INSERT [0xB6] and PAD_0 [0x62]). I don't have a terminal Model M at hand, but I assume the scan codes won't be too different, if at all, from these.

I think that supporting a split numpad0 in the USB-USB and IBMPC-USB online keymap editors shouldn't be too much of a hassle, and a good thing to add for everyone's benefit.

tmk, 2022-08-09:
The low-left corner key spits 0x68 as scan code in Set 2. The key will be handled as `ISO#` in IBMPC converter, so you can remap it with button next to Enter on online Editor.  OK. I may change Editor layout for this, but it is a workaround for now.

As for USB-USB converter I don't think USB keyboard need that support.

mbfarah, 2022-08-09:
Thanks for the prompt reply, hasu. I'd definitely appreciate adjusting TMK and the editor to handle the extra key in the numpad as its own key ("PI" or "PE", perhaps?), as it does have its uses, beyond the typical "00" or "000" entry ('K', 'x' and ':' come to mind immediately).

As for the USB-USB firmware, you may be right — the only USB keyboards I know that split 0 in the numpad are new-ish Unicomp Model M keyboards with a USB plug, and those are surely better served by the IBMPC-USB firmware. Thanks!

tmk, 2022-08-10:
> As for the USB-USB firmware, you may be right — the only USB keyboards I know that split 0 in the numpad are new-ish Unicomp Model M keyboards with a USB plug, and those are surely better served by the IBMPC-USB firmware. Thanks!

Interesting. What code does the split 0 key spit out in USB? If you have USB-USB converter you can see its code with `hid_listen`.

mbfarah, 2022-08-11:
I don't have an USB-USB converter with me (something that we shall be correcting soon), but I do have a Soarer's Converter with an **emtpy** .sc file, which should be good enough for this test.

I used hid_listen and pressed each half of the three splittable keys in the numpad (I know you only need the first one, but I am reporting all of them, to ensure that you can translate the output if needed, given I am using a Soarer's Converter). The output of hid_listen is:

```
Waiting for device:
Listening:
r68 +B6 dB6
rF0 r68 -B6 uB6
r70 +62 d62
rF0 r70 -62 u62
r29 +2C d2C
rF0 r29 -2C u2C
r79 +57 d57
r79 rF0 r79 -57 u57
r6D +85 d85
rF0 r6D -85 u85
r29 +2C d2C
rF0 r29 -2C u2C
rE0 r5A +58 d58
rE0 rF0 r5A -58 u58
r63 +92 d92
rF0 r63 -92 u92
r29 +2C d2C
rF0 r29 -2C u2C
r14 +E0 dE0
^C

```

[image]

[image]

This is a common, run-of-the-mill 1391401 Model M keyboard.

mbfarah, 2022-08-11:
I have also tested the splittable keys of an M122 keyboard (P/N 1390876) in the same order, and the output detected for the split 0 key coincides:

```
Waiting for device:
Listening:
r68 +B6 dB6
rF0 r68 -B6 uB6
r70 +62 d62
rF0 r70 -62 u62
r29 +2C d2C
rF0 r29 -2C u2C
r7C +55 d55
rF0 r7C -55 u55
r7B +56 d56
rF0 r7B -56 u56
r29 +2C d2C
rF0 r29 -2C u2C
r79 +57 d57
rF0 r79 -57 u57
r78 +B2 dB2
rF0 r78 -B2 uB2
r11 +E0 dE0
r21 +06 d06
^C
```

[image]

I couldn't test my F122's split 0 key, but I expect its scan codes to fully coincide with the M122.

tmk, 2022-08-12:
Thanks for the information. That is exactly conincident with what John Elliot researched.
- https://www.seasip.info/VintagePC/ibm_1391406.html
- https://www.seasip.info/VintagePC/ibm_1390876.html

It is still interesting about what code Unicomp controller uses for the hidden/split keys.

tmk, 2022-08-20:
Fixed IBMPC Full-key layout of keymap editor. https://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?ibmpc_usb

https://github.com/tmk/tmk_keyboard/commit/e12e8f3741086ad45f2cfe88b0e5e8de3d7486d9

Differences from orignal layout:
- ISO \ and US \ is identical
- Hidden key next to P0 is ISO#

closed 2022-08-20 by tmk