tmk/tmk_keyboard #747: Problems with SIEMENS keyboard

Ovaron1995, 2022-08-13:
Hi together,

I used hid_listen and got the following result:
Don't know how to read the console output.

Need help...

I have the IBM keyboard converter variant.

Device disconnected.
Waiting for new device:........................
Listening:
A5501 wFF rFA A5504 wFF R R R R R R R R R R
6224 ERR:11 ISR:8000 rAA W6555 rAA W6556 rAA W6558 wF2 isr:4800 R6558
ID:FFFF(XT) S6559 L6559 rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA rAA X6726

tmk, 2022-08-14:
That keyboard replies to 'Reset' command 'FF' with 'FA' correctly, but doesn't reply to 'ReadID' command 'F2'. And it sends many of 'AA' codes for some reason.
The keyboard seems to be kind of AT keyboard with some weirdness.

What happens when plugging the converter into USB port first, wait a bit, and then connect the keyboard to the converter?

Ovaron1995, 2022-08-14:
Thx for your replay,

I connected first the converter wait a bit and then connect the keyboard now i got this:

29252 ERR:20 ISR:C000 E29253 I29253
29635 PRT:10 ISR:EA90 rAA A32255 wFF rFA rFA W32258 rFA W32260 rFA W32261 wF2 isr:A400 R32261
ID:FFFF(XT) S32262 L32262 rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA...endless

Ovaron1995, 2022-08-14:
Next I tried to connect the keyboard first to Power get from USB, then connect the converter, wait a bit and then connect the keyboard. 
Now I get this:

TMK:56e88e/LUFA:d6a7df

USB configured.

Loop start.
I2500 I2500 A5501 wFF R R R R R R R R R R A6220 wFF R R R R R R R R R R
6939 ERR:11 ISR:8000
6939 ERR:11 ISR:8000 X7440 X7441
15170 PRT:10 ISR:EA90 rAA W15170 wF2 rFA rFA rFA R15676
Unknown ID: Report to TMK wF0 isr:FA40 wF0 rFA w03 isr:2000
ID:FAFA(AT) S15681 wED isr:2000 L15682 rFA !CS2_INIT!
E15682 I15683
15684 PRT:10 ISR:FE90 rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA rFA

tmk, 2022-08-15:
I can't see why the keyboard continues to send the last code endlessly, it is uncommon behaviour for AT keyboard.
This behaviour prevents command-response communication between the keyboard and the converter.

Can you try this firmware? This sends a command in different way.
https://gist.github.com/tmk/26d0eaada1f32fc74cd217dcbeb9e521

I don't think that power supply is a problem with the converter, you can draw around 500mA from USB port probably. But you may want to try a port of 'Powered USB hub' in case the keyboard needs more.

Ovaron1995, 2022-08-15:
Thanks for the firmware.

Now i see the following in the debugger:

TMK:dfe114/LUFA:d6a7df

USB configured.

Loop start.
I2500 I2500 A5501 wFF R R R R R R R R R R A6220 wFF R R R R R R R R R R
6939 ERR:11 ISR:8000
6939 ERR:11 ISR:8000 X7440 X7441  

connecting keyboard

1366 PRT:10 ISR:EA90 rAA W1367 wF2 rFA rFA rFA R1872
Unknown ID: Report to TMK wF0 isr:FA40 rFA w02 isr:9000 rFE wF0 rFA w03 isr:2000 rFE
ID:FAFA(AT) S1889 wED rFA w00 isr:9000 rFE L1895   

 r1C r1C r1C r1C r1C r1C r1C r1C... endless
 
 without debugger i see the character which I hit in the console, but endless. But I see the character.

tmk, 2022-08-16:
hmm, the keyboard seemt to work in different way from what my converter expects, at least.

Can you capture wave form when the keyboard is connected to computer that works well?
It would be helpful if you can take startup communication between them using logic analyzer.

Ovaron1995, 2022-08-16:
Ordered a logic analyzer for the keyboard. When it is available i will record the signals at startup on the pc where the keyboard works.

will be back with pics when it is done…
Thx

Ovaron1995, 2022-08-17:
Hi tmk,

so it's done, here the recorded signals from turning on the PC until Windows 10 is booted.
[Keyboard.zip](https://github.com/tmk/tmk_keyboard/files/9358684/Keyboard.zip)
It's recorded with PulseView.

Hope this is what you wanted.

Thx

Ovaron1995, 2022-08-17:
FYI this is the keyboard i want to get running:
[image]

tmk, 2022-08-17:
The data file doesn't seem to include correct signal capture due to low sample rate perhaps?

Can you capture signals with sample rate **1MHz**(or better) again?

Ovaron1995, 2022-08-17:
Sorry i missed this,

here the new one with 8MHz.
[Keyboard.zip](https://github.com/tmk/tmk_keyboard/files/9363016/Keyboard.zip)

Thanks
Michael

tmk, 2022-08-18:
Thanks for the capture.
The keyboard seems to  communicate with the computer in normal way except for a few unclear commands.
I could't find  promising clue to resolve the problem yet.

I added extra 'inhibit' time that the computer uses. Can you try this firmware?
[firmware_32u4.zip](https://github.com/tmk/tmk_keyboard/files/9372689/firmware_32u4.zip)

Ovaron1995, 2022-08-18:
Hi tmk,

these words are typed with the SIEMENS keyboard connected with your converter.
This means your firmware runs.

Many many thanks for your fast and successful help. You are the best.

Kind regards
Michael

Ovaron1995, 2022-08-18:
Hi tmk,

one additional question. As you can see  on the picture of the keyboard, i moved the minus key from the numpad upper right corner to the lower left side of the keyboard near by the left CTRL key. I used the free place on the circuit board and insert the minus-key. I hoped to get an key for the windows-key but it sends the same code as the CTRL-key.

Is it right that when I want to change this code I need to reed out the 8071 with eeprom and change the key matrix table?

Thx
Michael

tmk, 2022-08-19:
Great.
The SIEMENS PCD 2 keyboard seems to expect to be "inhibited" by host for a while. But I can't see why it requires the "inhibit". Other keyboards doesn't require the "inhibit" at all in most cases.

Can you share hid_listen debug log?

This is code modification for this for reference. I'll merge this into repository later unless this prevents other keyboards.
```
commit 5f3c9ed01b8b877c8d47f868e1f24e7c5d6314f7
Author: tmk <hasu@tmk-kbd.com>
Date:   Thu Aug 18 19:30:30 2022 +0900

    ibmpc: siemens: Add inhibit after send/recv

diff --git a/tmk_core/protocol/ibmpc.cpp b/tmk_core/protocol/ibmpc.cpp
index e9af15c6..d7286f2c 100644
--- a/tmk_core/protocol/ibmpc.cpp
+++ b/tmk_core/protocol/ibmpc.cpp
@@ -96,7 +96,7 @@ int16_t IBMPC::host_send(uint8_t data)
     //if (isr_state & 0x0FFF) {
     if (isr_state != 0x8000) {
         dprintf("isr:%04X ", isr_state);
-        return -1;
+        //return -1;
     }
 
     int_off();
@@ -140,6 +140,11 @@ RETRY:
     WAIT(data_hi, 300, 7);
     WAIT(clock_hi, 300, 8);
 
+    /* Inhibit for a while */
+    wait_us(15);
+    clock_lo();
+    wait_us(150);
+
     // clear buffer to get response correctly
     host_isr_clear();
 
@@ -360,6 +365,11 @@ void IBMPC::isr(void)
     }
 
 DONE:
+    // Inhibit
+    clock_lo();
+    wait_us(150);
+    clock_hi();
+
     // store data
     ringbuf_put(isr_state & 0xFF);
     if (ringbuf_is_full()) {
```

As for the 'free' key, yes, I think you need hardware modification if it sends same scan code as CTRL-key.

tmk, 2022-08-19:
MEMO:

The computer adds "inhibit" time after sending and receving data sessions. This appear to be remedy for the keyboard. Without the "inhibit" the keyboard seems to repeat the last codes endlessly.

This is capture from communication between the computer and the keyboard.
[image]

I don't know actually how long and when "inhibit" is needed for the keyboard. The converter fix simply emulates behaviour of the computer. 
[image]

This is capture from original converter without the fix.
[image]

closed 2022-09-08 by tmk

tmk, 2023-02-17:
Updated SIEMENS PCD keyboard support due to issue #751 at commit d51023b.

Let me know if you have a problem with the latest firmware.