tmk/tmk_keyboard #750: Keys not lining up based on keymap

anthonygualandri, 2022-12-16:
WHen I flash my hhkb pro3 using the usb to usb converter Several of the keys do not lie up with the tmk graphical interface placement I used in the hex file. For example I tried to make the space bar key a layer toggle and the one next to it is the layer toggle key. OR the delete key is the backtick from above it. Any idea what's going on here?

tmk, 2022-12-17:
Not sure about your problem. can you sahre your hex file with the problem?

Did you check DIP switches setting for delete key?

anthonygualandri, 2023-01-10:
Attached is the hex file. I had the dip switches set to have the delete key be backspace, but the hex file is making the pipe / backslash one u key above it the backspace. Can’t I just turn off all the dip switches and let the controller programming using the TMK editor control all the keys’ functionality? From: ***@***.***: Friday, December 16, 2022 6:45 PMTo: tmk/tmk_keyboardCc: Anthony Gualandri; AuthorSubject: Re: [tmk/tmk_keyboard] Keys not lining up based on keymap (Issue #750) Not sure about your problem. can you sahre your hex file with the problem?Did you check DIP switches setting for delete key?

anthonygualandri, 2023-01-10:
Everything seems to be working now with this hex file except the pip/backslash, the back tick and the back space keys. Slash is still backspace, tick is nothing and the Delete key is backtick. From: ***@***.***: Friday, December 16, 2022 6:45 PMTo: tmk/tmk_keyboardCc: Anthony Gualandri; AuthorSubject: Re: [tmk/tmk_keyboard] Keys not lining up based on keymap (Issue #750) Not sure about your problem. can you sahre your hex file with the problem?Did you check DIP switches setting for delete key?

anthonygualandri, 2023-01-10:
No matter what I set the backslash key to in the software, it’s always outputting the delete key action, the backtick is always nothing and the Delete key in the second row is always backtick. Sorry for the multiple email, but I can’t seem to figure this out! Do the dip switches need to be set to something specific for the converter. Changing them after flashing the firmware doesn’t seem to change anything though.  From: ***@***.***: Friday, December 16, 2022 6:45 PMTo: tmk/tmk_keyboardCc: Anthony Gualandri; AuthorSubject: Re: [tmk/tmk_keyboard] Keys not lining up based on keymap (Issue #750) Not sure about your problem. can you sahre your hex file with the problem?Did you check DIP switches setting for delete key?

tmk, 2023-01-14:
Still not sure I can understand your problem.

You can type the key as expected with default layout of Keymap Editor firmware. Delete(Backspace) key should work as Delete(Backspace).  Did that work for you?

First, physical key layout of HHKB and one of Keymap Editor is not completely coincident in terms of its placement, because HHKB layout is peculiar and Keymap Editor has generic layout to support as many keyboards as possible.

You have to handle those keys(Delete, \ and `) with specific care in Keymap Editor.
You can remap \ key on HHKB with \ key(one above Enter) on Editor(not JIS ￥), ` with `(one above Tab) or Delete(Backspace) with Delete(Backspace).

tmk, 2023-02-02:
Your comments look broken somehow.

Can you come this URL and edit your comments?
https://github.com/tmk/tmk_keyboard/issues/750

anthonygualandri, 2023-02-03:
I updated my previous two comments above.

tmk, 2023-02-03:
Is this image what you want to attach? probably not.

[image]

You can share your keymap using URL keymap editor.
Use 'Raw URL' button to get URL for your keymap. It is very long but you can post it here.
(if 'URL shortener' button works you can use)

Yes, ALT and GUI are reversed. It is because HHKB default layout(SW5=OFF) is reversed in comparison with PC keyboard. Keymap Editor layout is based on PC keyboard(not HHKB or Mac).  This is intended design of  USB-USB converter and keymap editor, you have to grok and bear with unfortunately. Or just turn SW5 on, this will be resolved if this is problem for you.

As for Fn key USB-USB converter can not remap the key.  The Fn key is processed completely inside of HHKB, it doesn't send any code to the converter.

tmk, 2023-02-03:
If you want to remap Esc of HHKB to A you have to use Esc key on **default layout** of Keymap Editor. I mean a key at top left corner, not ` key.

Also let me remap Backspace(SW3=ON) to B,  `(backtick) to C, \ to D and LControl to E.

See this Keymap Editor URL and pics below, hope this explains USB-USB converter idea clearly.

https://bit.ly/3wXc8Yk

Default layout of Keymap Editor:
[image]

Remapped for HHKB:
[image]

anthonygualandri, 2023-02-03:
Your most recent explanation clears everything up for me! Now I understand how the TMK editor works with the converter. I was trying to map the board with it like you do with a tool like VIA. Now it all makes sense. Thank you, and sorry for not getting it at first!

closed 2023-02-03 by anthonygualandri