tmk/tmk_keyboard #751: Model F 83-keys XT 83-key - Only seeing ERR:20, but detects each button press; doesn't work.

GrimbleFitz, 2023-02-14:
I've double-checked everything and am now completely stumped as to what issue may be occurring now. I only see the following when running hid-listen:

> Waiting for device:......
> Listening:
> 
> TMK:60b648/LUFA:d6a7df
> 
> USB configured.
> 
> Loop start.
> I2500 I2500 A5501 wFF A5504 wFF R R R R R R R R R R 
> 6223 ERR:12 ISR:8000 
> 6223 ERR:11 ISR:8000 X6724 X6725 
> 10134 ERR:20 ISR:2000 E10134 I10135 A13136 wFF 
> 13139 ERR:12 ISR:8000 X13640 
> 15163 ERR:20 ISR:2000 E15164 I15164 
> 15265 ERR:20 ISR:1000 E15265 I15266 
> 15453 ERR:20 ISR:2000 E15454 I15454 
> 15583 ERR:20 ISR:1000 E15584 I15584 
> 15787 ERR:20 ISR:2000 E15787 I15788 
> 15888 ERR:20 ISR:1000 E15888 I15889 A18890 wFF 
> 18893 ERR:12 ISR:8000 X19394 
> 23424 ERR:20 ISR:2000 E23425 I23425 
> 23627 ERR:20 ISR:2000 E23628 I23629 
> 23816 ERR:20 ISR:1000 E23816 I23817 
> 23990 ERR:20 ISR:2000 E23991 I23991 
> 24193 ERR:20 ISR:1000 E24193 I24194 
> 24381 ERR:20 ISR:1000 E24382 I24382 
> 24584 ERR:20 ISR:1000 E24585 I24585 
> 24787 ERR:20 ISR:2000 E24788 I24788 
> 25004 ERR:20 ISR:2000 E25005 I25005 
> 25222 ERR:20 ISR:2000 E25223 I25223 A28224 wFF 
> 28227 ERR:12 ISR:8000 X28728 

hid-listen detects all my keypresses, but only ever reports ERR:20. If I pause typing for a moment the keyboard outputs that wFF + ERR:12 text. Did I miss something simple? I've tried to scour what resources I can for information about this, without success. 

Thanks!

tmk, 2023-02-14:
Does a few replug of keyboard(not the conveter) work ?

tmk, 2023-02-14:
If repluging doesn't work  can you try this old firmware and post debug log?

https://github.com/tmk/tmk_keyboard/tree/bb955a819af89e74843743e268aa7b4ceb06cc4e/converter/ibmpc_usb/binary

GrimbleFitz, 2023-02-15:
That firmware works great! Haven't noticed any issues with it so far. Here's output from hid-listen showing me randomly pressing keys.

> Listening:
> 
> 
> TMK:6b71d881/LUFA
> 
> 
> USB configured.
> 
> Keyboard start.
> I673 wFF A1689 
> ERR:17 ISR:0000 X2191 
> PRT:21 ISR:AAA0 rAA W2226 wF2 R2762 
> ID:FFFF(1) 
> ERR:11 ISR:0000 S2763 L2763 r23 hr22 grf21 r20 drA3 rA2 rA1 rh23 rA0 r22 gr21 fr20 drA3 rA2 rA1 r23 hrA0 rg22 r21 fr20 drA3 rA2 rA1 rA0 r39  rB9 r39  rB9 r 39 rB9 r39  rB9 r39  rB9 r1C 
> r9C r
> 1C r9C r1C 
> r9C r25 krj24 rA5 rA4 r24 jr23 hrk25 rg22 rA5 rA4 rA3 rA2 r24 jr23 hr21 fr22 grA4 rA3 rA1 rA2 r39  rB9 

Is there any other information I could provide you that would help?

GrimbleFitz, 2023-02-15:
I should clarify, the previous firmware I was attempting to use was the latest hex from tmk-kbd. (http://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?ibmpc_usb_32u4_xt)

tmk, 2023-02-15:
Yes, the firmware offered on Keymap  Editor is the latest.
It seems the latest firmware(TMK:60b648) got something wrong with recnet changes.

To find bad change can you try these two firmwares and let me know result?

https://github.com/tmk/tmk_keyboard/tree/124bf700b43dae3d47f2d623b5b8e37327cc57d3/converter/ibmpc_usb/binary

https://github.com/tmk/tmk_keyboard/tree/44bf879285e1dc104fc47bc0226c38168fa8a8af/converter/ibmpc_usb/binary

GrimbleFitz, 2023-02-15:
The first firmware works; my keyboard is functional and reports this output via hid-listen:

> Listening:
> 
> TMK:6125ef/LUFA:d6a7df
> 
> USB configured.
> 
> Loop start.
> I780 I780 
> 807 PRT:21 ISR:AAA0 rAA A3781 wFF R A3851 wFF R R R R R R R R R R 
> 4570 ERR:17 ISR:AAA0 
> 4570 ERR:11 ISR:0000 X5071 X5072 rAA W5101 wF2 R R R R R R R R R R R6320 
> ID:FFFF(XT) 
> 6321 ERR:11 ISR:AAA0 S6321 L6322 rAA r27 ;r26 lr25 kr24 jrA6 rA7 rA5 rA4 r1E ar1F sr2d0 r21 fr9E r9F rA0 rA1 

The second firmware also works, here's the output from that:

> Listening:
> 
> 
> TMK:2a0471fd/LUFA
> 
> 
> USB configured.
> 
> Keyboard start.
> I917 A3918 wFF d:0 R R R R R R 
> 4138 ERR:15 ISR:0000 X4639 
> 4671 PRT:21 ISR:AAA0 rAA W4672 wF2 d:0 R R R R R R R R R5463 
> ID:FFFF(XT) 
> 5464 ERR:15 ISR:0000 S5464 L5464 r27 ;r2l6 r25 kr24 jrA6 rA7 r1E arA5 rA4 r1F sr20 dr21 fr9E r9F rA0 rA1 r1D r1D r1D r1D r1D r9D

tmk, 2023-02-16:
Thank you for your time.

I found suspicious part in my code and changed somethig.
I think this commit causes the issue probably. 52cdb909

Can you test attached firmwares? (you will need to change suffix to .hex. github uploader doesn't accept .hex file.)

[pcd2_disable_hex.txt](https://github.com/tmk/tmk_keyboard/files/10752781/pcd2_disable_hex.txt)
[pcd2_fix_hex.txt](https://github.com/tmk/tmk_keyboard/files/10752782/pcd2_fix_hex.txt)

GrimbleFitz, 2023-02-16:
`pcd2_disable` seems to work fine.

> Listening:
> 
> TMK:7fbf2c/LUFA:d6a7df
> 
> USB configured.
> 
> Loop start.
> I2500 I2500 
> 2514 PRT:21 ISR:AAA0 rAA A5501 wFF R R R R R R R R R R A6221 wFF 
> 6262 ERR:11 ISR:8000 
> 6263 ERR:17 ISR:8000 X6764 X6764 rAA W6795 wF2 R R R R R R R R R R R8014 
> ID:FFFF(XT) 
> 8015 ERR:11 ISR:8000 S8016 L8016 r27 ;r2l6 r2k5 rj24 rA7 rA6 rA5 rA4 r1aE rs1F r2d0 r2f1 r9E r9F rA0 rA1 r2;7 r2l6 r2k5 rj24 rA7 rA6 rA5 rA4 ra1E rs1F r2d0 r2f1 r9E r9F rA0 rA1 r;27 rl26 r2k5 rj24 rA7 rA6 rA5 rA4 r1aE r1sF r2d0 r2f1 r9F r9E rA0 rA1 

`pcd2_fix` works as well.

> Listening:
> 
> TMK:7fbf2c/LUFA:d6a7df
> 
> USB configured.
> 
> Loop start.
> I2500 I2500 
> 2515 PRT:21 ISR:AAA0 rAA A5501 wFF R A5571 wFF R R R R R R R R R R 
> 6290 ERR:17 ISR:AAA0 
> 6290 ERR:11 ISR:8000 X6791 X6792 rAA W6821 wF2 R R R R R R R R R R R8040 
> ID:FFFF(XT) 
> 8041 ERR:11 ISR:8000 S8042 L8042 r26 lr2k5 r2j4 rA6 rA5 rA4 r1aE rs1F r2d0 r2f1 r9E r9F rA0 rA1 r2l6 rk25 r2j4 rA6 rA5 rA4 r1sF r2d0 rf21 r9F rA0 rA1 r32 rB2 r18 r31 r98 rB1 r25 rA5 r12 r15 r92 r95 r1C r9C

tmk, 2023-02-17:
@GrimbleFitz Thank you for testing.
It seems to fix the issue.

Updated source code and prebuilt firmware in github repository.
The latest firmware is available also in Keymap Editor.

Let me know if you still have any problem.

closed 2023-02-17 by tmk