tmk/tmk_keyboard #752: build error when building ibmpc_usb

Gavus, 2023-02-23:
I have checkout 1820eff74 (master when writing) and I'm getting errors when trying to build ibmpc_usb.
[output.txt](https://github.com/tmk/tmk_keyboard/files/10817133/output.txt)
Basically says section .text overlaps section .keymap. 

I can build other stuff like hhkb without any issues.
avr-gcc version is 5.4.0.

tmk, 2023-02-24:
It seems that firmware is bloated for some reason. 
If you don't edit any souce code try updating gcc to 7.3.0 or any newer than 5.4.0. (If you do it would be a cause.)

I had used 5.4.0 long time and it can be built somehow. Recent changes may cause with 5.4.0. But I have used 7.3.0 now and have no problem.

```
-------- begin --------
avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

...

Linking: ibmpc_usb.elf
...
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: section .keymap loaded at [0000000000006800,00000000000069ff] overlaps section .text loaded at [0000000000000000,0000000000006823]
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: section .data loaded at [0000000000006824,00000000000068bd] overlaps section .keymap loaded at [0000000000006800,00000000000069ff]
collect2: error: ld returned 1 exit status
make: *** [../../tmk_core/rules.mk:556: ibmpc_usb.elf] Error 1

```

Gavus, 2023-02-26:
I tried to manually update avr-gcc on my Ubuntu setup but couldn't figure out out in a good way. 

A fedora 37 docker environment solved my issues instead. It had a newer version of avr-gcc. Thanks!

closed 2023-02-26 by Gavus