tmk/tmk_keyboard #756: Wake from suspend stopped working

mek-apelsin, 2023-03-11:
I suspect commit 781e2dfce905fcce36f5458b0a3478d98c449e64 introduced a bug which inhibits the ability to wake from sleep. I can't get my computer (running Ubuntu) to wake from sleep when using a m0118, a teensy2 and the "vanilla" keymap. 

I can reliably trigger the "buggy" behavior by doing the following:
`git checkout 781e2dfc && ( git submodule update  && cd converter/adb_usb/ && make --silent clean  && MCU=atmega32u4 make --silent teensy ) && sleep 5 && systemctl suspend`

I re-enable the ability to wake from suspend by doing the following: 
`git checkout 781e2dfc~ && ( git submodule update  && cd converter/adb_usb/ && make --silent clean  && MCU=atmega32u4 make --silent teensy ) && sleep 5 && systemctl suspend`

Haven't had the time to do much debugging, so far I got this:

* GOOD
** PWR then SPACE  during ordinary usage:
7F7F FFFF 31FF  B1FF 
** PWR then SPACE  during suspend:
F7F FFFF 1FF 

* BAD
** PWR then SPACE  during ordinary usage:
$E:7F7F $E:FFFF $E:31FF $E:B1FF 
** PWR then SPACE  during suspend:
$E:7F7F $E:FFFF $E:31B1

tmk, 2023-03-11:
Can you try adb_usb_rev1_unimap.hex and show full debug log?
https://github.com/tmk/tmk_keyboard/tree/master/converter/adb_usb/binary

tmk, 2023-03-11:
Confirmed the bug. You are right, the commit introduced it.

This change should fix the bug. I'll update code in repo soon. Thanks.
```
diff --git a/converter/adb_usb/matrix.c b/converter/adb_usb/matrix.c
index 73c47649..74470743 100644
--- a/converter/adb_usb/matrix.c
+++ b/converter/adb_usb/matrix.c
@@ -811,11 +811,6 @@ void matrix_init(void)
 {
 }
  
-uint8_t matrix_scan(void)
-{
-    return 0;
-}
-
 inline
 matrix_row_t matrix_get_row(uint8_t row)
 {
@@ -832,7 +827,7 @@ void led_set(uint8_t usb_led)
     adb_host_kbd_led(ADB_ADDR_KEYBOARD, ~usb_led);
 }
  
-void hook_main_loop(void)
+uint8_t matrix_scan(void)
 {
     static uint16_t poll_ms;
     static uint16_t detect_ms;
```

tmk, 2023-03-11:
@mek-apelsin Can you share M0118 debug log at startup like below?
It would be helpful to improve ISO-layout keyboard support.

```
TMK:e9c74f/LUFA:d6a7df

USB configured.

Scan:
$2 R3:6C01

Loop start.
R:$2:Found. R3:6D01
R:$2:Move to $F. R3:6A01
K:$F:Setup. R3:6101
K:$F:Layout: ANSI

Table:
A:H  a:h
---------
F:01 2:01
```

mek-apelsin, 2023-03-11:
Waiting for device:
Listening:
# Running teensy_loader_cli  --mcu=atmega32u4  -w converter/adb_usb/binary/adb_usb_rev1_unimap.hex in other window

Device disconnected.
Waiting for new device:....
Listening:

TMK:c46d7b/LUFA:d6a7df

USB configured.

Scan:
$2 R3:6804

Loop start.
R:$2:Found. R3:6A04
R:$2:Move to $F. R3:6504
K:$F:Setup. R3:6D04
K:$F:Layout: ISO

Table:
A:H  a:h
---------
F:04 2:04

# Testing a few keys... a  g  l  PWR and lshift+a
K:$F:04:00FF K:$F:04:80FF K:$F:04:05FF K:$F:04:85FF K:$F:04:25FF K:$F:04:A5FF K:$F:04:7F7F K:$F:04:FFFF 
Scan:
$F R3:6104

Table:
A:H  a:h
---------
F:04 2:04

K:$F:04:38FF K:$F:04:00FF K:$F:04:80FF K:$F:04:B8FF 

# Enabling debug:
K:$F:04:7F7F K:$F:04:02FF 
debug: off
K:$F:04:82FF K:$F:04:FFFF 
Scan:
$F R3:6004

Table:
A:H  a:h
---------
F:04 2:04

# oh, it is on by default, reenabling

K:$F:04:7F7F K:$F:04:02FF 
debug: on
K:$F:04:82FF K:$F:04:FFFF 
Scan:
$F R3:6304

Table:
A:H  a:h
---------
F:04 2:04

K:$F:04:33FF K:$F:04:B3FF K:$F:04:33FF K:$F:04:B3FF K:$F:04:33FF K:$F:04:B3FF K:$F:04:33FF K:$F:04:B3FF 
# A few Backspaces

# And lets run suspend...

K:$F:04:31B1 K:$F:04:31B1 K:$F:04:31B1 K:$F:04:31B1 

# Pressed quite a few times on space and PWR, and once on 'j'. Did not wake from suspend.

mek-apelsin, 2023-03-11:
The fix in  ecf52983 works as expected.

closed 2023-03-11 by mek-apelsin

tmk, 2023-03-11:
Thanks for the log.

I confirmed that M0118 has hanlder ID:0x04 and ISO Layout of the keyboard should be supported by current firmware.