tmk/tmk_keyboard #761: ACTION_LAYER_SET_CLEAR not working (anymore) when called during an ACTION_LAYER_TAP_KEY event?

mbfarah, 2023-06-29:
I have an Apple M0120 numpad, which I use solely as a numpad. I made all my arrangements with the TMK Keymap editor.

I have defined seven distinct layers (0..6), for differing modes of operation. I did not want a dedicated "Fn" key, so I made the COMMA key into an ACTION_LAYER_TAP_KEY, to "Turn on Layer 7 and ,(tap)", and select the layer I wanted by way of an ACTION_LAYER_SET_CLEAR assignment for each one, all in layer seven.

I made this about a year ago, and it worked as expected. During this week, I decided to make a few changes to one of the layers. I edited the layout on the editor, flashed, etc. — now changing from layer 0 to any layer 1 to 6 doesn't work. I know the momentary turn-on of layer 7 works, because all the other assignments I've made in that layer respond as expected.

I'm not sure if I've run into a new bug OR whether I had being doing this the wrong way and _now_ I can't get away with it.

I did not use ACTION_DEFAULT_LAYER at the time because AFAIU there could be more than one default layer active at the time.

What I want (which I could do before but not now) is to have _exactly one_ layer within 0..6 active at all times, with the COMMA key serving as the momentary toggle to layer 7, where I can set wich of the layers 0..6 is going to be the one active.

This is the generated JSON for the file.

[jsonM0110.txt](https://github.com/tmk/tmk_keyboard/files/11905716/jsonM0110.txt)

Thank you.

tmk, 2023-06-30:
M0110 firmware was updated recently for the first time in six years.
I think there are some inconsistencies in keymap system between the latest firmware built in Jan/2023 and previous firmware in 2016. You cannot load and use your old keymap in current Keymap Editor, perhaps. Did you try defining keymap from scratch, for the problem keys at least?

Can you share raw URL(or shortened URL) of your keymap?

mbfarah, 2023-06-30:
Based on your suggestion, I defined the keymap from scratch on the TMK editor; the problem persists.

This is the raw URL for the keymap I have done from scratch:

[new_M0120.txt](https://github.com/tmk/tmk_keyboard/files/11916344/new_M0120.txt)

This is the raw URL for the keymap I made back in July 2022:

[old_M0120.txt](https://github.com/tmk/tmk_keyboard/files/11916345/old_M0120.txt)

Thanks.

tmk, 2023-07-02:
Confirmed the problem.

`LAYER_SET_CLEAR` behaviour was modified by this bug fix in 2017. The action had worked incorrectly against my intention until the fix. You should use 'LAYER_SET` instead of it.
https://github.com/tmk/tmk_keyboard/commit/ba2883fd9ab040fc670ad729e6cddd4c67c8188c

`LAYER_SET_CLEAR`  enables a layer solely(other layers are disabled) on press and disable all layers on release(always return to layer 0).

 'LAYER_SET` enalbes a layer solely(other layers are disabled) on press(or release).

Keymap Editor had another bug and `LAYER_SET` had not been available on Editor. I just fixed Editor for `LAYER_SET` you can use it now. 

You can find  `LAYER_SET` action on 'Layer Switching' tab and 'Code Edit' tab.

mbfarah, 2023-07-02:
I just tested the layout using LAYER_SET instead of LAYER_SET_CLEAR and it works as expected.

Thank you very much.

closed 2023-07-02 by mbfarah