tmk/tmk_keyboard #762: ADB mouse movement slow

maltic42, 2023-07-01:
Hi,

I am using the converter on a Teensy 2.0 with an Apple ADB Mouse G5431. Everything works except that mouse movement is slow. Is it somehow possible to change the mouse speed?

Thanks!

Regards,
Malte

tmk, 2023-07-02:
You can implement mouse acceleration here.
https://github.com/tmk/tmk_keyboard/blob/master/converter/adb_usb/matrix.c#L607-L629

The current code intends to report original values from mouse and change speed configuration in computer.
You can refer to this for previous acceleration implement removed.
https://github.com/tmk/tmk_keyboard/commit/9643a76b8ec13dca2d01d0b1bd984c2abac6e9f1

Many of ADB mouses have very low cpi/dpi and will be less useful in modern desktop evironment.
Kensington Turbo mouse has relatively high cpi, just FYI.

maltic42, 2023-07-02:
Thanks - I tweaked the code a little bit and it's working now!

closed 2023-07-02 by tmk