tmk/tmk_keyboard #763: PS/2 converter needs to be unplugged and replugged upon restarting the computer.

mbfarah, 2023-07-04:
I have an IBM Model M SSK keyboard plugged to my computer through a TMK PS/2 converter, with the _IBMPC-USB Converter(TMK/ATmega32U2)_ firmware. It works as expected during regular usage, but every time I restart my computer (either through a reboot software command, or by turning it off and then back on), it ceases to work. Upong restart, the control panel shows that the converter IS recognized by the OS and shows up in the list of devices, even though it's not doing anything.

Once I deplug and replug the USB cable, the keyboard starts working again.

I also have an Apple M0120 numpad (using its own TMK converter) plugged to the same computer at the same time; it does *not* show this behaviour.

I have tested this, and obtained the same results, both with an SSK and a full-size Model M keyboard, on a laptop with Windows 10 and a laptop with Windows 11. Weirdly (?), this does not happen on a desktop computer running Linux.

I also made a test where I let the Windows computer be inactive for a while until its screen went blank. Then I pressed the Control key — Windows registered the event and turned the screen back on, but still did not otherwise take the output of the keyboard.

Converters recognized by Windows 11 (while the converter seems to be not sending keystrokes):

[image]

Converters recognized by Windows 10 (while the converter seems to be not sending keystrokes):

[image]

Just in case, this is the raw URL for the converter's mapping:

[SSKrawurl.txt](https://github.com/tmk/tmk_keyboard/files/11949559/SSKrawurl.txt)

Thanks.

tmk, 2023-07-05:
Confirmed the problem on Windows10.

This is related to NKRO. M0110 converter won't suffer from this because it doesn't support NKRO.

This describes the same issue in detail.
https://github.com/keyboardio/Kaleidoscope/issues/1314

This change seems to fix the issue. Try this if possible.
```
diff --git a/tmk_core/protocol/lufa/lufa.c b/tmk_core/protocol/lufa/lufa.c
index ec78e49a..4855fc3b 100644
--- a/tmk_core/protocol/lufa/lufa.c
+++ b/tmk_core/protocol/lufa/lufa.c
@@ -41,6 +41,7 @@
 #include "host_driver.h"
 #include "keyboard.h"
 #include "action.h"
+#include "action_util.h"
 #include "led.h"
 #include "sendchar.h"
 #include "ringbuf.h"
@@ -305,6 +306,17 @@ void EVENT_USB_Device_Reset(void)
 #ifdef TMK_LUFA_DEBUG
     print("[R]");
 #endif
+#ifndef NO_KEYBOARD
+    // Reset to report protocol(default) and clear keyboard report
+    keyboard_protocol = 1;
+    keyboard_idle = 0;
+    keyboard_led_stats = 0;
+    keyboard_report_sent = (report_keyboard_t){};
+    clear_keys();
+#endif
+#ifdef MOUSE_ENABLE
+    mouse_protocol = 1;
+#endif
 }
 
 void EVENT_USB_Device_Suspend()
```

tmk, 2023-07-05:
Updated IBMPC converter firmware on Keymap Editor.
Can you try this?

closed 2023-07-06 by tmk