tmk/tmk_keyboard #765: Can ACTION_DEFAULT_LAYER_SET be used in the Unimap Editor? (And how?)

mbfarah, 2023-07-07:
I am composing a layout definition for a full-size keyboard. I want three default layers (which differ only in the placements of Esc, Tab and Fn). I expected to be able to use the Unimap Editor to set either of the three as the default using ACTION_DEFAULT_LAYER_SET, but it seems it has not been fully implemented in the Unimap editor:

[image]

(I would expect to be able to pass one layer as an argument to ACTION_DEFAULT_LAYER_SET)

I don't want to use LAYER_SET here (unlike issue 761), because I have an additional layer defined for the numpad, which I can toggle on and off with its own keystroke (if I used LAYER_SET for the other three layers, the numpad's state would be reset).

Thanks.

Raw URL:

[rawurl765.txt](https://github.com/tmk/tmk_keyboard/files/11973658/rawurl765.txt)

tmk, 2023-07-07:
I was working on keymap editor to fix 'Code Edit' including this issue.
Keyboard Editor will be updated some later. I'll post here when the fix is available.

tmk, 2023-07-08:
Just updated Keyboard Editor to fix ACTION_DEFAULT_LAYER_SET support.

https://github.com/tmk/tmk_keyboard/commit/a6c87d2b6ae9ab736ea7a30df4bb3b26da579638

mbfarah, 2023-07-08:
I have tested the ACTION_DEFAULT_LAYER_SET and it's working as expected.

Thank you very much.

closed 2023-07-08 by mbfarah