tmk/tmk_keyboard #766: Model F 83-keys XT 83-key - Occasional stuck key, keyboard freezes

GrimbleFitz, 2023-08-11:
I've used my converter without issue since #751, but have recently experienced frequent freezes and stuck keys. It happens roughly every 5 to 15 minutes, lasting <10 seconds each time. Oftentimes, the key I was pressing when the issue occurs will repeat for several seconds while the the keyboard is locked up. Then the keyboard will resume normal function until it happens again.

Here's the keyboard booting up,

> Listening:
> 
> TMK:b0ca95/LUFA:d6a7df
> 
> USB configured.
> 
> Loop start.
> I2500 I2500 
> 2517 PRT:21 ISR:AAA0 rAA A5501 wFF R R R R R R R R R R A6220 wFF R R R R R R R R R R 
> 6939 ERR:11 ISR:8000 
> 6939 ERR:11 ISR:8000 X7440 X7441 rAA W7474 R7975

and later when the issue occurs. 

> r35 keyboard: 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> r35 
> 57608 ERR:20 ISR:C000 E57609 keyboard: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> keyboard: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> I57615 
> 57659 ERR:20 ISR:FC00 E57659 keyboard: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> I57663 
> 57669 PRT:21 ISR:0EA0 r0E r8E A60664 wFF R R R R R R R R R R 
> 61383 ERR:11 ISR:8000 X61884 rAA W61913 R62415 
> ID:FFFC(XT_IBM) S62415 L62416 r0E keyboard: 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r0E r8E keyboard: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Steps I've taken to remedy the issue: 1. Clean the keyboard, 2. Check all connections, 3. Unplug/replug, 4. Update with fresh firmware from tmk-kbd.com

tmk, 2023-08-12:
1. You said: ``` I've used my converter without issue since #751 ```
2. Firmware version showed in the log: ``` TMK:b0ca95/LUFA:d6a7df ``` is what fixed #751 back in Feb.

Judging from two points above firmware won't be problem here, hardware of converter or keyboard can cause probably. Change of temperature and humidity may reveal this.

tmk, 2023-08-12:
Post closeup pic of your covnerter hardware if you are not familiar with these things.
I can check it briefly for you.

tmk, 2023-08-14:
Note that you need two pull-up resistors on both DATA(PD0) and CLOCK(PD1) line for IBMPC converter.

Check this first post also.
https://geekhack.org/index.php?topic=103648.0