tmk/tmk_keyboard #767: ps2-usb: device not recognized at cold boot (bios and windows both)

jeongsu816, 2023-11-13:
hello.

thank you for making this great firmware. it is really helpful for me to make a ps2-usb conveter.

but i have a problem.
my converter is not recognized when the pc cold boot. but something strange is that when i connect the converter to front panel usb port, it is just working properly.

from back panel -> not recognized at cold boot. even bios. i unplug and plug again, it is recognized but unrecognized shortly. 
                             hid_listen just returned to listening mode when this happened
from front panel -> just working fine from bios and windows

how can i debug this problem? where is the good starting point?

jeongsu816, 2023-11-13:
oh, after writing this issue, i just pull your latest master branch and now it is working properly

jeongsu816, 2023-11-13:
it seems that it is not firmware issue. something strange but my case is the root cause.
my pc case is from the geometric, it is rotated by 90 degrees, means back panel is actually on the top and a cover is over it.
it makes usb cables are bended. 

i make it plugged firmly and now it is working properly.

jeongsu816, 2023-11-13:
it happened again. this time, i just unplug and plug it and now it works.

so please let me know how can i debug this problem and where is the good starting point.

tmk, 2023-11-14:
Let us know some info below if possible, these may be useful to help you.

Can you share some photos of your converter?
What's model no of your keyboard?
What's model no of your mother board?
The problem happnes only on back panel port, right?

First, ps2_usb won't be updated actively anymore. Use ibmpc_usb instead. It should work on your converter hardware as is.

As you know hid_listen is not much useful to debug a issue around boot.
You will have to use debug on UART.  Connect PD3(TXD) of MCU to computer(other than target pc) using TTL(5V)-USB serial converter. You can check debug prints even before target computer boots OS.

To enable debug on UART build firmware adding lines below to Makefile. With these options debug prints are brought into serial console. 
```
TMK_LUFA_DEBUG = yes
TMK_LUFA_DEBUG_UART = yes
```

It seems that the problem is located around USB startup/reset. Check USB Events in file tmk_core/protocol/lufa/lufa.c first.

https://github.com/tmk/tmk_keyboard/blob/master/tmk_core/protocol/lufa/lufa.c#L271

LUFA is used as USB stack library, you can find its details here.
 
https://www.fourwalledcubicle.com/files/LUFA/Doc/210130/html/
https://github.com/abcminiuser/lufa

jeongsu816, 2023-11-14:
thank you. i ordered TTL-USB serial converter. and i will try ibmpc_usb firmware, first.

jeongsu816, 2023-11-16:
i tried ibmpc_usb and it is working properly. if i have any problem later, i will update.

thank you :)

closed 2023-11-16 by jeongsu816