tmk/tmk_keyboard #771: The TMK mapping for an Apple M0110F ("anorexic ISO Enter") keyboard has some issues, including "missing" the space bar.

mbfarah, 2024-01-24:
I recently got my hands on an Apple M0110F keyboard.

[image]

[image]

I plugged this keyboard with a TMK converter, and ran into some weird mappings, two of which can be handled, and one that can't.

1) ISO# and Enter are swapped (this can be waded around by swapping them in the mapping as well; see below).
2) The R4 alphas are all shifted left (this can be waded around by shifting them right in the mapping; see below).

[image]

However, the third one is escaping my ability to wade around it:

3) The spacebar produces nothing; whatever is mapped to it goes to the ⌤ key to its right.

As mapped in the above image, both ⌥ keys produce LCTRL, ⌘ produces LALT, the spacebar produces nothing, and ⌤ produces ESC instead of RALT.

The keyboard is in excellent condition, so I don't believe it's a matter of a faulty PCB or something like that, and that ⌤ is producing the space bar's output as well strengthens my suspicion the problem lies in the key count within TMK itself.

I also plugged a regular TMK converter to a vanilla M0110 (en-US layout)keyboard,  and all five keys in the bottom row produced their expected results.

Thank you.

tmk, 2024-01-25:
Can you check this post to add support the keyboard?

https://geekhack.org/index.php?topic=24965.msg3152999#msg3152999

mbfarah, 2024-01-25:
I followed the instructions on the GH post and this is what I got:

[image]

tmk, 2024-01-25:
hmm, it seems that converter cannot detect ISO layout using model ID unfortunately.  One of my US layout M0110 uses same model ID `03`. (another uses `09`) This is not what I expected, and I'm disappointed at this situation.

For ISO layout you have to configure manually by one of methods below.

## Build firmware with `M0110_INTL`
Compile with build option `M0110_INTL` to enable ISO layout.

For example, edit Makefile like below and run `make -f Makefile.unimap.rev2`.

```
diff --git a/converter/m0110_usb/Makefile.unimap.rev2 b/converter/m0110_usb/Makefile.unimap.rev2
index 1416155c..83a57dfd 100644
--- a/converter/m0110_usb/Makefile.unimap.rev2
+++ b/converter/m0110_usb/Makefile.unimap.rev2
@@ -2,5 +2,5 @@ TARGET = m0110_usb_rev2_unimap
 UNIMAP_ENABLE = yes
 KEYMAP_SECTION_ENABLE = yes
 MCU = atmega32u2
-#OPT_DEFS = -DM0110_INTL
+OPT_DEFS = -DM0110_INTL
 include Makefile
```

## Edit keymap for ISO
This is possible as you did on keymap editor, but it is not intuitive nor useful.

Space bar should register `RAlt` with your configuration above, fyi.
 

## Switch between US layout and ISO
by pressing key combo

This is not implemented yet, but I will try later.

mbfarah, 2024-01-25:
Looking at the PDF, this impresses me (in a negative way):

[image]

Indeed, all the scan codes moved around for no reason... are as-is in the original. Shame on Apple! (almost as much shame as having both Shift keys and both Option keys sending the same scan code each). Your suggestion of tailor-editing the keymap _for ISO keyboards_ seems to be the best one. I would add a note in the Unimap editor regarding this, if someone else should show up with the same "issue" as myself.

You are right: my keyboard's space bar should be returning RALT, but I get nothing. The keyboard is in excellent shape, so I discounted the possibility of that switch being dead... it now seems to me that this may be the reason for the space bar not returning anything.

Thank you once more.

mbfarah, 2024-01-26:
I managed to force the switch under the space bar to actuate, and the result was, indeed, the one you referenced from the manual.

With this in mind, and considering that: a) "anorexic ISO Enter" M0110x models are comparatively scarce; b) they (so far) cannot be programmatically identified; c) they can be handled with the Unimap editor as-is (albeit with the caveat of shifting three groups of keys around), I'd say the best course of action is, for the time being, to leave the matter as-is, but with a note explaining how "anorexic ISO" M0110x keyboards differ from "short ANSI" M0110 units.

Thank you once more.

closed 2024-01-26 by mbfarah

tmk, 2024-01-27:
[Keymap Editor for International layout](https://www.tmk-kbd.com/tmk_keyboard/editor/unimap/?m0110_usb_intl#456C4oOW4Kmg6biL5oKGcOOOiNeA5rWMxoDjkr/suIDrg6DhjIDqsqTgu4PrlIDng6DinI7Isu6zs+KykuyDjOK8nO2kgOuoue2MtuqFmeu1puSxiu6EoeySl+GKvOSskNuK5I2J4aWS64um7r6564O07bK755et7aK34Kuc45226ree65ym5pCK4Yuo5Z6h5pqE7K2V75GB6qGJ5LGo5L2I5K6K7o6G5IeG5JeG5K+M7Iam5LGG5IWI6rGA5JaX6LGt6Z6X6IGf6YGJ5IOM5q2L6K2B7K2F7K2I6ZaL7K2M5K2B5K2B6q2P5K2E7IOL6IOB7IOF7IOL7IOM5IOE4oig7Iag7LWA7L2A7Iuh6oWd5JGY5aGb6Yq865yu5KmC5Jai5JKy66GI5KCs5qii7Jmx7Jim6ruu54uJ5Jqp5o2x542G7I2M5KWG46mx7KWF75eE76W97qO024XuoaTliojotYvojZDng5TlsbvouIruvIjqhqjjkarmqIfhnIfnhrnqp6jmqKbjlZjspYTkpp3sqLTimIDir63kkqHkooHunaPhv4bpvYzmlp3ioKHhhKXgt6/knbjjjrnnvLrmgLLqj4jml7LppKLpubHskYnvkJborbniianlkrHusK/pl4rosrLqlZTrhZzqvZbqrLXjqo3lqqTlnqntlIrkuYbttbbrvZTivLbprK3pi6vkhqrthazmmJvhtrTonZ3qhZbupbTlt4Xpobfnkrntn6nmgIbrt4Hrvajnh57hrrbkn7Xth5Pkvrbtl63nkKfPnuy9iueYtOqjjemuk+imleS5veOameCym+eMnu+ziOeCtuGirOSaq+WGmuyxruucmOa5p+isqe2Lq+anlO27j+ubm+KWmeWlsuu/nuq4juqPhy/pgIA=) is available now.

I'll add description in the Geekhack thread and README file for International layout support.