tmk/tmk_keyboard #772: Model M IBM PC Keyboard Converter Issues.

jdcaperon, 2024-03-02:
When installing the firmware the HID console outputs:

[image]

It scrolls a bit until a key press is entered. I suspect the board is cooked, but I have no way to know. I have attempted to use the https://github.com/tmk/tmk_keyboard/wiki/Signal-Capture-for-debug/#signal-capture-firmware but have had no success.

If advised I'll acquire a logic analyzer to see if I can get it to work. I also haven't tried to add pull up resistors yet but can do that as well.

[image]
[image]
[image]

dx7, 2024-09-14:
Did you place the pull-up resistors?
https://github.com/tmk/tmk_keyboard/wiki/Pull%E2%80%90Up-Resistor

jdcaperon, 2024-09-15:
@dx7 let me double check my setup and get back to you