tmk/tmk_keyboard #781: DCA IRMAkey/3270 issues

SlaktarnP90, 2024-11-15:
Hi, I've just got my hands on a DCA IRMAkey/3270. I've ben told that it speaks XT and works with a IBM PC/XT (5160). I've wired up my Teensy with 1k Ohm pull upp resistors and connected the reset cable between Pin 3 on the socket and PB7 on the Teensy.
I'm using the prebuilt .hex-file from GitHub [ibmpc_usb_atmega32u4.hex]. 

My 122-key terminal IBM Model F & G80-1500 with AT connector works with this setup. But when I connect my IRMAkey I manage to get one key to register and then I get a lot of errors, and after a while I can get another key to register followed by a lot of errors.
<details><summary>Photo of keyboard</summary>
<p>

[image]

</p>
</details>  

<details><summary>Photo of Teensy, close</summary>
<p>

[image]

</p>
</details> 

<details><summary>Photo of Teensy, overview</summary>
<p>

[image]

</p>
</details>

This is what I get from "Hid_Listen" when I press the "Q"-key.
I wait for the errors to stop before pressing Q again. Sometimes it types the letter Q and sometimes not.

```
Device disconnected.
Waiting for new device:...
Listening:

TMK:6f48ed/LUFA:d6a7df

USB configured.

Loop start.
I2500 I2500
2530 PRT:22 ISR:AAC0 rAA A5501 wFF R R R R R R R R R R A6209 wFF R R R R R R R R R R
6918 ERR:11 ISR:8000
6919 ERR:11 ISR:8000 X7420 X7420
7462 ERR:20 ISR:5580 E7463
[CLR] I7463 A10464 wFF R R R R R R R R R R
11172 ERR:11 ISR:8000 X11674
11716 ERR:20 ISR:5580 E11717
[CLR] I11717 A14718 wFF R R R R R R R R R R
15426 ERR:11 ISR:8000 X15928
15970 ERR:20 ISR:5580 E15971
[CLR] I15971 A18972 wFF R R R R R R R R R R
19680 ERR:11 ISR:8000 X20182
20224 PRT:22 ISR:AAC0 rAA W20225 R20726
ID:FFFB(XT_CLONE) S20726 L20727
34341 ERR:20 ISR:2180 E34342
[CLR] I34342
34480 ERR:20 ISR:2180 E34481
[CLR] I34481 A37482 wFF R R R R R R R R R R
38190 ERR:11 ISR:8000 X38692
38734 ERR:20 ISR:5580 E38735
[CLR] I38735 A41736 wFF R R R R R R R R R R
42444 ERR:11 ISR:8000 X42946
42988 ERR:20 ISR:5580 E42989
[CLR] I42989 A45990 wFF R R R R R R R R R R
46698 ERR:11 ISR:8000 X47200
47242 ERR:20 ISR:5580 E47243
[CLR] I47243 A50244 wFF R R R R R R R R R R
50952 ERR:11 ISR:8000 X51454
51496 ERR:20 ISR:5580 E51497
[CLR] I51497 A54498 wFF R R R R R R R R R R
55206 ERR:11 ISR:8000 X55708
55750 ERR:20 ISR:5580 E55751
[CLR] I55751 A58752 wFF R R R R R R R R R R
59460 ERR:11 ISR:8000 X59962
60004 PRT:22 ISR:AAC0 rAA W60005 R60506
ID:FFFB(XT_CLONE) S60506 L60507
62405 ERR:20 ISR:2180 E62406
[CLR] I62406
62693 ERR:20 ISR:2180 E62693
[CLR] I62694 A159 wFF R R R R R R R R R R
867 ERR:11 ISR:8000 X1369
1411 ERR:20 ISR:5580 E1412
[CLR] I1412 A4413 wFF R R R R R R R R R R
5121 ERR:11 ISR:8000 X5623
5665 ERR:20 ISR:5580 E5666
[CLR] I5666 A8667 wFF R R R R R R R R R R
9375 ERR:11 ISR:8000 X9877
9919 PRT:22 ISR:AAC0 rAA W9920 R10421
ID:FFFB(XT_CLONE) S10421 L10422
13451 ERR:20 ISR:2180 E13451
[CLR] I13452
13570 ERR:20 ISR:2180 E13571
[CLR] I13571 A16572 wFF R R R R R R R R R R
17280 ERR:11 ISR:8000 X17782
17824 PRT:22 ISR:AAC0 rAA W17825 R18326
ID:FFFB(XT_CLONE) S18326 L18327
24193 ERR:20 ISR:2180 E24194
[CLR] I24194
24302 PRT:22 ISR:90C0 r90 A27195 wFF R R R R R R R R R R
27903 ERR:11 ISR:8000 X28405 rAA W28447 R28948
ID:FFFB(XT_CLONE) S28948 L28949 r10 r90 r10 r90 r10 r90
51294 ERR:20 ISR:2180 E51295
[CLR] I51295
51443 ERR:20 ISR:2180 E51444
[CLR] I51444 A54445 wFF R R R R R R R R R R
55153 ERR:11 ISR:8000 X55655
55697 PRT:22 ISR:AAC0 rAA W55698 R56199
ID:FFFB(XT_CLONE) S56199 L56200
62582 ERR:20 ISR:2180 E62583
[CLR] I62583
62701 ERR:20 ISR:2180 E62702
[CLR] I62702 A167 wFF R R R R R R R R R R
875 ERR:11 ISR:8000 X1377
1419 ERR:20 ISR:5580 E1420
[CLR] I1420 A4421 wFF R R R R R R R R R R
5129 ERR:11 ISR:8000 X5631
5673 PRT:22 ISR:AAC0 rAA W5674 R6175
ID:FFFB(XT_CLONE) S6175 L6175
9344 ERR:20 ISR:2180 E9344
[CLR] I9345
9503 ERR:20 ISR:2180 E9503
[CLR] I9504 A12505 wFF R R R R R R R R R R
13213 ERR:11 ISR:8000 X13715
13757 PRT:22 ISR:AAC0 rAA W13758 R14259
ID:FFFB(XT_CLONE) S14259 L14260
18053 ERR:20 ISR:2180 E18053
[CLR] I18054
18172 ERR:20 ISR:2180 E18172
[CLR] I18173 A21174 wFF R R R R R R R R R R
21882 ERR:11 ISR:8000 X22384
22426 PRT:22 ISR:AAC0 rAA W22427 R22928
ID:FFFB(XT_CLONE) S22928 L22929
25869 ERR:20 ISR:2180 E25869
[CLR] I25870
26007 PRT:22 ISR:90C0 r90 A28871 wFF R R R R R R R R R R
29579 ERR:11 ISR:8000 X30081
30123 ERR:20 ISR:5580 E30124
[CLR] I30124 A33125 wFF R R R R R R R R R R
33833 ERR:11 ISR:8000 X34335
34377 ERR:20 ISR:5580 E34378
[CLR] I34378 A37379 wFF R R R R R R R R R R
38087 ERR:11 ISR:8000 X38589
38631 ERR:20 ISR:5580 E38632
[CLR] I38632 A41633 wFF R R R R R R R R R R
42341 ERR:11 ISR:8000 X42843
42885 PRT:22 ISR:AAC0 rAA W42886 R43387
ID:FFFB(XT_CLONE) S43387 L43388 r10 r90 r10
2071 ERR:20 ISR:2180 E2072
[CLR] I2073 A5074 wFF R R R R R R R R R R
5782 ERR:11 ISR:8000 X6284
6326 ERR:20 ISR:5580 E6327
[CLR] I6327 A9328 wFF R R R R R R R R R R
10036 ERR:11 ISR:8000 X10538
10580 ERR:20 ISR:5580 E10581
[CLR] I10581 A13582 wFF R R R R R R R R R R
14290 ERR:11 ISR:8000 X14792
14834 ERR:20 ISR:5580 E14835
[CLR] I14835 A17836 wFF R R R R R R R R R R
18544 ERR:11 ISR:8000 X19046
19088 ERR:20 ISR:AB00 E19088
[CLR] I19089 A22090 wFF R R R R R R R R R R
22798 ERR:11 ISR:8000 X23300
23342 PRT:22 ISR:AAC0 rAA W23343 R23844
ID:FFFB(XT_CLONE) S23844 L23845 r10
27231 ERR:20 ISR:2180 E27232
[CLR] I27233 A30234 wFF R R R R R R R R R R
30942 ERR:11 ISR:8000 X31444
31486 PRT:22 ISR:AAC0 rAA W31487 R31988
ID:FFFB(XT_CLONE) S31988 L31988

``` 
Any advice would be much apriciated! :)

/Christoffer

tmk, 2024-11-16:
Hi,
The converter seems to miss the last bit of signals from keyboard(TIMEOUT error).  Its signaling is very slow beyond what the conveter expects probably.

For furture reference can you capture signal form of your keyboard?

You can use Signal Capture Firmware to do that.
Flash this prebuilt firmware to get text data of signals from keyboard.

https://github.com/tmk/tmk_keyboard/wiki/Signal-Capture-for-debug
https://raw.githubusercontent.com/tmk/tmk_keyboard/refs/heads/signal_capture/converter/ibmpc_usb/binary/ibmpc_capture_atmega32u4.hex

To capture signals on startup and on pressing a key
connect  the keyboard with the converter first,
plug the converter into a computer, wait for several seconds
and press Q key.
You should get some amount of text on hid_listen.

If you have a logic analyzer(or oscilloscope) please use it.

Thanks,
Jun

SlaktarnP90, 2024-11-16:
Good morning,

Thanks for your help!

This is what I get. It outputs "0001" once per second. I waited for one row to fill upp with "0001" and then pressed Q but nothing happened. I waited a while before pressing again but nothing happened. At the end I tried spamming the Q-key, but still nothing. I did the same thing 4 times and it seemed unresponsive each time.

```
Waiting for device:
Listening:

Device disconnected.
Waiting for new device:...................
Listening:
0021 0001 0001 6401 D543 0103 8941 9EF2 9F42 AD70 C273 C2B3 D0E1 E6A2 E6F2 F520 0100 0A33 0A73 18A1 2E62 2EB2 3CE0 51D3 5223 6051 7612 7662 8490 9993 99D3 A801 0001 0001 DE01 7A50 8732 1002 6460 64B1 7190 7E72 1002 5BA0 5BF1 68D0 75B2 1002 52E0 5331 6010 6CF2 1002 4A20 4A81 5750 6472 1002 4160 41C1 4E90 5BB2 1002 38A0 3901 4610 52F2 1002 3020 3081 3D50 4A32 1002 2760 27C1 3490 4172 1002 1EA0 1F01 2BD0 38B2 1002 15E0 1641 2310 2FF2 1002 0D30 0D81 8B63 9581 7B01 3A53 0903 8291 9842 9892 A6C0 BBC3 BC03 CA41 DFF2 E042 EE70 0100 0383 03C3 11F1 27B2 27F2 3630 4B33 4B83 59B1 6F62 6FB2 7DE0 92E3 9333 A161 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001
Device disconnected.
Waiting for new device:......................................................
Listening:
0021 0001 0001 6401 D593 0103 87B1 9D62 9DB2 ABE0 C0E3 C123 CF51 E522 E562 F390 0100 08A3 08E3 1711 2CD2 2D12 3B40 5043 5093 5EC1 7482 74C2 8300 9803 9843 A671 0001 0001 DE01 7AA0 8782 1002 64B0 6501 71E0 7EC2 1002 5BF0 5C41 6920 7602 1002 5330 5381 6060 6D42 1002 4A70 4AC1 57A0 64C2 1002 41B0 4201 4EE0 5C02 1002 38F0 3951 4660 5342 1002 3070 30C1 3DA0 4A82 1002 27B0 2811 34E0 41C2 1002 1EF0 1F51 2C20 3902 1002 1630 1691 2360 3042 1002 0D80 0DD1 8BB3 95D1 7B01 3AA3 0903 82A1 9862 98A2 A6E0 BBD3 BC23 CA51 E012 E052 EE90 0100 0393 03E3 1201 27C2 2812 3640 4B43 4B93 59C1 6F82 6FC2 7E00 9303 9343 A171 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001
Device disconnected.
Waiting for new device:......................................................
Listening:
0021 0001 0001 6401 D563 0103 7BB1 9172 91B2 9FF0 B4E3 B533 C361 D922 D962 E7A0 FCA3 FCE3 0103 0B21 20D2 2122 2F50 4453 44A3 52D1 6892 68D2 7710 8C03 8C53 9A81 0001 0001 DE01 7790 8472 1002 61A0 6201 6EE0 7BB2 1002 58E0 5941 6620 72F2 1002 5020 5081 5D60 6A32 1002 4760 47C1 54A0 61B2 1002 3EA0 3F01 4BE0 58F2 1002 35E0 3641 4360 5032 1002 2D60 2DC1 3AA0 4772 1002 24A0 2501 31E0 3EC2 1002 1BF0 1C41 2920 3602 1002 1330 1381 2060 2D42 1002 0A70 0AD1 88B3 92D1 7B01 37A3 0903 7FD1 9582 95D2 A400 B903 B953 C781 DD42 DD82 EBC0 0100 00C3 0113 0F31 24F2 2542 3370 4873 48B3 56F1 6CB2 6CF2 7B20 9023 9073 9EA1 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001
Device disconnected.
Waiting for new device:.........................................................................................................................................
Listening:
0021 0001 0001 6401 D563 0103 8851 9E12 9E52 AC80 C183 C1D3 D001 E5C2 E602 F440 0100 0943 0993 17C1 2D72 2DC2 3BF0 50F3 5143 5F71 7532 7572 83B0 98A3 98F3 A721 0001 0001 DE01 78C0 85A2 1002 62D0 6321 7000 7CE2  [... cut ...] 01 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001 0001
Device disconnected.
Waiting for new device:.................................................
``` 

/Christoffer

tmk, 2024-11-16:
Thanks for the data. I could not find expected signal in that though. I may miss something.

Anyway, can you try attached firmware? I modified code to relax the timeout error(5ms to 10ms).
[timeout_10ms.zip](https://github.com/user-attachments/files/17787642/timeout_10ms.zip)

tmk, 2024-11-19:
Thanks for testing.
I'll update github for the timeout error later. Before that I like to know about how long the keyboard signaling takes.

Can you try attached Signal Capture firmware again when you have time? I hope this works well this time.

[ibmpc_capture_atmega32u4.zip](https://github.com/user-attachments/files/17810299/ibmpc_capture_atmega32u4.zip)

SlaktarnP90, 2024-11-19:
Getting a ton of text while pressing Q-button atleast. I pressed the button with a few seconds between each press.

```
Waiting for new device:....
Listening:
0021 0001 0001 6401 D4D3 0103 7801 8DB2 8E02 9C30 B133 B183 BFB1 D572 D5B2 E3F0 F8E3 F933 0103 0771 1D22 1D62 2BA0 40A3 40E3 4F11 64D2 6522 7350 8853 88A3 96D1 A7A3 0003 0003 DE03 6961 7650 8332 1002 2580 25E1 32C0 3F92 0F02 E1D0 E231 EF00 FBE2 1002 9E40 9E91 AB70 B882 1002 5AA0 5B01 6810 74F2 1002 1740 17A1 2470 3152 0F02 D390 D3E1 E0C0 EDA2 1002 8FF0 9051 9D20 AA02 1002 4C60 4CB1 5990 66A2 1002 08C0 0921 1630 2312 0F02 C550 C5A1 D280 DF52 1002 81C0 8211 FFF3 0103 0891 7A01 C3A3 0A03 0BD1 2182 21D2 3000 4503 4543 5381 6952 6992 77C0 8CB3 8D03 9B31 B0F2 B142 BF70 D473 D4B3 E2F1 F8A2 F8F2 0102 0730 1C23 1C73 2AA1 3B73 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 A003 FDC1 0101 1382 13D2 21F0 37B2 4630 5BF2 6A70 8032 8EB0 A3A3 A3F3 B221 C7E2 C822 D660 EC22 FAA0 0100 1062 1EE0 2FA3 2FF3 3603 F761 0101 0D22 0D62 1B90 3152 3FD0 5592 6410 79D2 8850 9D43 9D93 ABC1 C182 C1C2 D000 E5C2 F440 0100 0943 0993 17B1 2883 0003 0003 0003 0003 FE03 0921 1EE2 1F22 2D60 4322 5190 6752 75D0 8B92 9A10 AF13 AF53 BD91 D342 D392 E1C0 F782 0102 0610 1BC2 2A40 3B13 3B53 2303 82E1 9892 98D2 A710 BCD2 CB40 E112 EF80 0100 0552 13C0 28C3 2903 3741 4D02 4D42 5B70 7132 7FB0 94B3 9503 A331 B403 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 8103 6091 7642 7692 84C0 9A82 A900 BEC2 CD40 E302 F180 0100 0683 06C3 14F1 2AB2 2B02 3930 4EF2 5D70 7332 81A0 9273 92C3 2D03 9451 AA02 AA52 B880 CE42 DCC0 F282 0102 0100 16C2 2530 3A33 3A83 48B1 5E72 5EB2 6CF0 82B2 9130 A623 A673 B4A1 C573 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 3B03 98F1 AEA2 AEF2 BD20 D2F2 E160 F722 0102 05A0 1B62 29D0 3ED3 3F23 4D51 6312 6352 7190 8752 95D0 AB92 BA00 CAD3 CB23 3C03 63C1 7972 79C2 87F0 9DB2 AC30 C1F2 D070 E632 F4B0 0100 09B3 0A03 1821 2DE2 2E32 3C60 5222 60A0 75A3 75E3 8411 94E3 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 5203 60E1 7692 76E2 8510 9AD2 A950 BF12 CD90 E352 F1C0 0100 06D3 0713 1541 2B02 2B52 3980 4F42 5DC0 7382 81F0 92C3 9313 2603 4911 5EC2 5F12 6D40 8302 9180 A742 B5C0 CB82 DA00 EF03 EF43 FD71 0101 1342 1382 21B0 3772 45F0 5AF3 5B33 6961 7A33 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 C903 0401 19C2 1A02 2830 3E02 4C70 6232 70B0 8672 94F0 A9F3 AA33 B861 CE22 CE72 DCA0 F262 0102 00F0 16A2 2520 35E3 3633 2803 5B21 70D2 7122 7F50 9512 A390 B952 C7D0 DD92 EC00 0100 0113 0163 0F81 2542 2582 33C0 4982 5800 6D03 6D43 7B71 8C43 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 2403 6631 7BE2 7C32 8A60 A022 AEA0 C462 D2E0 E8A2 F720 0100 0C23 0C73 1A91 3052 30A2 3ED0 5492 6310 78D2 8750 9813 9863 2103 7221 87D2 8822 9650 AC12 BA90 D052 DED0 F492 0102 0310 1813 1853 2681 3C42 3C92 4AC0 6082 6F00 8403 8443 9271 A343 0003 0003 0003 0003 0003 0003 0003 0003 0003 2703 9841 ADF2 AE42 BC70 D232 E0B0 F672 0102 04F0 1AB2 2920 3E23 3E73 4CA1 6262 62A2 70E0 86A2 9520 AAE2 B960 CA23 CA73 1F03 3491 4A42 4A82 58C0 6E82 7CF0 92C2 A130 B6F2 C570 DA73 DAB3 E8F1 FEA2 FEF2 0102 0D40 22E2 3160 4663 46B3 54E1 65A3 0003 0003 0003 0003 0003 0003 0003 0003 0003 0003 7E03 3961 4F12 4F52 5D90 7352 81C0 9792 A600 BBC2 CA40 DF43 DF83 EDC1 0101 0382 03D2 11F0 27C2 3630 4BF2 5A70 6B43 6B83 2303 B311 C8C2 C902 D740 ED02 FB80 0100 1152 1FB0 3572 43F0 58F3 5933 6771 7D32 7D72 8BB0 A172 AFE0 C4E3 C533 D361 E433 0003 0003 0003 0003
Device disconnected.
```

tmk, 2024-11-20:
My firmware did not work again and seems to have unknown bug.

Soarer's logic analyzer should work with the XT keyboard then.

Download 'sctrace_v1.01_ATmega32U4.zip' from here and unzip it to get 'sctrace_v1.01_portD_ATmega32U4.hex'.
And flash the hex file.
https://deskthority.net/viewtopic.php?f=7&t=4567

Can you share its outputs on hid_listen?

Thank you for your patiance.

SlaktarnP90, 2024-11-20:
This is what I get when I pressed Q a bunch of times (8 times maybe), reconnected keyboard once.

Thanks a lot for your help! :)

```
Waiting for new device:....
Listening:
sctrace v1.01
0009FF1 0009FF1 48A0FD0 5E5EFE0 6CDFFC0 81DBFF0 9057FD0 A614FE0 B494FC0 C991FF0
D80CFD0 EDCAFE0 FC49FC0 0009FC1 1147FF0 1FC2FD0 3581FE0 43FFFC0 58FDFF0 6778FD0
7846FF0 0009FF1 0009FF1 306CFD0 462BFE0 54A9FC0 6A69FE0 78E5FC0 8EA7FE0 9D22FC0
B2E3FE0 C15FFC0 D65CFF0 E4D8FD0 FA97FE0 000AFE1 0915FC0 1ED6FE0 2D53FC0 4312FE0
518EFC0 6259FF0 000AFF1 0009FF1 CD19FD0 E2D6FE0 F154FC0 000AFC1 0715FE0 1593FC0
2B51FE0 39D1FC0 4F8FFE0 5E0BFC0 7307FF0 8185FD0 9742FE0 A5C1FC0 BB81FE0 C9FDFC0
DEFAFF0 ED77FD0 FE42FF0 000AFF1 000BFF1 D129FD0 E6E8FE0 F566FC0 000CFC1 0B26FE0
19A2FC0 2F63FE0 3DDFFC0 539FFE0 621CFC0 7719FF0 8595FD0 9B54FE0 A9D2FC0 BF92FE0
CE0EFC0 E3CFFE0 F24CFC0 0009FC1 0317FF0 0009FF1 0009FF1 90CEFD0 A68CFE0 B50CFC0
CACBFE0 D947FC0 EF08FE0 FD84FC0 0009FC1 1344FE0 21C1FC0 36BDFF0 453AFD0 5AF8FE0
6978FC0 7F37FE0 8DB3FC0 A2B0FF0 B12DFD0 C1F8FF0 0009FF1 0009FF1 0CC5FD0 2282FE0
3100FC0 46C1FE0 553EFC0 6AFDFE0 797BFC0 8F3AFE0 9DB8FC0 B2B4FF0 C131FD0 D6EEFE0
E56CFC0 FB2DFE0 0009FE1 09A9FC0 1F6BFE0 2DE6FC0 3EB0FF0 000AFF1 000AFF1 A96FFD0
BF2DFE0 CDACFC0 E36CFE0 F1E9FC0 000AFC1 07A9FE0 1625FC0 2BE7FE0 3A64FC0 4F5FFF0
5DDBFD0 739BFE0 8219FC0 97D8FE0 A655FC0 BB53FF0 C9D0FD0 DA99FF0 0009FF1 000AFF1
C3DFFD0 D99DFE0 E81CFC0 FDDDFE0 0009FE1 0C58FC0 221AFE0 3096FC0 4655FE0 54D2FC0
69CFFF0 784BFD0 8E0AFE0 9C88FC0 B248FE0 C0C4FC0 D686FE0 E502FC0 F5CCFF0 0009FF1
000AFF1 CF70FD0 E52EFE0 F3ACFC0 000CFC1 0970FE0 17EBFC0 2DAAFE0 3C26FC0 51E6FE0
6062FC0 755FFF0 83DCFD0 999BFE0 A81BFC0 BDDAFE0 CC55FC0 E153FF0 EFCFFD0 0009FD1
0099FF0 0009FF1 0009FF1 7627FD0 8BE6FE0 9A63FC0 B025FE0 BEA3FC0 D460FE0 E2DDFC0
F89EFE0 0009FE1 0719FC0 1C16FF0 2A94FD0 4051FE0 4ECFFC0 6490FE0 730CFC0 88CCFE0
9749FC0 A813FF0 0009FF1 000CFF1 356AFD0 4B28FE0 59A7FC0 6F67FE0 7DE3FC0 93A7FE0
A220FC0 B7E0FE0 C65DFC0 DB59FF0 E9D6FD0 FF94FE0 000AFE1 0E13FC0 23D5FE0 3251FC0
474CFF0 55CAFD0 6694FF0 0009FF1 0009FF1 444BFD0 5A09FE0 6886FC0 7E47FE0 8CC3FC0
A283FE0 B100FC0 C6C1FE0 D53DFC0 EA39FF0 F8B6FD0 0009FD1 0E74FE0 1CF4FC0 32B4FE0
412FFC0 56EFFE0 656CFC0 7636FF0 0009FF1 0009FF1 7335FD0 88F3FE0 9772FC0 AD32FE0
BBB0FC0 D170FE0 DFEBFC0 F5ACFE0 0009FE1 0429FC0 1926FF0 27A1FD0 3D60FE0 4BDEFC0
619EFE0 701BFC0 8517FF0 9394FD0 A45FFF0 000BFF1 0009FF1 F5CFFD0 0009FD1 0B8CFE0
1A0BFC0 2FCBFE0 3E49FC0 5408FE0 6284FC0 7845FE0 86C3FC0 9BBEFF0 AA3AFD0 BFF8FE0
CE77FC0 E438FE0 F2B4FC0 0009FC1 0874FE0 16F2FC0 27BCFF0 0009FF1 0009FF1 23F7FD0
39B5FE0 4833FC0 5DF4FE0 6C71FC0 8230FE0 90ADFC0 A66DFE0 B4E9FC0 C9E7FF0 D864FD0
EE22FE0 FC9FFC0 0009FC1 1260FE0 20DDFC0 35DAFF0 4455FD0 5520FF0 0009FF1 000AFF1
9158FD0 A715FE0 B593FC0 CB54FE0 D9D1FC0 EF91FE0 FE0EFC0 0009FC1 13CDFE0 224AFC0
3746FF0 45C3FD0 5B81FE0 6A00FC0 7FC0FE0 8E3DFC0 A3FCFE0 B279FC0 C343FF0 0009FF1
0009FF1 509BFD0 6658FE0 74D8FC0 8A98FE0 9914FC0 AED5FE0 BD50FC0 D310FE0 E18DFC0
F689FF0 000AFF1 0507FD0 1AC6FE0 2943FC0 3F04FE0 4D7FFC0 627CFF0 70F9FD0 81C4FF0
0009FF1 000AFF1 8FD2FD0 A590FE0 B40FFC0 C9D0FE0 D84CFC0 EE0DFE0 FC88FC0 000AFC1
1248FE0 20C5FC0 35C1FF0 443FFD0 59FCFE0 687AFC0 7E3BFE0 8CB8FC0 A277FE0 B0F5FC0
C1BFFF0 0009FF1 0009FF1 720EFD0 87CDFE0 964BFC0 AC0BFE0 BA88FC0 D048FE0 DEC4FC0
F486FE0 000AFE1 0301FC0 17FEFF0 267AFD0 3C38FE0 4AB7FC0 6077FE0 6EF4FC0 83F0FF0
926DFD0 A338FF0 000AFF1 0009FF1
Device disconnected.
Waiting for new device:.....
Listening:
sctrace v1.01
0009FF1 0009FF1 484DFD0 5E0CFE0 6C8AFC0 8189FF0 9004FD0 A5C2FE0 B440FC0 C93DFF0
D7B9FD0 ED78FE0 FBF6FC0 0009FC1 10F4FF0 1F70FD0 352FFE0 43ADFC0 58AAFF0 6727FD0
77F2FF0 0009FF1 0009FF1 6329FD0 78E7FE0 8766FC0 9D25FE0 ABA2FC0 C162FE0 CFDEFC0
E5A1FE0 F41BFC0 0009FC1 0918FF0 1795FD0 2D53FE0 3BD1FC0 5192FE0 600EFC0 75CEFE0
844BFC0 9515FF0 0009FF1 000CFF1 D808FD0 EDC5FE0 FC43FC0 000AFC1 1204FE0 2081FC0
3642FE0 44BDFC0 5A7DFE0 68FAFC0 7DF7FF0 8C74FD0 A233FE0 B0AFFC0 C673FE0 D4EFFC0
E9E9FF0 F865FD0 0009FD1 0930FF0 000AFF1 0009FF1 F7ADFD0 0009FD1 0D6CFE0 1BE9FC0
31ABFE0 4026FC0 55E7FE0 6463FC0 7A23FE0 889FFC0 9D9DFF0 AC19FD0 C1D8FE0 D055FC0
E616FE0 F493FC0 0009FC1 0A52FE0 18D0FC0 2999FF0 0009FF1 0009FF1 B158FD0 C718FE0
D595FC0 EB55FE0 F9D3FC0 000AFC1 0F92FE0 1E0EFC0 33CFFE0 424BFC0 5748FF0 65C4FD0
7B84FE0 8A03FC0 9FC2FE0 AE3FFC0 C33BFF0 D1B7FD0 E282FF0 000AFF1 000CFF1 37D6FD0
4D94FE0 5C12FC0 71D3FE0 804FFC0 960FFE0 A48CFC0 BA4CFE0 C8C9FC0 DDC6FF0 EC44FD0
0009FD1 0200FE0 107FFC0 263FFE0 34BBFC0 4A7BFE0 58F8FC0 69C2FF0 0009FF1 0009FF1
1A12FD0 2FD0FE0 3E4FFC0 5410FE0 628BFC0 784DFE0 86C9FC0 9C88FE0 AB08FC0 C003FF0
CE7EFD0 E43CFE0 F2BBFC0 0009FC1 087CFE0 16F8FC0 2BF5FF0 3A72FD0 4B3CFF0 0009FF1
000AFF1 0550FD0 1B0EFE0 298DFC0 3F4DFE0 4DC9FC0 638AFE0 7206FC0 87C6FE0 9643FC0
AB3FFF0 B9BCFD0 CF7DFE0 DDFAFC0 F3BAFE0 000BFE1 0236FC0 17F6FE0 2672FC0 373DFF0
0009FF1 000AFF1 E78DFD0 FD4BFE0 0009FE1 0BC9FC0 218AFE0 3006FC0 45C7FE0 5443FC0
6A06FE0 787FFC0 8D7CFF0 9BF9FD0 B1B7FE0 C035FC0 D5F6FE0 E473FC0 F96FFF0 000AFF1
07EBFD0 18B6FF0 0009FF1 0009FF1 71EEFD0 87ABFE0 962AFC0 ABEAFE0 BA66FC0 D027FE0
DEA4FC0 F464FE0 000BFE1 02DFFC0 17DCFF0 2659FD0 3C17FE0 4A95FC0 6056FE0 6ED2FC0
8493FE0 9310FC0 A3DAFF0 0009FF1 0009FF1 0901FD0 1EBFFE0 2D3FFC0 42FEFE0 517AFC0
673BFE0 75B7FC0 8B78FE0 99F4FC0 AEF2FF0 BD6DFD0 D32BFE0 E1AAFC0 F76AFE0 0009FE1
05E6FC0 1AE4FF0 2961FD0 3A2BFF0 000AFF1 000AFF1 5A6FFD0 702BFE0 7EA8FC0 9469FE0
A2E6FC0 B8A5FE0 C722FC0 DCE4FE0 EB5FFC0 000CFC1 005BFF0 0ED9FD0 2497FE0 3314FC0
[... cut ...]
C133FE0 CFB2FC0 E572FE0 F3EFFC0 0009FC1 09AEFE0 182AFC0 2DECFE0 3C6AFC0 5165FF0
5FE1FD0 759FFE0 841DFC0 99DEFE0 A85AFC0 BD57FF0 CBD4FD0 DC9EFF0 0009FF1 0009FF1
B351FD0 C910FE0 D78EFC0 ED4EFE0 FBCBFC0 0009FC1 118CFE0 2008FC0 35CAFE0 4444FC0
5942FF0 67BDFD0 7D7BFE0 8BFAFC0 A1BBFE0 B037FC0 C5F7FE0 D473FC0 E53EFF0 000AFF1
0009FF1 0473FD0 1A33FE0 28B1FC0 3E71FE0 4CECFC0 62AEFE0 7129FC0 86EAFE0 9566FC0
AA62FF0 B8DFFD0 CE9DFE0 DD1CFC0 F2DCFE0 0009FE1 0159FC0 1656FF0 24D2FD0 359EFF0
0009FF1 000AFF1
Device disconnected.
```

tmk, 2024-11-20:
I could see signal form of the keyboard finally. Thank you.

It takes about 5ms to send a scan code, this is 5 times slow comparing to original IBM XT keyboard. Interesting.

https://github.com/tmk/tmk_keyboard/wiki/IBM-PC-XT-Keyboard-Protocol#ibm-pc-xt-and-lynk-122-key-terminallx-122-us

[image]

I updated code in github repo at commit https://github.com/tmk/tmk_keyboard/commit/3c7ca5b64bda3b7c98ce021c6e5f7d0a7a708877 .

You can download the fixed firmware from Keymap Editor now.

closed 2024-11-20 by tmk